margelo/react-native-vision-camera #1730: 🐛 V3: Object-oriented frame processors restrict return types

rkmackinnon, 2023-08-29:
### What were you trying to do?

I am trying to update the vision-camera-code-scanner library to be compatible with current RC builds of react-native-vision-camera V3.

### Reproduceable Code

```tsx
// Relevant types in FrameProcessorPlugins.ts
type BasicParameterType = string | number | boolean | undefined;
type ParameterType = BasicParameterType | BasicParameterType[] | Record<string, BasicParameterType | undefined>;

// Top-level type that vision-camera-code-scanner previously used to return
type Barcode = {
  boundingBox?: Rect;
  cornerPoints?: Point[];
  displayValue?: string;
  rawValue?: string;
  format: BarcodeFormat;
  content:
    | {
        type:
          | BarcodeValueType.UNKNOWN
          | BarcodeValueType.ISBN
          | BarcodeValueType.TEXT;
        data: string;
      }
    | {
        type: BarcodeValueType.CONTACT_INFO;
        data: ContactInfo;
      }
    | {
        type: BarcodeValueType.EMAIL;
        data: Email;
      }
    | {
        type: BarcodeValueType.PHONE;
        data: Phone;
      }
    | {
        type: BarcodeValueType.SMS;
        data: Sms;
      }
    | {
        type: BarcodeValueType.URL;
        data: UrlBookmark;
      }
    | {
        type: BarcodeValueType.WIFI;
        data: Wifi;
      }
    | {
        type: BarcodeValueType.GEO;
        data: GeoPoint;
      }
    | {
        type: BarcodeValueType.CALENDAR_EVENT;
        data: CalendarEvent;
      }
    | {
        type: BarcodeValueType.DRIVER_LICENSE;
        data: DriverLicense;
      };
};
```

### What happened instead?

I am not sure if this is technically a bug or a feature request. Previously, frame processors could return any sort of complex object under the `unknown` type, and now they are restricted to `ParameterType` as defined in FrameProcessorPlugins.ts.

We cannot recompile vision-camera-code-scanner to be compatible with RC4 and future versions of react-native-vision-camera without causing breaking changes for users of previous versions of vision-camera-code-scanner due to this. The current API would require us to create a new return type flattening the `Barcode` return type into something that is compatible with `ParamterType`. Since the `Barcode` type mirrors Google's MLKit Barcode-Scanning return type, we could previously pass all this information back to the JS side directly. It would be nice to be able to continue to do this.

Would it be possible to change the return type back to `unknown`?

### Relevant log output

```shell
N/A
```

### Device

N/A

### VisionCamera Version

3.0.0-rc.4 and higher

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-08-30:
Hey! The change does make sense since the return type of a frame processor plugin cannot be anything else other than what's defined in `ParameterType`. It is either a string, number, boolean, or an array or object containing such primitives. 

You can typecheck/cast on the JS/TS side yourself, but this is now more type-safe.

Or am I missing something?

rkmackinnon, 2023-08-30:
I took another look and I think this is the issue: your typings don't allow for an object that contains other objects that contain those primitives. Our previous return types had, e.g., Rect and Point, which were themselves objects of primitives.

rkmackinnon, 2023-08-30:
Also, if it helps, BarcodeFormat and BarcodeValueType are enums.

rkmackinnon, 2023-09-02:
Something like this:

```
diff --git a/src/FrameProcessorPlugins.ts b/src/FrameProcessorPlugins.ts
index cb5366f5cbebb6472b77b4450c50f9ca8e965ef2..d49f709e26a8e7970501826f0ca86f70e44b0628 100644
--- a/src/FrameProcessorPlugins.ts
+++ b/src/FrameProcessorPlugins.ts
@@ -7,17 +7,16 @@ import type TWorklets from 'react-native-worklets-core';
 import { CameraModule } from './NativeCameraModule';
 import { assertJSIAvailable } from './JSIHelper';
 
-type BasicParameterType = string | number | boolean | undefined;
-type ParameterType = BasicParameterType | BasicParameterType[] | Record<string, BasicParameterType | undefined>;
+type ParameterType = string | number | boolean | undefined | ParameterType[] | { [key: string]: ParameterType };
 
-interface FrameProcessorPlugin {
+interface FrameProcessorPlugin<T extends ParameterType> {
   /**
    * Call the native Frame Processor Plugin with the given Frame and options.
    * @param frame The Frame from the Frame Processor.
    * @param options (optional) Additional options. Options will be converted to a native dictionary
    * @returns (optional) A value returned from the native Frame Processor Plugin (or undefined)
    */
-  call: (frame: Frame, options?: Record<string, ParameterType>) => ParameterType;
+  call: (frame: Frame, options?: Record<string, ParameterType>) => T;
 }
 
 interface TVisionCameraProxy {
@@ -27,7 +26,7 @@ interface TVisionCameraProxy {
    * Creates a new instance of a Frame Processor Plugin.
    * The Plugin has to be registered on the native side, otherwise this returns `undefined`
    */
-  getFrameProcessorPlugin: (name: string) => FrameProcessorPlugin | undefined;
+  getFrameProcessorPlugin: <T extends ParameterType>(name: string) => FrameProcessorPlugin<T> | undefined;
 }
 
 let hasWorklets = false;

```

allows me to write code like this:

```typescript
const plugin = VisionCameraProxy.getFrameProcessorPlugin<Barcode[]>('scanCodes');

export function scanBarcodes(
  frame: Frame,
  types: BarcodeFormat[],
  options?: CodeScannerOptions
): Barcode[] {
  'worklet';

  if (plugin == null)
    throw new Error('Failed to load Frame Processor Plugin "scanCodes"!');

  return plugin.call(frame, {
    ...(options || defaultOptions),
    types: Array.from(types),
  });
}
```

and not get any typescript errors with mismatched types from `plugin.call(...)` and the return type of `scanBarcodes`. What do you think of this approach? If you like it, feel free to use the code, or I can make a PR with it.

mrousavy, 2023-09-04:
Yea I get what you mean but essentially this isn't type-safe since your native API could return `"bla"` and you expect `Barcode[]`, you know what I mean?

rkmackinnon, 2023-09-04:
Ahh, I finally understand what you're saying now. _I_ know my native code doesn't do that, but if there was a bug in it, the side effects for the vision camera code could be handled more gracefully if you didn't allow me to extend that return type and forced me to do my own typechecking (or let me shoot myself in the foot and explicitly cast it).

I think the first modification to handle that "object that contains object that contains primitive type" case is still valid though. If I were to implement my own typechecking code, I wouldn't currently be able to do it without explicitly casting to overcome that limitation. Would you be okay with making just that first change and not the parts where I added the generics?

Thanks for taking the time to read this. I definitely learned something. Always a good day when you learn something!

mrousavy, 2023-09-05:
Yepp 100%! If you want to intentionally not be safe you can always do a `as any` cast :)