margelo/react-native-vision-camera #1751: ❓ FaceDetect frameProcessor with v3

vahidid, 2023-09-02:
### Question

Hi, I'm waiting for version 3 for a long time and now I want to add a face detector frame processor but i'm not a good Java developer. Can any body write an example for android and iOS? Please!!!

### What I tried

I tried to use the old lib [vision-camera-face-detector](https://github.com/rodgomesc/vision-camera-face-detector) to migrate them to v3 but i failed.

### VisionCamera Version

3.0.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

manolo-battista, 2023-09-04:
on iOS I have this error when I try to build with `"react-native-vision-camera": "^3.0.0-rc.10" / "3.0.0", and "vision-camera-face-detector": "^0.1.8"

[image]

duyanhv, 2023-09-05:
i've found a solution base on this guide: https://github.com/mrousavy/react-native-vision-camera/blob/b300209e3618a0ae6e46db8e0e9290d2beb09baf/docs/docs/guides/FRAME_PROCESSOR_CREATE_PLUGIN_IOS.mdx
you will have to setup a plugin like in the guide, and paste all of the content from VisionCameraFaceDetector callback function and it will work again 👍

manolo-battista, 2023-09-06:
@duyanhv I tried to follow the guide through the automatic and manual process, but I'm not so strong on XCode and I found some problems. I used the automatic process, I copied callback and relatives functions, all build correctly, but I received undefined from my `detectFaces` function. 

I have to do something on `AppDelegate.m`? Cause when I tried to add my function I have problems on build.

Can you share your solution?

duyanhv, 2023-09-06:
> @duyanhv I tried to follow the guide through the automatic and manual process, but I'm not so strong on XCode and I found some problems. I used the automatic process, I copied callback and relatives [...]

that's exactly what i did.
after running `npx vision-camera-plugin-builder ios` i got 2 file: 
[image]

FaceDetectorFrameProcessorPlugin.m
[image]

FaceDetectorFrameProcessorPlugin.swift
[image]

You don't have to do anything else, VisionCameraProxy will automatically pickup the plugin. 
On the JS side, i have something like this (declared it outside the react component)
[image]

And this is what i have in useFrameProcessor
[image]

Hope this help.

closed 2023-09-30 by mrousavy