margelo/react-native-vision-camera #1752: Android on Init Error🐛

hamdij0maa, 2023-09-02:
### What's happening?

[image]

### Reproduceable Code

```tsx
<ReanimatedCamera
                  style={globalStyles.cameraStyle}
                  device={device}
                  lowLightBoost={
                    device.supportsLowLightBoost && enableNightMode
                  }
                  isActive={isActive && cameraEnabled}
                  onInitialized={onInitialized}
                  onError={onError}
                  enableZoomGesture={false}
                  animatedProps={cameraAnimatedProps}
                  orientation="portrait"
                  photo={true}
                  video={true}
                  audio={hasMicrophonePermission}
                  //@ts-ignore
                  ref={camera}
                />
```

### Relevant log output

```shell
<img width="529" alt="image" src="https://github.com/mrousavy/react-native-vision-camera/assets/20734472/ca120a36-9a23-4b81-9fd6-8d34b10c37dc">
```

### Camera Device

```shell
Android Simulator Pixel 5
```

### Device

Pixel 5

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-04:
You can reproduce this in the example app? I can't. C++ part of VideoPipeline is being built properly...

Yasha33, 2023-09-05:
I have the same error when pass `video={true}`

hamdij0maa, 2023-09-05:
yes same problem

hamdij0maa, 2023-09-05:
Here is the error showing when start recording : 
[image]

mrousavy, 2023-09-05:
Sorry I have no idea what you guys are trying to do (no reproduction steps) and I don't have the full code. The example app works.

kocaagasefa, 2023-09-06:
https://github.com/mrousavy/react-native-vision-camera/issues/1769#issuecomment-1708240712