margelo/react-native-vision-camera #1755: Applying color filters to the camera❓

pTinosq, 2023-09-02:
### Question

I have a very simple system set up with a frame processor plugin set up to return a string (just to test communication between processor and ts is functioning).

I would like to somehow apply effects to my images (contrast, grain, brightness etc.) and I am not too sure how to go about doing this.

Would you do this in the frame processor plugin or is it not needed at all?

### What I tried

I have tried to return an Image from the frame processor with errors being thrown. Not too sure how to send an edited image back,

### VisionCamera Version

3.0.0

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

mrousavy, 2023-09-04:
Hey there was some work to bring Skia into VisionCamera which allowed you to do all of that. I decided to remove it as it greatly increased the complexity of VisionCamera. If you want that for your business, reach out to my agency (margelo.io) and we can build a tailored solution for you :)

closed 2023-09-04 by mrousavy

pTinosq, 2023-09-04:
Ah that's fair enough. So for hobby/free projects there is no support for this sort of thing?

mrousavy, 2023-09-05:
nope, not at the moment.

mrousavy, 2024-04-12:
See https://github.com/mrousavy/react-native-vision-camera/pull/2727 👀