margelo/react-native-vision-camera #1756: 🐛 : The 'VisionCamera/Frame.h' was not found in the V3 version.

rbayuokt, 2023-09-04:
### What's happening?

first of all, thank you for your hard work in making V3! I'm genuinely thrilled about it, and I appreciate the countless hours you've dedicated to this project. I'm currently working on creating a `vision-camera-pose-detection` project using MLKit for V3. I've already written the native code for iOS, but I've encountered an issue that says
`'VisionCamera/Frame.h' file not found.'` 

is there a solution to this problem? will there be new documentation available soon? I've been following the GitHub setup for the latest V3 version, but when I check the Vision Camera documentation on the web, it still seems to be for V2. 

could you please provide some guidance on this issue or perhaps share a link to any relevant resources? I couldn't find it in the GitHub issues. Thank you!

I'm using RN 0.72.4

### Relevant log output

```shell
ios:
/Users/xxxx/Desktop/xxxxx/xxxxx/ios/PoseDetection.h:14:9: fatal error: 'VisionCamera/Frame.h' file not found

normal x86_64 objective-c com.apple.compilers.llvm.clang.1_0.compiler
```

### Device

iOS 16.2

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-04:
Ah Frame.h is only exposed if worklets are enabled hold on

mrousavy, 2023-09-04:
Wait are you sure you installed `react-native-worklets-core` in your test project? thats required for frame processors

rbayuokt, 2023-09-04:
> Wait are you sure you installed `react-native-worklets-core` in your test project? thats required for frame processors

ah, you're right! `react-native-worklets-core` was indeed the missing piece I overlooked 😅. 
now, the error is gone. however, I still have a question, are we still supposed to use `VISION_EXPORT_FRAME_PROCESSOR` to export our native function on iOS? I'm encountering an error related to that.

[image]

Previously, in V2, I didn't encounter this error, even when I added `@end ` at the end of the line as suggested by Xcode.

rbayuokt, 2023-09-04:
oops sorry, my bad! fixed now, thanks a lot @mrousavy 🚀

[image]