margelo/react-native-vision-camera #1757: 🐛  V3 setState using useFrameProcessor doesn't work

manolo-battista, 2023-09-04:
### What's happening?

I'm trying to use `useFrameProcessor` to analyze the frame and detect if the luminosity of camera is enough. When light is not enough i want to set a state variable and show a message on screen. I saw for `setState` it is possible use `runOnJs` function as described here #305 but it seems doesn't work

### Reproduceable Code

```tsx
const [checkLight, setCheckLight] = useState<boolean>(false);

const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    console.log(`Received a ${frame.width} x ${frame.height} Frame!`);
    const pixels = frame.toArrayBuffer();
    let luminanceSum = 0;
    // Get luminance from pixel average
    for (let i = 0; i < pixels.length; i++) {
      luminanceSum += pixels[i];
    }
    const result = luminanceSum / pixels.length;
    runOnJS(setCheckLight)(result < 80);
  }, []);
```

### Relevant log output

```shell
ERROR  Frame Processor threw an error: Regular javascript function '_f' cannot be shared. Try decorating the function with the 'worklet' keyword to allow the javascript function to be used as a worklet.
```

### Camera Device

```shell
{
  "maxZoom": 128.875,
  "formats": [
    {
      "photoWidth": 4032,
      "supportsVideoHDR": false,
      "photoHeight": 3024,
      "fieldOfView": 73.29196166992188,
      "maxFps": 60,
      "videoHeight": 144,
      "supportsPhotoHDR": false,
      "autoFocusSystem": "none",
      "videoWidth": 192,
      "minFps": 60,
      "pixelFormats": [
        "yuv",
        "yuv",
        "rgb",
        "unknown",
        "unknown",
        "unknown"
      ],
      "maxZoom": 189,
      "videoStabilizationModes": [
        "auto",
        "off"
      ],
      "maxISO": 2208,
      "minISO": 23
    },
    {
      "fieldOfView": 73.29196166992188,
      "photoWidth": 4032,
      "minFps": 60,
      "minISO": 23,
      "pixelFormats": [
        "yuv",
        "yuv",
[... 1475 lines cut ...]
    "wide-angle-camera"
  ],
  "hasTorch": false
}
```

### Device

iPhone 14 Pro

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-04:
V3 doesn't use reanimated. see docs 
[image]

closed 2023-09-04 by mrousavy

manolo-battista, 2023-09-04:
@mrousavy I saw it, but I not found a way to set variable on state. WIth or without reanimated doesn't work.