margelo/react-native-vision-camera #1763: Como acessar a base64 da imagem através do frame que vem do useFrameProcessor❓

MateusHoffman, 2023-09-04:
### Question

```javascript
  const frameProcessor = useFrameProcessor(async (frame: any) => {
    'worklet'
    
  }, []);
``` 
I have this code above, it has the frame in this format below:

```javascript
frame === {"bytesPerRow": 1280, "close": [Function hostFunction], "height": 720, "isValid": true, "planesCount": 3, "toString": [Function hostFunction], "width": 1280}
```

I need to get the base64 or blob of this frame, what can I do to achieve this?

```javascript
  "dependencies": {
    "expo": "~48.0.18",
    "react": "18.2.0",
    "react-native": "0.71.8",
    "react-native-reanimated": "~2.14.4",
    "react-native-vision-camera": "^2.15.4"
  },
  ```

### VisionCamera Version

"react-native-vision-camera": "^2.15.4"

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

vonovak, 2023-09-05:
hello; I don't think you want to do that. base64 is slooow. Please don't use it 🙂

closed 2023-09-05 by mrousavy

MateusHoffman, 2023-09-05:
@mrousavy Please do not set to complete. I know base64 is slow but I still need it, can you help me?

mrousavy, 2023-09-05:
Didn't I reply two times already to your message? You also posted this question on other issues and tagged me.

Again; you can convert the raw GPU frame to a byte buffer using frame.toByteArray().
Do whatever you want with the bytes from there on

LPitonakova, 2024-05-29:
I would also like to know this. I need to upload base64 data to my API. How do you proceed after `const file = await this._camera.takePhoto();` ?

v1n33th, 2024-08-13:
Frame processor plugins such as [tony-xlh/vision-camera-cropper](https://github.com/tony-xlh/vision-camera-cropper) can also be used to get base64 from frame.

SANDEEP1938, 2025-04-23:
> Didn't I reply two times already to your message? You also posted this question on other issues and tagged me.
> 
> Again; you can convert the raw GPU frame to a byte buffer using frame.toByteArray(). Do whatever you want with the bytes from there on

frame.toByteArray()  is throwing error: Property 'toByteArray' does not exist on type 'Frame'.

kitolog, 2025-07-20:
You can do it this way:
`
import {
    Camera,
    useCameraDevice,
    useCameraFormat,
    useCameraPermission,
} from "react-native-vision-camera";
import * as FileSystem from 'expo-file-system';

....

const photo = await cameraRef.current?.takePhoto();

let base64Image = null;
try {
    base64Image = await FileSystem.readAsStringAsync(`file:/${photo.path}`, {encoding: 'base64'});
} catch (e: any) {
    console.log('e');
    console.log(e);
}

//do smth with the base64 image
...
`