margelo/react-native-vision-camera #1766: App crash while using useFrameProcessor (V3:Android) 🐛

akshay8033, 2023-09-05:
### What's happening?

My app is crashing when I'm trying to enable FrameProcessor. The app works fine without FrameProcessor. 

### Reproduceable Code

```tsx
import * as React from 'react';
import { StyleSheet, View, Text, Button } from 'react-native';
import {
  Camera,
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';

export default function App() {
  // Permissions
  React.useEffect(() => {
    async function getPermissions() {
      await Camera.requestCameraPermission();
    }
    getPermissions();
  }, []);

  // Device
  const devices = useCameraDevices('wide-angle-camera');
  const device = devices.back;

  // Frame processor
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    console.log('Captured frame');
  }, []);

  if (device == null) return <Text>Loading camera...</Text>;
  return (
    <>
      <View style={styles.container}>
        <Camera
          frameProcessor={frameProcessor}
          device={device}
          isActive={true}
          style={StyleSheet.absoluteFill}
        />
      </View>
      <View>
        <Button title="Capture" />
      </View>
    </>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    alignItems: 'center',
    justifyContent: 'center',
  },
});
```

### Relevant log output

```shell
There is not error logged.
```

### Device

Samsung Galaxy S10+ 

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-05:
Ah just saw this.

You didn't put any logs in here. Put the logs in here, or I can't help you.

akshay8033, 2023-09-05:
Downgraded react-native to 0.72.3 and it worked @mrousavy

mashad6, 2024-02-17:
the issue still exist !

mashad6, 2024-02-17:
[image]
[image]

mashad6, 2024-02-17:
@mrousavy these were the logs when i use frameProcessor

mashad6, 2024-02-17:
@akshay8033 were you able to find the solution?

mrousavy, 2024-02-19:
This is a cache issue. `yarn start --reset-cache`