margelo/react-native-vision-camera #1768: 🐛  [V3] (Android) Initilization Error On Galaxy S23 Ultra

elirichey, 2023-09-05:
### What's happening?

I'm trying to setup V3 on a Galaxy S23 Ultra. After launching and accepting permissions, the camera does not initialize properly. 

onInitialized camera property never fires off. 

I've tried using both front and back cameras. Result is the same

### Reproduceable Code

```tsx
I'm running the example app without frame processors enabled.
```

### Relevant log output

```shell
ERROR  [unknown/unknown: [unknown/unknown] No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)]
```

### Camera Device

```shell
{
  "formats": [REMOVED BECAUSE OUTPUT IS TOO LONG - WILL NOT ACCEPT TICKET],
  
  "hardwareLevel": "level-3",
  "pixelFormats": [
    "RAW_SENSOR",
    "JPEG",
    "PRIVATE",
    "YUV_420_888",
    "RAW_PRIVATE",
    "RAW10",
    "YCBCR_P010"
  ],
  "maxZoom": 10,
  "minZoom": 0.6000000238418579,
  "supportsLowLightBoost": true,
  "sensorOrientation": "landscape-right",
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "hasTorch": true,
  "devices": [
    "ultra-wide-angle-camera"
  ],
  "hasFlash": true,
  "name": "BACK (0)",
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Galaxy S23 Ultra

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

elirichey, 2023-09-05:
I see someone posted the same issue here: https://github.com/mrousavy/react-native-vision-camera/issues/1769

I can confirm that if video={false}, then the camera works. But if video is set to true, it produces the error above.

kocaagasefa, 2023-09-06:
https://github.com/mrousavy/react-native-vision-camera/issues/1769#issuecomment-1708240712

fedu44, 2023-09-12:
[image]

mrousavy, 2023-09-30:
Hey - this should be fixed with the latest update in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy