margelo/react-native-vision-camera #1769: 🐛 V3 - Camera not initialized in Android when video prop is set to true.

dsipahioglu, 2023-09-05:
### What's happening?

I'm trying to record videos, but the camera output is black in Android even when the permissions are given. I'm giving the permissions from app settings to create this issue faster. I have no issues in taking photos, and onInitialized is fired as expected, but when the **video prop is set to true**, onInitialized is never fired.

Project information:

    "react-native": "0.72.3",
    "react-native-reanimated": "^3.3.0",
    "react-native-vision-camera": "^3.0.0",

Node version: v16.20.1

### Reproduceable Code

```tsx
import { useIsFocused } from '@react-navigation/native';
import React, {
  useCallback,
  useEffect, useRef
} from 'react';
import { Button, View } from 'react-native';
import { Camera, useCameraDevices } from 'react-native-vision-camera';

function TryNewCameraScreen() {
  const cameraRef = useRef(null);
  const isFocused = useIsFocused();

  const devices = useCameraDevices('wide-angle-camera');
  const device = devices.back;

  const onCameraInitialized = useCallback(() => console.log('camera initialized'), []);

  useEffect(() => {
    if (device) {
      console.log('device is', device);
    }
  }, [device]);

  useEffect(() => {
    if (cameraRef) {
      console.log('cameraRef is', cameraRef);
    }
  }, [cameraRef]);

  const takePicture = async () => {
    if (cameraRef.current !== null) {
      const image = await cameraRef.current.takePhoto();
      console.log('image', image);
    }
  };

  const recordVideo = async () => {
    try {
      cameraRef.current.startRecording({
        flash: 'on',
        onRecordingFinished: (video2) => console.log('success', video2),
        onRecordingError: (error) => console.log('error', error),
      });
    } catch (e) {
      console.log('err', e);
    }
  };

  const stopRecording = async () => {
    try {
      await cameraRef.current.stopRecording();
    } catch (e) {
      console.log('err', e);
    }
  };

  return (
    <View>
      {device == null ? <></> : (
        <>
          <Camera
            ref={cameraRef}
            style={{ height: '80%' }}
            isActive={isFocused}
            video
            audio={false}
            photo
            device={device}
            onInitialized={onCameraInitialized}
          />
          <View style={{ flexDirection: 'row', justifyContent: 'space-evenly' }}>
            <Button
              color="#841584"
              accessibilityLabel="TAKE PICTURE"
              title="TAKE PICTURE"
              onPress={takePicture}
            />
            <Button
              color="#841584"
              accessibilityLabel="CAPTURE VIDEO"
              title="CAPTURE VIDEO"
              onPress={recordVideo}
            />
            <Button
              color="#841584"
              accessibilityLabel="CAPTURE VIDEO"
              title="STOP VIDEO"
              onPress={stopRecording}
            />
          </View>
        </>
      )}
    </View>
  );
}

export default TryNewCameraScreen;
```

### Relevant log output

```shell
error {
    "userInfo": null,
    "message": "[session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!",
    "cause": {
        "stacktrace": "com.mrousavy.camera.CameraNotReadyError: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!\n\tat com.mrousavy.camera.core.CameraSession.startRecording(CameraSession.kt:265)\n\tat com.mrousavy.camera.CameraView_RecordVideoKt.startRecording(CameraView+RecordVideo.kt:47)\n\tat com.mrousavy.camera.CameraViewModule$startRecording$1.invokeSuspend(CameraViewModule.kt:85)\n\tat kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)\n\tat kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)\n\tat kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)\n",
        "message": "[session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!"
    },
    "code": "session/camera-not-ready"
}
```

### Device

Xiaomi Redmi Note 8 - Android 11 - MIUI Global 12.5.2

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jide, 2023-09-05:
Same issue on Redmi Note 5G

bjornhutter, 2023-09-06:
Same issue on Asus Zenfone 8

kocaagasefa, 2023-09-06:
enabling [frameProcessor](https://www.react-native-vision-camera.com/docs/guides/frame-processors) and installing [react-native-worklets-core](https://github.com/margelo/react-native-worklets-core) solved the issue. 

Maybe required C++ part of VideoPipeline is being built if only frameProcessors is enabled.

Reproduced same issue with example project by removing `react-native-worklets-core` and disabling `frameProcessor`

iernstig, 2023-09-08:
I observed the same behavior, video only works when frameProcessors are enabled!

mattlewer, 2023-09-12:
> enabling [frameProcessor](https://www.react-native-vision-camera.com/docs/guides/frame-processors) and installing [react-native-worklets-core](https://github.com/margelo/react-native-worklets-core) [...]

This solution allows me to open the camera with video, but as soon as I press to record I get a ton of errors about width, not sure if this is related or I should open my own issue.
[image]

fedu44, 2023-09-12:
[image]

mrousavy, 2023-09-14:
Please share your adb logcat logs.

dsipahioglu, 2023-09-14:
> Please share your adb logcat logs.

```
No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)
E/CameraView: Failed to configure session: No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)
    java.lang.UnsatisfiedLinkError: No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)
        at com.mrousavy.camera.core.VideoPipeline.initHybrid(Native Method)
        at com.mrousavy.camera.core.VideoPipeline.<init>(VideoPipeline.kt:51)
        at com.mrousavy.camera.core.outputs.CameraOutputs.<init>(CameraOutputs.kt:122)
        at com.mrousavy.camera.core.CameraSession.configureSession(CameraSession.kt:137)
        at com.mrousavy.camera.CameraView.configureSession(CameraView.kt:205)
...
```

I searched for this issue, and that you have said https://github.com/mrousavy/react-native-vision-camera/pull/1789 might help. I'll try this asap.

mrousavy, 2023-09-30:
Hey - this should be fixed with the latest update in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

R4tch3t, 2023-10-17:
bug on example camera no reload when app is in background and back to foreground fixed adding `{device != null && **isActive &&** (...`

ombogdanJoinToIt, 2023-10-17:
not fixed

R4tch3t, 2023-10-18:
> not fixed

any sugestion ?... ONLY for worklets try, background to foreground: `
useEffect(()=>{
    if(!isActive){
      setReRender(true)
    }else{
      setTimeout(()=>{
         setReRender(false)
       },333)
    }
  },[isActive])`

other feature example change FPS: 
`onPress={() => {
      setReRender(true)              
      setTimeout(()=>{
        setTargetFps((t) => (t === 30 ? 60 : 30))
        setReRender(false)
      },50)
}}`

this updates not are a good experience... :/

sumitthakur11, 2023-10-29:
Has anyone fixed this? why this issue is closed if not fixed?

R4tch3t, 2023-10-29:
> Has anyone fixed this? why this issue is closed if not fixed?

is fixed update all yours deps RNReanimated and Vision camera to latest, now if you're using worklets yes exist a problem when pixelFormat="yuv" but it's not the topic of this thread

amanrathore18, 2023-11-16:
@dsipahioglu 

Any update?

amanrathore18, 2023-11-16:
@mrousavy 

This issue still persist for me. Any solution?

DavichoStar, 2023-11-16:
I still have this problem, this is what I receive in Sentry only when I generate the apk for production, in debug mode it works fine. how stressful.

```bash
unknown/unknown: [unknown/unknown] no "Lcom/facebook/jni/HybridData;" field "mHybridData" in class "Lcom/mrousavy/camera/core/VideoPipeline;" or its superclasses
```

DavichoStar, 2023-11-16:
Update on my specific issue: The production app now works!!! I did several things to it but in the end it seems that it was **deactivating proguard**, I don't like it so much that I have to deactivate it but I like that the video works better.

amanrathore18, 2023-11-17:
@DavichoStar 

Your solution is working fine for APK created by command line or android studio but when i've uploaded .aab bundle on PlayStore and downloaded the same from PlayStore in my device then i'm facing same issue.

mrousavy, 2023-11-17:
@DavichoStar thanks, just fixed this in https://github.com/mrousavy/react-native-vision-camera/pull/2169 :) If you appreciate my work and dedication to the library please consider [💖 sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to keep the project running!