margelo/react-native-vision-camera #1770: 🐛 ld: error: undefined symbol: typeinfo for RNWorklet::JsiHostObject

DeveloperMCD, 2023-09-06:
### What's happening?

Getting all kinds of linking errors after I upgraded to Vision Camera 3.0.0 (release)
RC6 was building just fine for me.

        "react-native": "0.72.4",
        "react-native-vision-camera": "3.0.0",
        "react-native-worklets-core": "^0.2.0",

### Reproduceable Code

```tsx
npx react-native start
```

### Relevant log output

```shell
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/.cxx/Debug/5g5c3o2b/arm64-v8a'
  [1/1] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/5g5c3o2b/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/5g5c3o2b/obj/arm64-v8a/libVisionCamera.so 
  : && /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/5g5c3o2b/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/cpp/JSITypedArray.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VideoPipeline.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/PassThroughShader.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLContext.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/OpenGLRenderer.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/FrameProcessorPluginHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/VisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrame.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraScheduler.cpp.o  /home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so  -landroid  /home/matthew/.gradle/caches/transforms-3/ed5971bff14b029db9d61e0025901397/transformed/jetified-react-android-0.72.4-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /home/matthew/.gradle/caches/transforms-3/ed5971bff14b029db9d61e0025901397/transformed/jetified-react-android-0.72.4-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /home/matthew/.gradle/caches/transforms-3/d2945effe2c76af5abf2618fac389cac/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  -lGLESv2  -lEGL   -latomic -lm && :
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::getWorkletRuntime()
  >>> referenced by JFrameProcessor.cpp:37 (/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:37)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by JVisionCameraProxy.cpp:57 (/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:57)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(vision::JVisionCameraProxy::~JVisionCameraProxy())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnJsThread(std::__ndk1::function<void (facebook::jsi::Runtime&)>&&)
  >>> referenced by JFrameProcessor.cpp:50 (/home/matthew/dev/stackbayapp/node_modules/react-native-vision-camera/android/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp:50)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(vision::JFrameProcessor::callWithFrameHostObject(std::__ndk1::shared_ptr<vision::FrameHostObject> const&) const)
  >>> referenced by WKTJsiWorklet.h:358 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:358)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::runtimeMappings
  >>> referenced by __tree:0 (/home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/__tree:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long))
  >>> referenced by __tree:0 (/home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/__tree:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::call(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long))
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::invokeOnWorkletThread(std::__ndk1::function<void (RNWorklet::JsiWorkletContext*, facebook::jsi::Runtime&)>&&)
  >>> referenced by WKTJsiWorklet.h:361 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:361)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::WorkletInvoker::~WorkletInvoker())
  
  ld: error: undefined symbol: RNWorklet::JsiWorkletContext::defaultInstance
  >>> referenced by memory:3887 (/home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:3887)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  >>> referenced by memory:3887 (/home/matthew/programs/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:3887)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  
  ld: error: undefined symbol: RNWorklet::JsiHostObject::JsiHostObject()
  >>> referenced by WKTJsiWorkletContext.h:31 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorkletContext.h:31)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JFrameProcessor.cpp.o:(RNWorklet::JsiWorkletContext::getDefaultInstance())
  >>> referenced by WKTJsiWorklet.h:77 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorklet.h:77)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o:(RNWorklet::JsiWorklet::JsiWorklet(facebook::jsi::Runtime&, facebook::jsi::Value const&))
  
  ld: error: undefined symbol: vtable for RNWorklet::JsiWorkletContext
  >>> referenced by WKTJsiWorkletContext.h:31 (/home/matthew/dev/stackbayapp/node_modules/react-native-worklets-core/android/build/headers/rnworklets/react-native-worklets-core/WKTJsiWorkletContext.h:31)
[... cut ...]
### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

DeveloperMCD, 2023-09-06:
I thought CMakeLists.txt would be fixed now -- apparently it isn't.
You still need to add:

set(ENABLE_FRAME_PROCESSORS ON)

towards the beginning of the file.

barghi, 2023-09-06:
Thanks @DeveloperMCD 
It fixed the problem

mrousavy, 2023-09-30:
Hey - I've seen similar issues with RN Worklets cc @chrfalch

mrousavy, 2023-10-19:
I don't get how this can be off when it even logs that it's on...

juanevp, 2023-11-03:
@mrousavy I don't quite grasp the details, but it seems that the actual value passed to ENABLE_FRAME_PROCESSORS when defined is relevant.
Currently, at https://github.com/mrousavy/react-native-vision-camera/blob/8f986a45ea0aa3398e45a806a264fbcc03278971/package/android/build.gradle#L110C17-L110C18, it is being passed the value of hasWorklets directly which translates to the literal ```true``` in my case as I want to include frameprocessors.  And the build fails because it cannot find the worklets lib during linking. If I change the argument passed at that line to the expression: ```hasWorklets ? "ON" : "OFF"``` (which has an analogous effect to hard-coding the value ON as suggested by @DeveloperMCD), then the build finishes successfully.  I don't know much about cmake, but my guess is that the values ON/OFF are associated with the boolean type, whereas the literal ```true``` would be of string type. And that distinction seems to be relevant here.

mrousavy, 2023-11-07:
@juanevp wow, if that's the case then that's ridiculous 😂 

Did you test if it works fine with and without Worklets using your change?

closed 2023-11-09 by mrousavy

yknx4, 2024-05-03:
I still experience flaky build issues. It works perfectly for hours, then I have to redo the build and it fails again. It is very inconsistent, is there any other workarounds?

christophemenager, 2024-09-09:
Same as @yknx4, I have never patched this lib (as suggested by @DeveloperMCD) and had frame processors enabled without trouble for about a year now.

Without any reason, I had to apply this patch (which works fine) today...

ronickg, 2025-01-31:
The only working fix for me was clearing the global .gradle cache, node_modules + bun.lock (yarn.lock..) and the android generated build, cxx folders (With expo prebuild its easier, just prebuild --clean).

devunlimit-austin, 2025-04-02:
Mobile development is a rapidly evolving field, making projects like these highly complex, especially with limited contributor resources. If you're experiencing frustration, consider contributing a fix yourself or providing financial support to mrousavy. He's addressing complex issues the community hasn't yet resolved.

zrina1314, 2025-04-03:
I created my own branch and integrated ZXing into it, not through the method of frame-processor-plugins. Instead, I directly modified the part of QR code recognition in the source code, which can automatically determine whether the mobile device supports Google services. If the device supports Google services, MLKit will be used; if not, ZXing will be used.

Reason: Although Google MLKit is indeed very useful, in our country, Google services are inaccessible. Almost all Android mobile devices do not have Google services. Therefore, I had to use ZXing.

If you think this approach is feasible, I can submit a Pull Request (PR).

Here is the branch of my code:
https://github.com/zrina1314/react-native-vision-camera