margelo/react-native-vision-camera #1771: ❓`frame.image.imageFormat` in frame processor is PRIVATE(34). Is it intended?

bglgwyng, 2023-09-06:
### Question

I found the code below from MLKit's face detector failed with the following error message.
```kotlin
val image = InputImage.fromMediaImage(frame.image, frame.imageInfo.rotationDegrees)
```
```
Exception in HostFunction: java.lang.IllegalArgumentException: Only JPEG and YUV_420_888 are supported now
```
So I investigated the image format of the `frame.image` and it was [`PRIVATE`(34)](https://developer.android.com/reference/android/graphics/ImageFormat#PRIVATE).
I've already observed that MLKit's face detector worked fine with RNVC V2. So I suppose that this is a change from RNVC v3.

I don't fully understand the description of `PRIVATE` format in the android official document linked, but it doesn't appear to be a commonly employed value.
Is it a bug? Or did I miss something?

I tested it on a Galaxy S20.

### VisionCamera Version

3.0.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

rocket13011, 2023-09-08:
same issue

xseignard, 2023-09-08:
`Orientation.valueOf(frame.getOrientation()).toDegrees()`

GaylordP, 2023-09-09:
Same issue :/

jjbwii, 2023-09-11:
Hello,

‘yuv’ crash my app (some internal function… it seems when VideoPipeline.kt tries to trabsform surface to image)

‘rgb’ works for me but there is no image bytes to copy behind… I need this info in order to implement some image post-processing with opencv…

jlsneto, 2023-09-12:
> Hello,
> 
> ‘yuv’ crash my app (some internal function… it seems when VideoPipeline.kt tries to transform surface to image)
> 
> ‘rgb’ works for me but there is no image bytes to copy behind… I need this info in order to implement some image post-processing with opencv…

I believe it has to do with the ordering of camera formats. In version 2.x the frame was 1920x1280, in this version only RGB "works" and the frame is 1920x1920. The bytes are 14,745,600 I believe it is (1920*1920*4)

rocket13011, 2023-09-12:
hello everyone, with this PR there are no more YUV worries, 

Basically, the problem is that the OpenGL pipeline was RGB-only.

so this pipeline was abandoned and switched to ImageWriter.

https://github.com/mrousavy/react-native-vision-camera/pull/1789

mrousavy, 2023-09-13:
Hey- yea this PR fixes the frame processor issue once and for all, but it breaks for video recordings. Not sure why, as the `ImageWriter` should be writing Frames perfectly fine to the `MediaRecorder`'s surface-  I need to investigate this; https://github.com/mrousavy/react-native-vision-camera/pull/1799

jlsneto, 2023-09-13:
> Hey- yea this PR fixes the frame processor issue once and for all, but it breaks for video recordings. Not sure why, as the `ImageWriter` should be writing Frames perfectly fine to the [...]

When will the release come out?

kyle-bowden, 2023-09-13:
I tested this change([#1799](https://github.com/mrousavy/react-native-vision-camera/pull/1799)) 

Pixel format set as "yuv" for frameprocessor
`pixelFormat="yuv"`

and I get this error now

`com.facebook.jni.CppException: Compiling JS failed: 1:1:invalid empty parentheses '( )' Buffer size 3 starts with: 280a29 and has protection mode(s): rw-p
                                                                                                    	at com.mrousavy.camera.frameprocessor.FrameProcessor.call(Native Method)
                                                                                                    	at com.mrousavy.camera.core.VideoPipeline.onImageAvailable(VideoPipeline.kt:79)`

It is failing in the VideoPipeline.kt on `fp.call(frame)`

itujo, 2023-09-28:
Hey, I have the same issue, have you guys figured out how to solve that?

mrousavy, 2023-09-29:
fixed with 3.2.0 :)

closed 2023-09-29 by mrousavy

mgcrea, 2023-10-02:
@mrousavy Is the format supposed to be something else now? In my frame processor using `3.2.2` I still get `34` eg. `ImageFormat.PRIVATE`.

mrousavy, 2023-10-02:
@mgcrea yup you need to pass the pixel format, I could maybe explain this a bit better in the docs?