margelo/react-native-vision-camera #1772: ❓  Camera: `device` is null!

Panigale, 2023-09-06:
### Question

I have already obtained the permissions, but encountered this error when opening the camera. I tried using Camera.getAvailableCameraDevices() to see the camera modules I can use, but received the following response:
{"_h": 0, "_i": 0, "_j": null, "_k": null}

my testing device is iphone 13 pro max  and ios 16.6. Below is the complete code:

```
import React from 'react';
import {
  Camera,
  useCameraDevices
} from 'react-native-vision-camera';
import { StyleSheet } from 'react-native';

export default function CameraScreen(){
  // const devices = useCameraDevices()
  const devices = Camera.getAvailableCameraDevices()
  console.log(devices)
  const device = devices.back

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={true}
    />
  )
}
```

and here is my packages:
```
  "dependencies": {
    "@react-navigation/bottom-tabs": "^6.5.8",
    "@react-navigation/native": "^6.1.7",
    "@react-navigation/native-stack": "^6.9.13",
    "@rneui/base": "^4.0.0-rc.7",
    "@rneui/themed": "^4.0.0-rc.7",
    "expo": "~48.0.18",
    "expo-barcode-scanner": "^12.5.3",
    "expo-splash-screen": "~0.18.2",
    "expo-status-bar": "~1.4.4",
    "react": "18.2.0",
    "react-native": "0.72.3",
    "react-native-drawer-layout": "^3.2.1",
    "react-native-gesture-handler": "^2.12.0",
    "react-native-pager-view": "^6.2.0",
    "react-native-reanimated": "^3.5.0-nightly-20230903-79201c8c0",
    "react-native-safe-area-context": "4.5.0",
    "react-native-screens": "~3.20.0",
    "react-native-tab-view": "^3.5.2",
    "react-native-vector-icons": "^9.2.0",
    "react-native-vision-camera": "^3.0.0",
    "react-native-web": "^0.19.6",
    "react-native-webview": "^13.2.2"
  },
```

### VisionCamera Version

3.0.0

### Additional information

- [X] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

bjornhutter, 2023-09-06:
`const devices = Camera.getAvailableCameraDevices();`
Returns a promise so you would have to await it.
`const devices = await Camera.getAvailableCameraDevices();`
But that can't be done directly in the component body so easier to just use the useCameraDevices hook like this:

```typescript
const [cameraPosition, setCameraPosition] = React.useState<'front' | 'back'>(
    'back',
  );

const devices = useCameraDevices();
const device = devices[cameraPosition];
```

You could also check the example app: https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/src/CameraPage.tsx

closed 2023-09-06 by Panigale

steavenb, 2025-05-19:
Faced this on Android, so I realised we need to show the Camera component only if the device is ready.

Instead of 

`
  <Camera
          style={styles.absoluteFill}
          device={device}
          isActive={true}
          codeScanner={codeScanner}
        />
`

I put a check like this to handle the error.

`
      {device ? (
        <Camera
          style={styles.absoluteFill}
          device={device}
          isActive={true}
          codeScanner={codeScanner}
        />
      ) : null}
`