margelo/react-native-vision-camera #1773: 🐛  'CaptureRequest contains unconfigured Input/Output Surface!' on camera device switched

bglgwyng, 2023-09-06:
### What's happening?

I tried to update `Camera`s `device` prop.

### Reproduceable Code

```tsx
https://github.com/bglgwyng/Expo49RNVC3Sandbox

Push `flip` button to switch the camera between front and back.

I wasn't able to reproduce the issue in the VisionCamera Example app due to some build errors. I expect it enough, since it's a tiny example.
```

### Relevant log output

```shell
java.lang.IllegalArgumentException: CaptureRequest contains unconfigured Input/Output Surface!
  android.hardware.camera2.CaptureRequest.convertSurfaceToStreamId(CaptureRequest.java:765)
  android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:1289)
  android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:1345)
  android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:313)
  com.mrousavy.camera.core.CameraSession.updateRepeatingRequest(CameraSession.kt:548)
  com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:526)
  com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:51)
  com.mrousavy.camera.core.CameraSession$configureFormat$1.invokeSuspend(CameraSession.kt:185)
  kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
  kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
  android.os.Handler.handleCallback(Handler.java:942)
  android.os.Handler.dispatchMessage(Handler.java:99)
  android.os.Looper.loopOnce(Looper.java:226)
  android.os.Looper.loop(Looper.java:313)
  android.os.HandlerThread.run(HandlerThread.java:67)
```

### Device

Galaxy S20

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

gravitypersists, 2023-09-07:
I also ran into this issue by trying to use a mix of the example code on the root readme and the example project - what fixed this for me was specifying a `format`

OpachiEg, 2023-09-17:
I have the same problem. Setting the key and format properties did not help

mrousavy, 2023-09-27:
This is so weird. I think I am really locking the access to the surfaces and mutate fully atomically, so I have no idea where that issue is coming from (I dont have that device, so I cannot reproduce the issue).

jkaufman, 2023-09-27:
I encounter the same issue on a Pixel 3a. It reproduces any time I set video to true, not only when switching.

jkaufman, 2023-09-27:
I encounter the same issue using RNVC 3.1 on a Pixel 3a when both video and photo are enabled.

jkaufman, 2023-09-27:
I'm seeing this same error on a Pixel 3a running v3.1 when I enable both photo and video _and_ toggle between FPS settings, even if the capture format remains the same. I ran a suite of tests and could not reproduce it with any other combination of settings.

CarlosMayoralMX, 2023-09-29:
> adding <Camera key={device.id} // add this device={device} ... />
> 
> fixed it for me

This works for me when switching front/back cameras, but it crashes when using camera in a Tab Navigator and navigating between tabs.

mrousavy, 2023-09-29:
1. This is where the session gets re-built if you switch the device: https://github.com/mrousavy/react-native-vision-camera/blob/9bad0bb99eaf0ad88e6d045a7d46f585d510d815/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L511-L561
2. And this is where I update the capture request: https://github.com/mrousavy/react-native-vision-camera/blob/9bad0bb99eaf0ad88e6d045a7d46f585d510d815/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L563-L581

As you can see it's both under a mutex, so I wonder where this is coming from. Are you taking a photo or recording a video while flipping the camera or does this just happen in the preview?

I need the device to test this, I really wonder what happens here.

If anyone wants to debug this and put some breakpoints or logs there, please feel free and share your results with me

mrousavy, 2023-09-30:
Let me know if anyone could throw breakpoints or logs in there and reproduce the issue please! I need your help to fix this, as I cannot reproduce the issue.

bglgwyng, 2023-09-30:
@mrousavy  I just tested the following code. See the injected code for debugging.
```kotlin
  private suspend fun startRunning() {
    isRunning = false
    val cameraId = cameraId ?: return
    if (!isActive) return

    Log.i(TAG, "Starting Camera Session...")
    Log.d("foo", "startRunning, try to obtain lock")
    try {

      mutex.withLock {
        try {
          Log.d("foo", "startRunning, lock obtained")
          val outputs = outputs
          // ...
        } finally {
          Log.d("foo", "startRunning, lock released")
        }

```

```
2023-09-30 20:18:51.628  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, try to obtain lock
2023-09-30 20:18:51.628  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, lock obtained
2023-09-30 20:18:51.628  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, lock released
2023-09-30 20:18:51.641  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, try to obtain lock
2023-09-30 20:18:51.641  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock obtained
2023-09-30 20:18:51.895  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock released
2023-09-30 20:18:51.895  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, try to obtain lock
2023-09-30 20:18:51.895  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, lock obtained
2023-09-30 20:18:51.896  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, lock released

// switch device from back to front

2023-09-30 20:19:01.642  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, try to obtain lock
2023-09-30 20:19:01.642  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock obtained
2023-09-30 20:19:01.642  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock released
2023-09-30 20:19:01.642  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, try to obtain lock
2023-09-30 20:19:01.642  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, lock obtained
2023-09-30 20:19:01.643  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  updateRepeatingRequest, lock released
2023-09-30 20:19:01.688  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, try to obtain lock
2023-09-30 20:19:01.688  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock obtained
2023-09-30 20:19:01.981  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, try to obtain lock
2023-09-30 20:19:01.981  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock obtained
2023-09-30 20:19:01.986  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock released
2023-09-30 20:19:01.988  8705-8889  foo                     com.gwyng.Expo49RNVC3Sandbox         D  startRunning, lock released
```
Before `switch device from back to front`, the pairs of `lock obtained` / `lock released` are valid.
After `switch device from back to front`, there's a `lock obtained` before the release of the previous lock.
Is this what you expected to see?

bglgwyng, 2023-09-30:
Also, I can't reproduce this error in the example app. The camera flips work fine there.
This log is from a very minimal project with Expo 49 + RNVC V3.

kavuri, 2023-10-01:
I can reproduce it everytime with 2 ButtomNavigation tabs. When the app starts, the camera is on and everything is fine. When I switch to another tab and move back to the Record tab, app crashes with the error from this post
`<Camera
            key={device.id}
            ref={camera}
            device={device}
            isActive={true}
            video={true}
            style={{
              width: SCREEN_WIDTH,
              height: SCREEN_HEIGHT,
            }}
            format={format}
            audio={false}
            videoStabilizationMode={
              supportsVideoStabilization ? 'cinematic-extended' : undefined
            }
          />`

I have the check `{device != null && hasPermission ? (` before I show the camera

kavuri, 2023-10-01:
Solved the problem by unmounting the camera on blur. Used this option in the navigation tab:
`unmountOnBlur: true`

CarlosMayoralMX, 2023-10-01:
I realised that using the basic implementation of the camera in a Tab Navigator it will start ok, if you switch the camera (Front / Back) it will crash so I added  `key={device.id}` and it works if you only switch front/back. Navigate between tabs make the app crashes. The I started to add element by element and until realise that `isActive = {true} ` is the option is causing trouble. You have to use : `isActive = {isActive} `

```
    // check if camera page is active
    const isFocussed = useIsFocused()
    const isForeground = useIsForeground()
    const isActive = isFocussed && isForeground
```
But is not stable all the times, sometimes crash (I think that maybe is caused by the RN life cycle). Adding `unmountOnBlur: true` as @kavuri commented makes more stable.

I'm using:

"expo": "^49.0.13",
"react-native": "0.72.5",
"react-native-reanimated": "3.5.0",
 "react-native-vision-camera": "3.1.0",
 
 
 classpath('com.android.tools.build:gradle:7.4.1')
 distributionUrl=https\://services.gradle.org/distributions/gradle-7.5.1-all.zip

mrousavy, 2023-10-03:
Hm, weird. I think this might be happening when the Surface from the `PreviewView` gets destroyed by Android (Android kills `SurfaceView` when its no longer on screen, idk why - iOS is much better at this). https://github.com/mrousavy/react-native-vision-camera/blob/9bad0bb99eaf0ad88e6d045a7d46f585d510d815/package/android/src/main/java/com/mrousavy/camera/core/PreviewView.kt#L25-L37

Can you guys please all share your **full** adb logs so I can follow the stacktrace? Please paste them inside a `<details>` section like this:

```
<details>
<summary>My adb logs</summary>

\``` <-- without the \
LOGS GO HERE
\``` <-- without the \

</details>
```

bglgwyng, 2023-10-04:
@mrousavy Here's my log. It's too large to fit in a comment.
https://gist.github.com/bglgwyng/9bb5b5120a304167df52439ecca250c3

mrousavy, 2023-10-05:
> Calling JS function after bridge has been destroyed: RCTEventEmitter.receiveEvent

Huh, does your app fully crash? Or are you doing a hot reload/CMD+S here?

mrousavy, 2023-10-05:
Okay - some info in your logs:

1. For the first time/configuration, it destroys the Preview Surface, then creates it again and update with size is called. (all in `NativePreviewView`)
2. For the second time, it destroys the Preview Surface, but never re-creates it.
 
    https://gist.github.com/bglgwyng/9bb5b5120a304167df52439ecca250c3#file-gistfile1-txt-L1008

I have no idea why it doesn't re-create it here, I need to debug this further

hsjoberg, 2023-10-06:
I'm also experiencing this problem, but with flipping camera while using the code scanner.

It can be observed with this code:

```TSX
import React, { useState, useEffect } from "react";
import { Button, StyleSheet, Text, View } from "react-native";
import { Camera, useCodeScanner, CameraPosition, useCameraDevice, useCameraPermission } from "react-native-vision-camera";

export default function SendCamera() {
  const [cameraType, setCameraType] = useState<CameraPosition>("back");
  const device = useCameraDevice(cameraType);
  const { hasPermission, requestPermission } = useCameraPermission();
  const codeScanner = useCodeScanner({
    codeTypes: ["qr"],
    onCodeScanned: (codes) => {}
  });

  useEffect(() => {
    if (hasPermission === false) {
      console.log("Does not have camera permission");
      requestPermission().catch(() => {});
    }
  }, [requestPermission, hasPermission]);

  const onCameraSwitchClick = () => {
    setCameraType(
      cameraType === "front"
        ? "back"
        : "front"
    );
  };

  if (!hasPermission) {
    return (<Text>Camera permission was denied</Text>);
  }

  if (!device) {
    return (<></>);
  }

  return (
    <>
      <Camera
        style={StyleSheet.absoluteFill}
        codeScanner={codeScanner}
        device={device}
        isActive={true}
      />
      <View style={{ marginTop: 100 }}>
        <Button title="flip" onPress={onCameraSwitchClick} />
      </View>
    </>
  );
};
```

mrousavy, 2023-10-06:
I might have an idea; https://github.com/mrousavy/react-native-vision-camera/pull/1944

(not tested yet! also needs some code cleanups, I wanna make this absolutely atomic so it never goes out of sync)

mrousavy, 2023-10-06:
Thanks for the repro @hsjoberg! I'll try to reproduce that on my Huawei P10

flexingCode, 2023-10-06:
I have the problem when i try to comeback to the camera screen, i mean, when i do the thing with the camera and go to next screen, when i try to comeback, show the critical error screen

maxs1m, 2023-10-12:
The same problem was solved {device != null && isFocussed && (<Camera .../>)}

CarlosMayoralMX, 2023-10-12:
Regarding this issue I achieve build a stable version (Moto device and API 33, 31 simulators) with this settings https://github.com/mrousavy/react-native-vision-camera/issues/1988#issuecomment-1759970445

hsjoberg, 2023-10-12:
The bug is observable in an Android 14 emulator, and presumably also on <14, so you can just try that.

I tried looking into the bug last week, but perhaps with some pointers I could look at it again and try solving the bug (I don't have any prior experience with the codebase).

ldstein, 2023-10-18:
I ran into the `CaptureRequest contains unconfigured Input/Output Surface!`  issue while upgrading an old project from RNVC 2.12.0 to 3.3.1.

The problem disappeared after updating to 3.4.0. Huzzah!

closed 2023-10-18 by mrousavy