margelo/react-native-vision-camera #1774: 🔧 ninja: error: manifest 'build.ninja' still dirty after 100 tries

xulihang, 2023-09-06:
### How were you trying to build the app?

I tried to build the app for Android and I encountered this error: ninja: error: manifest 'build.ninja' still dirty after 100 tries

### Full build logs

```tsx
C:\Users\admin\AppData\Local\Android\Sdk\cmake\3.22.1\bin\ninja.exe -d explain -C G:\git\vision-camera-dynamsoft-barcode-reader\example\node_modules\react-native-vision-camera\android\.cxx\Debug\6l4u185s\armeabi-v7a VisionCamera
[0/1] Re-running CMake...
VisionCamera: Frame Processors enabled!
-- Configuring done
-- Generating done
-- Build files have been written to: G:/git/vision-camera-dynamsoft-barcode-reader/example/node_modules/react-native-vision-camera/android/.cxx/Debug/6l4u185s/armeabi-v7a
ninja explain: output ../prefab/armeabi-v7a/prefab/lib/arm-linux-androideabi/cmake/react-native-worklets-core/react-native-worklets-coreConfig.cmake of phony edge with no inputs doesn't exist
ninja explain: ../../../../CMakeLists.txt is dirty
ninja explain: ../prefab/armeabi-v7a/prefab/lib/arm-linux-androideabi/cmake/ReactAndroid/ReactAndroidConfig.cmake is dirty
ninja explain: ../prefab/armeabi-v7a/prefab/lib/arm-linux-androideabi/cmake/ReactAndroid/ReactAndroidConfigVersion.cmake is dirty
ninja explain: ../prefab/armeabi-v7a/prefab/lib/arm-linux-androideabi/cmake/fbjni/fbjniConfig.cmake is dirty
ninja explain: ../prefab/armeabi-v7a/prefab/lib/arm-linux-androideabi/cmake/fbjni/fbjniConfigVersion.cmake is dirty
ninja explain: ../prefab/armeabi-v7a/prefab/lib/arm-linux-androideabi/cmake/react-native-worklets-core/react-native-worklets-coreConfig.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeCInformation.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeCXXInformation.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeCommonLanguageInclude.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeGenericSystem.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeInitializeConfigs.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeLanguageInformation.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeSystemSpecificInformation.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/CMakeSystemSpecificInitialize.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Compiler/CMakeCommonCompilerMacros.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Compiler/Clang-C.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Compiler/Clang-CXX.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Compiler/Clang.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Compiler/GNU.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Android-Clang-C.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Android-Clang-CXX.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Android-Clang.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Android-Common.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Android-Initialize.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Android.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/Linux.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/cmake/3.22.1/share/cmake-3.22/Modules/Platform/UnixPaths.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/adjust_api_level.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/flags.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/hooks/post/Android-Clang.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/hooks/post/Android.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/hooks/pre/Android-Clang.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/hooks/pre/Android-Initialize.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/hooks/pre/Android.cmake is dirty
ninja explain: C:/Users/admin/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/platforms.cmake is dirty
ninja explain: CMakeCache.txt is dirty
ninja explain: CMakeFiles/3.22.1-g37088a8-dirty/CMakeCCompiler.cmake is dirty
ninja explain: CMakeFiles/3.22.1-g37088a8-dirty/CMakeCXXCompiler.cmake is dirty
ninja explain: CMakeFiles/3.22.1-g37088a8-dirty/CMakeSystem.cmake is dirty
```

### Project dependencies

```tsx
{
  "name": "VisionCameraDynamsoftBarcodeReaderExample",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "pods": "pod-install --quiet"
  },
  "dependencies": {
    "react": "18.2.0",
    "react-native": "0.72.4",
    "react-native-vision-camera": "^3.0.0",
    "react-native-worklets-core": "^0.2.0"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0",
    "@babel/preset-env": "^7.20.0",
    "@babel/runtime": "^7.20.0",
    "@react-native/eslint-config": "^0.72.2",
    "@react-native/metro-config": "^0.72.11",
    "babel-plugin-module-resolver": "^5.0.0",
    "metro-react-native-babel-preset": "0.76.8"
  },
  "engines": {
    "node": ">=16"
  }
}
```

### Target platforms

Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

- [ ] I can build the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-30:
Hey - this should be fixed with the latest update in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy

osaidfaisal12, 2023-10-03:
VisionCamera: Frame Processors enabled!
  ninja: error: manifest 'build.ninja' still dirty after 100 tries
  still getting in issue in 3.2.2 update

xulihang, 2023-10-24:
This can be resolved by changing the name of the project's folder. https://github.com/mrousavy/react-native-vision-camera/issues/1941

For me, I have to change `vision-camera-dynamsoft-barcode-reader` to `1`.

xulihang, 2024-02-06:
> Any updates?

Use macOS or Linux, or use a shorter path on Windows.