margelo/react-native-vision-camera #1775: 🔧

fedpre, 2023-09-06:
### How were you trying to build the app?

I tried to build my app on Android with `npx react-native run-android` and it fails.

### Full build logs

```tsx
info JS server already running.
info Installing the app...

> Configure project :react-native-firebase_app
:react-native-firebase_app package.json found at /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_app:firebase.bom using default value: 30.1.0
:react-native-firebase_app:play.play-services-auth using default value: 20.2.0
:react-native-firebase_app package.json found at /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_app:version set from package.json: 14.11.1 (14,11,1 - 14011001)
:react-native-firebase_app:android.compileSdk using custom value: 33
:react-native-firebase_app:android.targetSdk using custom value: 33
:react-native-firebase_app:android.minSdk using custom value: 26
:react-native-firebase_app:reactNativeAndroidDir /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native/android

> Configure project :react-native-firebase_crashlytics
:react-native-firebase_crashlytics package.json found at /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/@react-native-firebase/crashlytics/package.json
:react-native-firebase_app package.json found at /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/@react-native-firebase/app/package.json
:react-native-firebase_crashlytics:firebase.bom using default value: 30.1.0
:react-native-firebase_crashlytics package.json found at /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/@react-native-firebase/crashlytics/package.json
:react-native-firebase_crashlytics:version set from package.json: 14.11.1 (14,11,1 - 14011001)
:react-native-firebase_crashlytics:android.compileSdk using custom value: 33
:react-native-firebase_crashlytics:android.targetSdk using custom value: 33
:react-native-firebase_crashlytics:android.minSdk using custom value: 26
:react-native-firebase_crashlytics:reactNativeAndroidDir /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native/android

> Configure project :react-native-notifications
WARNING:DSL element 'dexOptions' is obsolete and should be removed.
It will be removed in version 8.0 of the Android Gradle plugin.
Using it has no effect, and the AndroidGradle plugin optimizes dexing automatically.

> Configure project :react-native-reanimated
No AAR for react-native-reanimated found. Attempting to build from source.
Android gradle plugin: 7.4.2
Gradle: 8.0.1
WARNING:Software Components will not be created automatically for Maven publishing from Android Gradle Plugin 8.0. To opt-in to the future behavior, set the Gradle property android.disableAutomaticComponentCreation=true in the `gradle.properties` file or use the new publishing DSL.

> Configure project :react-native-vision-camera
[VisionCamera] react-native-worklets-core found, Frame Processors enabled!

> Task :react-native-vision-camera:compileDebugKotlin
'compileDebugJavaWithJavac' task (current target is 11) and 'compileDebugKotlin' task (current target is 1.8) jvm target compatibility should be set to the same Java version.
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (7, 44): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (13, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (31, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (37, 28): 'RCTEventEmitter' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/CameraView+Events.kt: (37, 57): 'receiveEvent(Int, String!, WritableMap?): Unit' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt: (209, 9): Variable 'previewOutput' is never used
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/core/RecordingSession.kt: (49, 96): 'constructor MediaRecorder()' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraCharacteristics+getOutputSizes.kt: (46, 36): 'get(Int, Int): CamcorderProfile!' is deprecated. Deprecated in Java
w: /Users/federicopregnolato/moveshelf/home-monitoring/HomeMonitoring/node_modules/react-native-vision-camera/android/src/main/java/com/mrousavy/camera/extensions/CameraDevice+createCaptureSession.kt: (92, 12): 'createCaptureSessionByOutputConfigurations((Mutable)List<OutputConfiguration!>!, CameraCaptureSession.StateCallback!, Handler?): Unit' is deprecated. Deprecated in Java

> Task :react-native-vision-camera:configureCMakeDebug[arm64-v8a] FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.

You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.

See https://docs.gradle.org/8.0.1/userguide/command_line_interface.html#sec:command_line_warnings
530 actionable tasks: 20 executed, 510 up-to-date

info 💡 Tip: Make sure that you have set up your development environment correctly, by running react-native doctor. To read more about doctor command visit: https://github.com/react-native-community/cli/blob/main/packages/cli-doctor/README.md#doctor 
[... 209 lines cut ...]
  "engines": {
    "node": ">=16"
  }
}
```

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

- [ ] I can build the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

barghi, 2023-09-06:
version 3.0.0-rc10 fixed the problem for me

fedpre, 2023-09-06:
> version `3.0.0-rc10` fixed the problem for me

Still getting the issue. If I disable the frame processor, I am able to build the app. That's a strange error too that I couldn't find any answer online.

lfrallon, 2023-09-07:
Encountered the same issue, but it turns out that I forgot to update some configs:

Note: Take a look at this [issue](https://github.com/mrousavy/react-native-vision-camera/issues/1770#issuecomment-1707550362) first. (I used patch-package for this one)

### Forgot to update my imports in root `index.js`
```js
import 'react-native-reanimated';
import 'react-native-worklets-core/src'; // for frame processor with react-native-vision-camera@^3.0.0

import {AppRegistry} from 'react-native';
import {GestureHandlerRootView} from 'react-native-gesture-handler';
import {persistor, store} from './src/store';
import {Provider} from 'react-redux';
import {PersistGate} from 'redux-persist/integration/react';
import App from './src/App';
import {name as appName} from './app.json';

const Root = () => (
  <GestureHandlerRootView style={{flex: 1}}>
    <Provider store={store}>
      <PersistGate loading={null} persistor={persistor}>
        <App />
      </PersistGate>
    </Provider>
  </GestureHandlerRootView>
);

AppRegistry.registerComponent(appName, () => Root);
```

### Updated my `babel.config.js`
```js
module.exports = api => {
  const babelEnv = api.env();
  const plugins = [
    ['react-native-worklets-core/plugin'], // this one 
    ['react-native-reanimated/plugin'], 
    ['@babel/plugin-transform-optional-chaining'],
  ];
  //change to 'production' to check if this is working in 'development' mode
  if (babelEnv !== 'development') {
    plugins.push(['transform-remove-console', {exclude: ['error', 'warn']}]);
  }
  return {
    presets: ['module:metro-react-native-babel-preset'],
    plugins,
  };
};
```

Based on [react-native-worklets-core setup](https://github.com/margelo/react-native-worklets-core).

### Deps

```json
"react-native": "0.72.4",
"react-native-reanimated": "^3.4.2",
"react-native-vision-camera": "^3.0.0",
"react-native-worklets-core": "^0.2.0"
```

barghi, 2023-09-07:
After adding `set(ENABLE_FRAME_PROCESSORS ON)` to CMakeLists.txt I'm getting this error.
without frameProcessor prop everything works fine, I think there is an issue with react-native-worklets-core.

```
"react-native-vision-camera": "3.0.0-rc.10",
"react-native-worklets-core": "^0.2.0",
"react-native": "0.72.0",
"react-native-reanimated": "^3.4.2",
```

babel.config.js
```
module.exports = {
	presets: ['module:metro-react-native-babel-preset'],
	plugins: [
		['react-native-reanimated/plugin'],
		['react-native-worklets-core/plugin'],
	],
};
```

[image]

karri-lehtiranta, 2023-09-07:
In my experience, `set(ENABLE_FRAME_PROCESSORS ON) `actually disables the the frameprocessors instead of enabling them. This is why the build succeed as none of the worklets-core code is imported. The actual problem seemed to be that the CMake is not able to find the worklets-core library during the linking phase (at least for me the actual clang command did not include the worklets-core built library). This can be fixed by adding the direct path to the android/CMakeLists.txt file:

```
# Link everything together
target_link_libraries(
        ${PACKAGE_NAME}
        ${LOG_LIB}                          # <-- Logcat logger
        android                             # <-- Android JNI core
        ReactAndroid::jsi                   # <-- RN: JSI
        ReactAndroid::reactnativejni        # <-- RN: React Native JNI bindings
        ReactAndroid::folly_runtime         # <-- RN: For casting JSI <> Java objects
        fbjni::fbjni                        # <-- fbjni
 --->"${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/debug/obj/${ANDROID_ABI}/librnworklets.so"
)
```

I think that there might be a cleaner way to do this but I found that this will work so left it at that.
Be sure to remove the  `set(ENABLE_FRAME_PROCESSORS ON) ` if trying this.

barghi, 2023-09-07:
@karri-lehtiranta could you please send your CMakeLists.txt ?

karri-lehtiranta, 2023-09-07:
``` 
project(VisionCamera)
cmake_minimum_required(VERSION 3.9.0)

set(CMAKE_VERBOSE_MAKEFILE ON)
set(PACKAGE_NAME "VisionCamera")
set(BUILD_DIR ${CMAKE_SOURCE_DIR}/build)
set(CMAKE_VERBOSE_MAKEFILE ON)
set(CMAKE_CXX_STANDARD 17)

# Folly
include("${NODE_MODULES_DIR}/react-native/ReactAndroid/cmake-utils/folly-flags.cmake")
add_compile_options(${folly_FLAGS})

# Third party libraries (Prefabs)
find_package(ReactAndroid REQUIRED CONFIG)
find_package(fbjni REQUIRED CONFIG)
find_library(LOG_LIB log)
find_package(react-native-worklets-core REQUIRED CONFIG)

add_definitions(-DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=${ENABLE_FRAME_PROCESSORS} -DVISION_CAMERA_ENABLE_SKIA=${ENABLE_SKIA})

# Add react-native-vision-camera sources
add_library(
        ${PACKAGE_NAME}
        SHARED
        ../cpp/JSITypedArray.cpp
        src/main/cpp/FrameHostObject.cpp
        src/main/cpp/FrameProcessorPluginHostObject.cpp
        src/main/cpp/JSIJNIConversion.cpp
        src/main/cpp/VisionCamera.cpp
        src/main/cpp/VisionCameraProxy.cpp
        src/main/cpp/skia/SkiaRenderer.cpp
        src/main/cpp/java-bindings/JFrame.cpp
        src/main/cpp/java-bindings/JFrameProcessor.cpp
        src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp
        src/main/cpp/java-bindings/JHashMap.cpp
        src/main/cpp/java-bindings/JVisionCameraProxy.cpp
        src/main/cpp/java-bindings/JVisionCameraScheduler.cpp
)

# Header Search Paths (includes)
target_include_directories(
        ${PACKAGE_NAME}
        PRIVATE
        "../cpp"
        "src/main/cpp"
        "${NODE_MODULES_DIR}/react-native/ReactCommon"
        "${NODE_MODULES_DIR}/react-native/ReactCommon/callinvoker"
        "${NODE_MODULES_DIR}/react-native/ReactAndroid/src/main/jni/react/turbomodule" # <-- CallInvokerHolder JNI wrapper
)

# Link everything together
target_link_libraries(
        ${PACKAGE_NAME}
        ${LOG_LIB}                          # <-- Logcat logger
        android                             # <-- Android JNI core
        ReactAndroid::jsi                   # <-- RN: JSI
        ReactAndroid::reactnativejni        # <-- RN: React Native JNI bindings
        ReactAndroid::folly_runtime         # <-- RN: For casting JSI <> Java objects
        fbjni::fbjni                        # <-- fbjni
        "${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/debug/obj/${ANDROID_ABI}/librnworklets.so"
)

# Optionally also add Frame Processors here
if(ENABLE_FRAME_PROCESSORS)
    find_package(react-native-worklets-core REQUIRED CONFIG)
    target_link_libraries(
            ${PACKAGE_NAME}
            react-native-worklets-core::rnworklets
    )
    message("VisionCamera: Frame Processors enabled!")

    # Optionally also add Skia Integration here
    if(ENABLE_SKIA)
        set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -DSK_GL -DSK_GANESH -DSK_BUILD_FOR_ANDROID")

        find_package(shopify_react-native-skia REQUIRED CONFIG)

        set(SKIA_PACKAGE shopify_react-native-skia::rnskia)
        set(RNSKIA_PATH ${NODE_MODULES_DIR}/@shopify/react-native-skia)
        set (SKIA_LIBS_PATH "${RNSKIA_PATH}/libs/android/${ANDROID_ABI}")
        add_library(skia STATIC IMPORTED)
        set_property(TARGET skia PROPERTY IMPORTED_LOCATION "${SKIA_LIBS_PATH}/libskia.a")
        add_library(svg STATIC IMPORTED)
        set_property(TARGET svg PROPERTY IMPORTED_LOCATION "${SKIA_LIBS_PATH}/libsvg.a")
        add_library(skshaper STATIC IMPORTED)
        set_property(TARGET skshaper PROPERTY IMPORTED_LOCATION "${SKIA_LIBS_PATH}/libskshaper.a")

        # We need to include the headers from skia
        # (Note: rnskia includes all their files without any relative path
        #  so for example "include/core/SkImage.h" becomes #include "SkImage.h".
        #  That's why for the prefab of rnskia, we flatten all cpp files into
        #  just one directory. HOWEVER, skia itself uses relative paths in
        #  their include statements, and so we have to include the path to skia)
        target_include_directories(
                ${PACKAGE_NAME}
                PRIVATE
                "${RNSKIA_PATH}/cpp/skia"
                "${RNSKIA_PATH}/cpp/skia/include/config/"
                "${RNSKIA_PATH}/cpp/skia/include/core/"
                "${RNSKIA_PATH}/cpp/skia/include/effects/"
                "${RNSKIA_PATH}/cpp/skia/include/utils/"
                "${RNSKIA_PATH}/cpp/skia/include/pathops/"
                "${RNSKIA_PATH}/cpp/skia/modules/"
                # "${RNSKIA_PATH}/cpp/skia/modules/skparagraph/include/"
                "${RNSKIA_PATH}/cpp/skia/include/"
                "${RNSKIA_PATH}/cpp/skia"
        )

        target_link_libraries(
                ${PACKAGE_NAME}
                GLESv2                              # <-- Optional: OpenGL (for Skia)
                EGL                                 # <-- Optional: OpenGL (EGL) (for Skia)
                ${SKIA_PACKAGE}                     # <-- Optional: RN Skia
                jnigraphics
                skia
                svg
                skshaper
        )

        message("VisionCamera: Skia enabled!")
    endif()
endif()

```

The worklets-core import should probably be under if(ENABLE_FRAME_PROCESSORS) but didn't try that as this version worked.

barghi, 2023-09-07:
Thank you @karri-lehtiranta I will try it

barghi, 2023-09-07:
finally fixed with this line
```
"${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/debug/obj/${ANDROID_ABI}/librnworklets.so"
```

mrousavy, 2023-09-30:
Hey - this should be fixed with the latest update in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

mrousavy, 2023-09-30:
I have seen similar build errors in RN Worklets cc @chrfalch

closed 2023-09-30 by mrousavy

val089, 2023-10-04:
> finally fixed with this line
> 
> ```
> "${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/debug/obj/${ANDROID_ABI}/librnworklets.so"
> ```

It's necessery now?

mrousavy, 2023-10-04:
Wait whats wrong with that line?

I think this could be a PR to RN Worklets Core to fix it cleaning up, no?

BuiHung1612, 2023-10-04:
> > finally fixed with this line
> > ```
> > "${NODE_MODULES_DIR}/react-native-worklets-core/android/build/intermediates/cmake/debug/obj/${ANDROID_ABI}/librnworklets.so"
> > ```
> 
> It's necessery now?

I tried this and it works. But after opening the camera with frameprocessor it crashed.

Crash log is below

[image]

rodgomesc, 2023-10-04:
@BuiHung1612 this is a different issue, follow https://github.com/mrousavy/react-native-vision-camera/issues/1776

nnabinh, 2023-10-14:
@mrousavy I'm getting the same issue here. Using v3 and React Native 0.72.

```
> Task :react-native-vision-camera:compileDebugKotlin
'compileDebugJavaWithJavac' task (current target is 11) and 'compileDebugKotlin' task (current target is 1.8) jvm target compatibility should be set to the same Java version.
```

I saw you're defining `JavaVersion.VERSION_1_8` in `build.gradle`. Why is that so? I'm using latest version of Expo and some of its libraries are requiring at least version 11.

GaylordP, 2023-10-18:
Hello @nnabinh ,
I've had the same problem since the last version.
Have you found a solution? :) 

thanks

RP-alissonpaschoal, 2023-10-27:
I am facing the same error here 

```
'compileDebugJavaWithJavac' task (current target is 11) and 'compileDebugKotlin' task (current target is 1.8) jvm target compatibility should be set to the same Java version.
```

nbolender, 2023-11-09:
@RP-alissonpaschoal This worked for me: https://github.com/mrousavy/react-native-vision-camera/issues/1547#issuecomment-1787296748