margelo/react-native-vision-camera #1777: 🐛 Error: While trying to resolve module `react-native-vision-camera` for iOS

xulihang, 2023-09-06:
### What's happening?

I am trying to create a frame processor plugin for vision camera v3 using a project created with create-react-native-library.

When running the example project for iOS, I encountered the following error:

```
error: Error: While trying to resolve module `react-native-vision-camera` from file `/Users/xulihang/vision-camera-dynamsoft-barcode-reader/src/index.tsx`, the package `/Users/xulihang/vision-camera-dynamsoft-barcode-reader/node_modules/react-native-vision-camera/package.json` was successfully found. However, this package itself specifies a `main` module field that could not be resolved (`/Users/xulihang/vision-camera-dynamsoft-barcode-reader/node_modules/react-native-vision-camera/src/index`. Indeed, none of these files exist:

  * /Users/xulihang/vision-camera-dynamsoft-barcode-reader/node_modules/react-native-vision-camera/src/index(.native|.ios.js|.native.js|.js|.ios.jsx|.native.jsx|.jsx|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)
  * /Users/xulihang/vision-camera-dynamsoft-barcode-reader/node_modules/react-native-vision-camera/src/index/index(.native|.ios.js|.native.js|.js|.ios.jsx|.native.jsx|.jsx|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)
    at DependencyGraph.resolveDependency (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/node-haste/DependencyGraph.js:289:17)
    at Object.resolve (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/lib/transformHelpers.js:169:21)
    at Graph._resolveDependencies (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:473:35)
    at Graph._processModule (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:261:38)
    at async Graph._addDependency (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:372:20)
    at async Promise.all (index 6)
    at async Graph._processModule (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:322:5)
    at async Graph._addDependency (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:372:20)
    at async Promise.all (index 2)
    at async Graph._processModule (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:322:5)
    at async Graph._traverseDependenciesForSingleFile (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:249:5)
    at async Promise.all (index 0)
    at async Graph.initialTraverseDependencies (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/Graph.js:233:5)
    at async DeltaCalculator._getChangedDependencies (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/DeltaCalculator.js:229:25)
    at async DeltaCalculator.getDelta (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler/DeltaCalculator.js:112:16)
    at async DeltaBundler.buildGraph (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/DeltaBundler.js:54:5)
    at async IncrementalBundler.buildGraphForEntries (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/IncrementalBundler.js:66:19)
    at async IncrementalBundler.buildGraph (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/IncrementalBundler.js:146:19)
    at async /Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/IncrementalBundler.js:187:34
    at async IncrementalBundler.initializeGraph (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/IncrementalBundler.js:204:24)
    at async build (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/Server.js:649:35)
    at async Server.requestProcessor [as _processBundleRequest] (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/Server.js:559:18)
    at async Server._processRequest (/Users/xulihang/vision-camera-dynamsoft-barcode-reader/example/node_modules/@react-native-community/cli-plugin-metro/node_modules/metro/src/Server.js:383:7)
```

### Reproduceable Code

```tsx
https://github.com/tony-xlh/vision-camera-dynamsoft-barcode-reader/tree/ef6763bbc69e8a09dcf7167cbaa86f2cf636c1c8
```

### Device

iPhone SE iOS 15

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-06:
Hey! That's weird, maybe the `.tsconfig` doesn't create the relative import properly?

mendesbarreto, 2023-09-15:
I have the latest version of the project working [here](https://github.com/mendesbarreto/ml-mobile-camera-template/tree/master), maybe you can compare the files.

closed 2023-09-30 by mrousavy