margelo/react-native-vision-camera #1779: An old version of CMake is being used that cannot automatically detect  compiler attributes🔧

PieterUyz, 2023-09-06:
### How were you trying to build the app?

I tried running the app with [nx](https://nx.dev/) but it utilizes yarn to start the app

### Full build logs

```tsx
CMake Warning at CMakeLists.txt:250 (message):
    VisionCamera linking: FOR_HERMES=true

  CMake Warning at C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
    An old version of CMake is being used that cannot automatically detect
    compiler attributes.  Compiler identification is being bypassed.  Some
    values may be wrong or missing.  Update to CMake 3.19 or newer to use
    CMake's built-in compiler identification.
  Call Stack (most recent call first):
    C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
    .cxx/Debug/5x615p1k/arm64-v8a/CMakeFiles/3.18.1-g262b901-dirty/CMakeSystem.cmake:6 (include)
    CMakeLists.txt:1 (project)

  CMake Warning at CMakeLists.txt:250 (message):
    VisionCamera linking: FOR_HERMES=true

  CMake Warning at C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
    An old version of CMake is being used that cannot automatically detect
    compiler attributes.  Compiler identification is being bypassed.  Some
    values may be wrong or missing.  Update to CMake 3.19 or newer to use
    CMake's built-in compiler identification.
  Call Stack (most recent call first):
    C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
    .cxx/Debug/5x615p1k/arm64-v8a/CMakeFiles/3.18.1-g262b901-dirty/CMakeSystem.cmake:6 (include)
    CMakeLists.txt:1 (project)

  CMake Warning at CMakeLists.txt:250 (message):
    VisionCamera linking: FOR_HERMES=true

  CMake Warning at C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
    An old version of CMake is being used that cannot automatically detect
    compiler attributes.  Compiler identification is being bypassed.  Some
    values may be wrong or missing.  Update to CMake 3.19 or newer to use
    CMake's built-in compiler identification.
  Call Stack (most recent call first):
    C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
    .cxx/Debug/5x615p1k/arm64-v8a/CMakeFiles/3.18.1-g262b901-dirty/CMakeSystem.cmake:6 (include)
    CMakeLists.txt:1 (project)

  CMake Warning at CMakeLists.txt:250 (message):
    VisionCamera linking: FOR_HERMES=true

  CMake Warning at C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
    An old version of CMake is being used that cannot automatically detect
    compiler attributes.  Compiler identification is being bypassed.  Some
    values may be wrong or missing.  Update to CMake 3.19 or newer to use
    CMake's built-in compiler identification.
  Call Stack (most recent call first):
    C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
    .cxx/Debug/5x615p1k/arm64-v8a/CMakeFiles/3.18.1-g262b901-dirty/CMakeSystem.cmake:6 (include)
    CMakeLists.txt:1 (project)

  CMake Warning at CMakeLists.txt:250 (message):
    VisionCamera linking: FOR_HERMES=true

  CMake Warning at C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android-legacy.toolchain.cmake:416 (message):
    An old version of CMake is being used that cannot automatically detect
    compiler attributes.  Compiler identification is being bypassed.  Some
    values may be wrong or missing.  Update to CMake 3.19 or newer to use
    CMake's built-in compiler identification.
  Call Stack (most recent call first):
    C:/Users/username/AppData/Local/Android/Sdk/ndk/23.1.7779620/build/cmake/android.toolchain.cmake:55 (include)
    .cxx/Debug/5x615p1k/arm64-v8a/CMakeFiles/3.18.1-g262b901-dirty/CMakeSystem.cmake:6 (include)
[... 623 lines cut ...]
    "util": "^0.12.5",
    "uuidv4": "^6.2.13",
    "vision-camera-code-scanner": "^0.2.0",
    "zod": "^3.21.4"
```

### Target platforms

iOS, Android

### Operating system

Windows

### Can you build the VisionCamera Example app?

- [ ] I can build the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

PieterUyz, 2023-09-06:
If I upgrade to 3.0.0-rc10 or 3.0.0 I get the following

BUILD FAILED in 39s

error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081
node_modules\vision-camera-code-scanner\android\src\main\java\com\visioncameracodescanner\VisionCameraCodeScannerPlugin.java:40: error: VisionCameraCodeScannerPlugin is not abstract and does not override abstract method callback(Frame,Map<String,Object>) in FrameProcessorPlugin
public class VisionCameraCodeScannerPlugin extends FrameProcessorPlugin {
       ^
node_modules\vision-camera-code-scanner\android\src\main\java\com\visioncameracodescanner\VisionCameraCodeScannerPlugin.java:62: error: method does not override or implement a method from a supertype
  @Override
  ^
node_modules\vision-camera-code-scanner\android\src\main\java\com\visioncameracodescanner\VisionCameraCodeScannerPlugin.java:251: error: constructor FrameProcessorPlugin in class FrameProcessorPlugin cannot be applied to given types;
    super("scanCodes");
    ^
  required: no arguments
  found: String
  reason: actual and formal argument lists differ in length
node_modules\vision-camera-code-scanner\android\src\main\java\com\visioncameracodescanner\VisionCameraCodeScannerPluginPackage.java:19: error: cannot find symbol
    FrameProcessorPlugin.register(new VisionCameraCodeScannerPlugin());
                        ^
  symbol:   method register(VisionCameraCodeScannerPlugin)
  location: class FrameProcessorPlugin
4 errors

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':vision-camera-code-scanner:compileDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

2: Task failed with an exception.
-----------
* What went wrong:
java.lang.StackOverflowError (no error message)

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 39s

    at makeError (C:\Users\piete\Documents\OpenVantage Projects\City-ev\node_modules\execa\index.js:174:9)
    at node_modules\execa\index.js:278:16
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async runOnAllDevices (node_modules\react-native\node_modules\@react-native-community\cli-platform-android\build\commands\runAndroid\runOnAllDevices.js:82:7)
    at async Command.handleAction (node_modules\react-native\node_modules\@react-native-community\cli\build\index.js:108:9)
info Run CLI with --verbose flag for more details.

mrousavy, 2023-09-06:
You need to upgrade your CMake version to the same one I use in VisionCamera example (here)

closed 2023-09-06 by mrousavy

PieterUyz, 2023-09-06:
I have 3.18 and the latest installed.

[image]

And If I uninstall 3.18 your react-native-vision-camera re-installs it

sabarinath-m, 2023-09-08:
Getting the same issue here. I have both CMake 3.22.1 and 3.18.1

TiagoDanin, 2023-09-26:
> If I upgrade to 3.0.0-rc10 or 3.0.0 I get the following

[image]
[image]

Getting the same issue here. I have CMake 3.22.1.
"react-native": "0.72.5"
"react-native-vision-camera": "^3.1.0"

indranilGghosh, 2023-09-28:
Any solution for this getting the same error

learncourse555, 2023-10-16:
> I have 3.18 and the latest installed.
> 
> [image]
> 
> And If I uninstall 3.18 your react-native-vision-camera re-installs it @mrousavy

getting Same Issue....Any Solution for resolving this ?

arsimmsejdiu, 2023-10-24:
Same here, @mrousavy is the library still using old versions of CMake and you need to update the library so that it uses newer verons ? or is it independent of that, because this error seems still unsolvable

prithvisharma, 2023-10-27:
Even facing this CMAKE issue, even thought higher version is installed it forces 3.18 and reinstall if not present

Spxc, 2023-11-15:
Hi @mrousavy, seems like you forgot the link here:
> You need to upgrade your CMake version to the same one I use in VisionCamera example (here)

So was there any solution for this? (i see it's closed) Still having some `CMAKE` build issue on android. Followed the troubleshooter and everything. Latest `3.22` installed from Android Studio.

Using the following:
`react-native-vision-camera: 3.6.6`
`react-native: 0.69.12`

It builds, runs and installs fine on iOS 17.1, however i'm not able to get it running on Android.

Error from `./gradlew assembleRelease`:
```
> Task :react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a] FAILED
C/C++: ninja: Entering directory `/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/205o2y4k/arm64-v8a'
C/C++: In file included from ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:5:
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
C/C++: In file included from ../../../../src/main/cpp/VisionCamera.cpp:2:
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
C/C++: In file included from ../../../../src/main/cpp/frameprocessor/VisionCameraProxy.cpp:5:
C/C++: In file included from ../../../../src/main/cpp/frameprocessor/VisionCameraProxy.h:9:
C/C++: ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
w: Detected multiple Kotlin daemon sessions at build/kotlin/sessions

FAILURE: Build completed with 2 failures.
```

Furthermore:
```
1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeRelWithDebInfo[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native-vision-camera/android/.cxx/RelWithDebInfo/205o2y4k/arm64-v8a'
  [1/4] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o 
  /Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --gcc-toolchain=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=false -DVisionCamera_EXPORTS -I../../../../../cpp -I../../../../src/main/cpp -I../../../../src/main/cpp/frameprocessor -I../../../../src/main/cpp/frameprocessor/java-bindings -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactCommon/callinvoker -I/Users/stianwiikinstebo/Documents/Backup/YourProjectNameBak-RN096/node_modules/react-native/ReactAndroid/src/main/jni/react/turbomodule -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/jsi/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/44f87529cbf9f777306931dc59c5d740/transformed/jetified-react-android-0.73.0-rc.4-release/prefab/modules/reactnativejni/include -isystem /Users/stianwiikinstebo/.gradle/caches/transforms-3/c7598e7d957f8ec4dcaab85973495b27/transformed/jetified-fbjni-0.5.1/prefab/modules/fbjni/include -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -O2 -g -DNDEBUG -fPIC -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp.o -c ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp
  In file included from ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.cpp:5:
  ../../../../src/main/cpp/frameprocessor/java-bindings/JVisionCameraProxy.h:7:10: fatal error: 'ReactCommon/CallInvokerHolder.h' file not found
  #include <ReactCommon/CallInvokerHolder.h>
           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  1 error generated.
  [2/4] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  FAILED: CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o 
  /Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++ --target=aarch64-none-linux-android26 --gcc-toolchain=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/Users/stianwiikinstebo/Library/Android/sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -DVISION_CAMERA_ENABLE_FRAME_PROCESSORS=false -DVis [... cut ...] sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O2 -frtti -fexceptions -Wall -Wno-unused-variable -fstack-protector-all -O2 -g -DNDEBUG -fPIC -std=gnu++17 -MD -MT CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -MF CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o.d -o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o -c ../../../../src/main/cpp/VisionCamera.cpp
```

Elijah699, 2024-05-04:
@mrousavy this issue still persists on RN 0.71.4. I have a C Make version of 3.22.1. 

How can i get this resolved? @mrousavy

mrousavy, 2024-05-04:
please stop tagging me - I don't know, when you just give me a version and no reproduction I cannot help you.
The example app builds fine, try comparing your versions with that.

syaiful46, 2024-05-28:
@mrousavy please helpppp