margelo/react-native-vision-camera #1781: 🐛 Cannot convert Frame with Format 34 to byte array!

bankbiz, 2023-09-07:
### What's happening?

I tried to get an image ArrayBuffer from frameProcessor but did not succeed. I got the error: `Cannot convert Frame with Format 34 to byte array! `

### Reproduceable Code

```tsx
const devices = useCameraDevices();

const frameProcessor = useFrameProcessor((frame) => {
   'worklet'
   let byteImage = frame.toArrayBuffer();
}, []);

useEffect(() => {
   if (devices.back !== undefined) {
      setCameraDevice(devices.back);
   }
}, [devices]);

...
<Camera
   device={cameraDevice}
   isActive={true}
   frameProcessor={frameProcessor}
/>
```

### Device

Xiaomi 11

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

barghi, 2023-09-07:
Same issue here

jjbwii, 2023-09-09:
Same issue here. Additional info:

- When I set ‘yuv’ (format:34) my program crash
- when I set ‘rgb’ program works but GetPlanes() allways null, it seems the buffer is not available
- when I set ‘native’ it works and obvious there is no buffer available (private format)

I think there is any problem with VideoPipeline.kt Surface tranform to Image

jlsneto, 2023-09-12:
> Same issue here. Additional info:
>
> * When I set ‘yuv’ (format:34) my program crash
> * when I set ‘rgb’ program works but GetPlanes() allways null, it seems the buffer is not available
> * when I set [...]

Same issue.

sarahsporck, 2023-09-13:
format 34 should be the [private](https://developer.android.com/reference/android/graphics/ImageFormat#PRIVATE) format. I guess that's also the format the emulator uses?

Waseem6409, 2023-09-14:
@mrousavy  Can you check the issue , i am working on something but stuck due to this and also because the tflite and playtorch functionalities like imageFromFrame is not yet implemented

mrousavy, 2023-09-14:
@Waseem6409 sure, contact us through my app development/consulting agency https://margelo.io and we can figure something out for a fair $$$.

oliverphardman, 2023-09-16:
Same issue here with Expo.

oliverphardman, 2023-09-17:
> Same issue here. Additional info:
>
>     * When I set ‘yuv’ (format:34) my program crash
>
>     * when I set ‘rgb’ program works but GetPlanes() allways null, it seems the buffer is not available
>
>     * [...]

`rgb` crashes my app.

basharmalkawi, 2023-09-25:
Same issue.

rodgomesc, 2023-09-27:
@oliverphardman  please check if #1875 solves your crash while using `rgb`

mrousavy, 2023-09-30:
Hey - this should be fixed in V3.2.0 as I now use the GPU HardwareBuffer APIs in YUV_420!

closed 2023-09-30 by mrousavy