margelo/react-native-vision-camera #1787: 🐛 No implementation found for com.facebook.jni.HybridData method when video prop set to true

Phillupz, 2023-09-10:
### What's happening?

I'm encountering an issue when using the react-native-vision-camera library. Specifically, the error arises when the video prop in the Camera component is set to true. The screen turns black, and an error is thrown indicating no implementation found for a specific method. The photo functionality appears to be working correctly when its prop is set to true.

### Reproduceable Code

```tsx
import React from 'react'
import { Camera, CameraDevice } from 'react-native-vision-camera';
import { StyleSheet, View } from 'react-native';

interface CameraCompProps {
  device: CameraDevice | undefined;
  setDevice: (device: CameraDevice) => void;
  cameraRef: any;
  activate: boolean;
}

export const CameraComp: React.FC<CameraCompProps> = ({ cameraRef, device, activate }) => {

  return (
    <View style={[styles.container]}>
      {activate && device && ( 
        <Camera
          ref={cameraRef}
          style={styles.camera}
          device={device}
          photo={true}
          video={true}
          enableZoomGesture
          isActive={activate}
          orientation={'portrait'}
          onInitialized={() => {
            console.log('initialized')
          }}
          onError={(error) => {console.log(error)}}
        />
      )}
    </View>
  );
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    height: '100%'
  },
  camera: {
    flexGrow: 1,
    height: '100%'
  },

})
```

### Relevant log output

```shell
LOG  [unknown/unknown: [unknown/unknown] No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)
```

### Camera Device

```shell
{
   "pixelFormats": [
    "rgb",
    "native",
    "yuv",
    "unknown"
   ],
   "videoStabilizationModes": [
    "off",
    "standard",
    "off"
   ],
   "autoFocusSystem": "contrast-detection",
   "supportsPhotoHDR": false,
   "photoHeight": 144,
   "supportsVideoHDR": false,
   "maxISO": 1550,
   "minISO": 100,
   "minFps": 1,
   "videoWidth": 176,
   "videoHeight": 144,
   "photoWidth": 176,
   "fieldOfView": 67.80954434399631,
   "maxFps": 30
  }
 ],
 "hardwareLevel": "level-3",
 "pixelFormats": [
  "RAW_SENSOR",
  "JPEG",
  "PRIVATE",
  "YUV_420_888",
  "RAW_PRIVATE",
  "RAW10",
  "HEIC"
 ],
 "maxZoom": 6,
 "minZoom": 1,
 "supportsLowLightBoost": false,
 "sensorOrientation": "landscape-left",
 "supportsDepthCapture": false,
 "neutralZoom": 1,
 "supportsFocus": true,
 "supportsRawCapture": true,
 "isMultiCam": false,
 "hasTorch": false,
 "devices": [
  "wide-angle-camera"
 ],
 "hasFlash": false,
 "name": "FRONT (1)",
 "position": "front",
 "id": "1"
}
```

### Device

OnePlus Nord n20 5g Android Version 12

### VisionCamera Version

"react-native-vision-camera": "^3.0.0",

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [X] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-11:
This looks like you either didn't build the native part, or the library didn't load at this point - I'll take a look.

Btw.; this might also fix your issue: https://github.com/mrousavy/react-native-vision-camera/pull/1789/files

Romick2005, 2023-09-11:
> This looks like you either didn't build the native part, or the library didn't load at this point - I'll take a look.
> 
> Btw.; this might also fix your issue: #1789

Yes that solve the issue with ability to show video stream, but when when I try to stop recording I got this error on Android, on iOS it work prefect! :
[image]

mrousavy, 2023-09-11:
Please don't attach screenshots, use formatted codeblocks instead. This looks like a separate issue, so open a separate issue here please.

closed 2023-09-11 by mrousavy

sriumalladi, 2023-09-15:
[unknown/unknown: [unknown/unknown] No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)]
 LOG  Re-rendering camera page with inactive camera. Device: "FRONT (1)" (6528x4896 photo / 1920x1080 video @ 30fps)