margelo/react-native-vision-camera #1790: 🐛[v3][iOS] withArguments:Arguments in the FrameProcessorPlugin callback function is always nil

ArturKalach, 2023-09-11:
### What's happening?

Using example I tried to write and use frameProcessor
https://www.react-native-vision-camera.com/docs/guides/frame-processors-plugins-ios

iOS:
```
- (id)callback:(Frame*)frame withArguments:(NSDictionary*)arguments {
  CMSampleBufferRef buffer = frame.buffer;
  UIImageOrientation orientation = frame.orientation;
  // code goes here
  return @[];
}
```

TS:
```
const plugin = VisionCameraProxy.getFrameProcessorPlugin('__scanCode');
export function examplePlugin(frame: Frame): string[] {
  'worklet';

  if (plugin == null) {
    throw new Error('Failed to load Frame Processor Plugin "example_plugin"!');
  }

  return plugin.call(frame, {
    someString: 'hello!',
    someBoolean: true,
    someNumber: 42,
    someObject: {test: 0, second: 'test'},
    someArray: ['another test', 5],
  }) as string[];
}

...
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';

    console.log(
      `${frame.timestamp}: ${frame.width}x${frame.height} ${frame.pixelFormat} Frame (${frame.orientation})`,
    );
    examplePlugin(frame);
  }, []);
```

The callback fn is called but arguments object is always `nil`.
Same for this plugin: https://github.com/rkmackinnon/vision-camera-code-scanner/tree/react-native-vision-camera-v3

```
 return plugin.call(frame, {
    ...options,
    types,
  });
 ```

But: 
[image]

 
 
 
 

### Reproduceable Code

```tsx
You could check this library:
https://github.com/rkmackinnon/vision-camera-code-scanner/blob/react-native-vision-camera-v3

https://github.com/rkmackinnon/vision-camera-code-scanner/blob/react-native-vision-camera-v3/src/index.ts
[... 1162 lines cut ...]
  ],
  "isMultiCam": false,
  "supportsLowLightBoost": false
}
```

### Device

iPhone 8 (16.5.1)

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

ObidosDev, 2023-09-14:
Same for me. Also working on patch for qr code scanner :D, but `arguments` is always `nil` with no reason

ArturKalach, 2023-09-15:
I have seen this patch, I guess it should help to fix the issue:
https://github.com/rodgomesc/vision-camera-code-scanner/blob/f2933a7af88affaa9b5f874a2076ec26bd1d3a7c/.yarn/patches/react-native-vision-camera-npm-3.0.0-21b4ab30d7.patch

ObidosDev, 2023-09-18:
@ArturKalach did you try this?

ObidosDev, 2023-09-18:
https://github.com/mrousavy/react-native-vision-camera/pull/1750
I think it's related and hasn't been released yet

ArturKalach, 2023-09-19:
I guess this issue is not relevant anymore. Let's wait for release

closed 2023-09-19 by ArturKalach