margelo/react-native-vision-camera #1793: 🐛  [Android] - Camera does not work due to profile retreiving error

idrissakhi, 2023-09-12:
### What's happening?

Opening the camera after giving permissions fails on android.

### Reproduceable Code

```tsx
import { ReactElement, useCallback, useEffect, useState } from 'react';
import { ImageRequireSource, Pressable, View, ViewStyle } from 'react-native';

import { Gesture, GestureDetector } from 'react-native-gesture-handler';
import Animated, {
  Extrapolate,
  interpolate,
  useAnimatedProps,
  useAnimatedRef,
  useAnimatedStyle,
  useSharedValue,
  withSpring,
  withTiming,
} from 'react-native-reanimated';

import { useSafeAreaInsets } from 'react-native-safe-area-context';
import {
  Camera,
  CameraDevice,
  CameraPosition,
  CameraProps,
  PhotoFile,
  TakePhotoOptions,
} from 'react-native-vision-camera';

import { filePrefix, isAndroid } from 'src/Shared/Constants';
import { useHapticEffect, usePrevious } from 'src/Shared/Hooks/Common';
import { HapticType } from 'src/Shared/NativeModules';
import { Color } from 'src/Shared/Types';
import { IconButton } from 'src/Shared/UI/Atoms';
import { FullScreenLoader } from 'src/Shared/UI/Molecules';

import { styles } from './CameraView.styles';

export interface CameraViewProps {
  onTakePhoto: (photo: PhotoFile) => void;
  testID?: string;
}

const flashIcon = new Map<TakePhotoOptions['flash'], ImageRequireSource>([
  ['auto', require('assets/icons/camera-flash-auto-icon/camera-flash-auto-icon.png')],
  ['on', require('assets/icons/camera-flash-on-icon/camera-flash-on-icon.png')],
  ['off', require('assets/icons/camera-flash-off-icon/camera-flash-off-icon.png')],
]);

/**
 * Returns next value for flash mode based on current, passed as argument
 *
 * @param flashMode - Current flash mode
 */
const getNextFlashMode = (flashMode: TakePhotoOptions['flash']): TakePhotoOptions['flash'] => {
  const flashModesOrder = Array.from(flashIcon.keys());
  const currentIndex = flashModesOrder.findIndex((mode) => mode === flashMode);
  if (currentIndex > -1) {
    return flashModesOrder[currentIndex >= flashModesOrder.length - 1 ? 0 : currentIndex + 1];
  }
  return flashModesOrder[0];
};

/**
 * Returns icon of flash mode
 *
 * @param flashMode - Flash mode to get icon for
[... 210 lines cut ...]
### Camera Device

```shell
[unknown/unknown: [unknown/unknown] Error retrieving camcorder profile params]
```

### Device

Samsung S20

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

valn1, 2023-09-28:
im also having this issue in my samsung s20 +
vision camera v3 and bare minimum camera setup
```

const Scanner = (props: NavigationProps) => {
    const [device, setDevice] = useState(undefined as CameraDevice | undefined);
    const { hasPermission, requestPermission } = useCameraPermission();

    useEffect(() => {
        if (!hasPermission) {
            requestPermission().then(console.log).catch(console.warn);
            const devices = Camera.getAvailableCameraDevices(); //the error uccurs here(Error: Exception in HostObject::get for prop 'CameraDevices': java.lang.RuntimeException: Error retrieving camcorder profile params, js engine: hermes)
            setDevice(devices.find((d) => d.position === "back"));
        }
    }, []);

    return (
        <Screen>
            <Text>qrcode Scanner</Text>
            {hasPermission && device && <Camera style={StyleSheet.absoluteFill} device={device} isActive={true} />}
        </Screen>
    );
};

export default Scanner;

```

valn1, 2023-09-28:
i also got this:
```
09-28 12:16:39.955 17411 17411 E Berry_Drawer: getDrawerListName() : Exception : java.lang.NullPointerException: Attempt to read from field 'java.lang.String com.samsung.accessory.hearablemgr.common.uhm.DeviceRegistryData.deviceId' on a null object reference in method 'java.lang.String com.samsung.accessory.hearablemgr.module.home.drawer.Drawer.getDrawerListName(com.samsung.accessory.hearablemgr.common.uhm.DeviceRegistryData, java.util.List)'

```

valn1, 2023-09-28:
i found out that in the file "node_modules/react-native-vision-camera/src/CameraDevices.ts", NativeModules returns {} for some reason

mrousavy, 2023-09-30:
Hey - this should be fixed with the latest update in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

valn1, 2023-10-02:
the error is still happening after updating the lib.
any advice on how i can debug that and help fix this issue?

valn1, 2023-10-02:
I tested in a Samsung s20 FE with nova launcher instead of oneui and it worked fine(only fails to switch back to back camera).
in my Samsung s20 plus, it didnt fails to get the camera device

mrousavy, 2023-10-02:
That exact error? I am no longer using Camcorder, this sounds impossible...

Could you maybe create a new issue and paste the new logs you get in there?
Thanks!

NiravSa, 2023-10-03:
I was having the same error on the latest version. not able to load the camera device. I also got a runtime error when trying to load a camera

mrousavy, 2023-10-03:
Ahh, my bad! I oversaw this. Check this PR out and let me know if that works for you: https://github.com/mrousavy/react-native-vision-camera/pull/1907

closed 2023-10-03 by mrousavy