margelo/react-native-vision-camera #1794: 🐛 If you use the frame processor, the app will automatically shut down

dongseobb, 2023-09-12:
### What's happening?

I added frame processor to implement vision-camera-ocr.

However, when building, the app shuts down at the same time when you operate the camera even though you saw the phrase "frame processor" is available.

Exits without outputting any error logs.

What should I do?

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
  'worklet'
  console.log("log test")
}, [])

...

return(
  ...
  <Camera
    ref={camera}
    style={{
      height: '100%',
      width: windowWidth,
    }}
    device={device}
    isActive={showCamera}
    photo={true}
    fps={240}
    frameProcessor={frameProcessor}
  />
  ...
)
```

### Device

IPhone XS (ios 16.6.1)

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fedu44, 2023-09-12:
[image]

mrousavy, 2023-09-14:
> Exits without outputting any error logs.

This is wrong. The issue template tells you how to read logs, read the issue template before opening an issue.

Open adb logcat logs and post them here, otherwise I can't help you.

mrousavy, 2023-09-30:
Should be fixedi n latest v3

closed 2023-09-30 by mrousavy