margelo/react-native-vision-camera #1795: 🐛

alsdbtjd0103, 2023-09-12:
### What's happening?

I used takePhoto() for Camera. In the case of the back camera, when orientation='portrait' is set, it is taken and stored normally, but even if the portrait is set, the front camera is rotated 90 degrees to the left and stored. (anti-clockwise)

### Reproduceable Code

```tsx
const photo = await cameraRef.current?.takePhoto({
      enableShutterSound: false,
    });

          <View className="w-screen h-[500px] rounded-lg overflow-hidden mb-10">
            <Camera
              ref={cameraRef}
              device={device}
              isActive={active}
              className=" w-full h-full"
              photo={true}
              style={{ flex: 1 }}
              orientation={'portrait'}
            />
          </View>
```

### Relevant log output

```shell
when I used Front: 
photo {"height": 2460, "isMirrored": true, "isRawPhoto": false, "orientation": "landscape-left", "path": "/data/user/0/com.sleepy.mollu/cache/mrousavy854540589847414516.jpg", "width": 3280}

when I used Back:
photo {"height": 3492, "isMirrored": false, "isRawPhoto": false, "orientation": "landscape-right", "path": "/data/user/0/com.sleepy.mollu/cache/mrousavy1616464148184895130.jpg", "width": 4656}
```

### Camera Device

```shell
"formats": [
    {
      "pixelFormats": [
        "rgb",
        "native",
        "yuv"
      ],
      "videoStabilizationModes": [
        "off",
        "standard",
        "off"
      ],
      "autoFocusSystem": "contrast-detection",
      "supportsPhotoHDR": false,
      "photoHeight": 2460,
      "supportsVideoHDR": false,
      "maxISO": 800,
      "minISO": 50,
      "minFps": 1,
      "videoWidth": 1920,
      "videoHeight": 1440,
      "photoWidth": 3280,
      "fieldOfView": 68.83830743678364,
      "maxFps": 30
    },
```

### Device

Android

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

fedu44, 2023-09-12:
+1 using react-native-reanimated same issue

fedu44, 2023-09-12:
With library react-native-reanimated and orientation="portrait" that stop happening

alsdbtjd0103, 2023-09-13:
> With library react-native-reanimated and orientation="portrait" that stop happening

How did you fix it??

I currently used imageResizer to rotate image, but I can't assure that this situation always happen in all of devices.

mrousavy, 2023-09-14:
Yea I think orientation is still a bit weird on some devices. Unfortunately there's no clear solution provided by the Android system so I have to do a bunch of workarounds. I just don't have the devices here to test it on

mrousavy, 2023-09-30:
Hey - orientation is quite tricky to implement, if I get enough sponsors I can take a look at it but there's a ton of edge cases. JFYI; Camera sensors are always 90deg oriented.

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Hey - I'm tracking Orientation in this feature request/issue now: https://github.com/mrousavy/react-native-vision-camera/issues/1891 ✨