margelo/react-native-vision-camera #1796: 🐛 Switching camera in the example app crashes

szmarczak, 2023-09-12:
### What's happening?

Black screen after pressing the "switch camera" button in the top right.

### Reproduceable Code

```tsx
`packages/example`
```

### Relevant log output

```shell
09-12 16:43:22.865 12498 12536 I ReactNativeJS: Re-rendering camera page with active camera. Device: "BACK (0)" (4000x2250 photo / 3840x2160 video @ 30fps)
09-12 16:43:22.913 12498 12558 I CameraSession: Camera became available: 0
09-12 16:43:22.913 12498 12558 I CameraSession: Camera became available: 1
09-12 16:43:22.913 12498 12558 I CameraSession: Camera became available: 2
09-12 16:43:22.913 12498 12558 I CameraSession: Camera became available: 3
09-12 16:43:22.916 12498 12498 I CameraView: Props changed: [enableFrameProcessor, audio, cameraId, photo, orientation, video, isActive, fps, format, lowLightBoost, hdr]
09-12 16:43:22.932 12498 12498 I PreviewSize: Phone has a 1080 x 1920 screen.
09-12 16:43:22.935 12498 12498 I PreviewView: Using Preview Size 1920 x 1080.
09-12 16:43:22.936 12498 12498 D PreviewView: onMeasure(0, 0)
09-12 16:43:22.938 12498 12498 D PreviewView: Measured dimensions set: 0 x 0
09-12 16:43:22.938 12498 12498 I CameraView: Configuring Camera Device...
09-12 16:43:22.940 12498 12498 I CameraSession: Setting Format (fps: 30 | videoStabilization: null | hdr: false | lowLightBoost: false)...
09-12 16:43:22.941 12498 12498 I CameraSession: Setting isActive: false (isRunning: false)
09-12 16:43:22.965 12498 12498 D CompatibilityChangeReporter: Compat change id reported: 210923482; UID 10002; state: ENABLED
09-12 16:43:23.012 12498 12498 I CameraSession: Setting isActive: true (isRunning: false)
09-12 16:43:23.017 12498 12498 D PreviewView: onMeasure(1080, 1920)
09-12 16:43:23.018 12498 12498 D PreviewView: Measured dimensions set: 1080 x 1920
09-12 16:43:23.023  1062  1201 I ThermalEngine: handle_thresh_sig: SS Id SS-CLUSTER1-SP0, Read xo_therm 39000mC
09-12 16:43:23.041 12498 12498 I PreviewView: Surface created! Surface(name=null)/@0x14a5a9f
09-12 16:43:23.041 12498 12498 I CameraView: Configuring Camera Device...
09-12 16:43:23.042 12498 12498 I CameraSession: Configuring Session for Camera 0...
09-12 16:43:23.044 12498 12498 I CameraOutputs: Preparing Outputs for Camera 0...
09-12 16:43:23.044 12498 12498 I CameraOutputs: Adding native preview view output.
09-12 16:43:23.049 12498 12498 I PreviewSize: Phone has a 1080 x 1920 screen.
09-12 16:43:23.056 12498 12498 I CameraOutputs: Adding 4000x2250 photo output. (Format: 256)
09-12 16:43:23.062 12498 12498 I CameraOutputs: Adding 3840x2160 video output. (Format: 34)
09-12 16:43:23.062 12498 12498 I CameraOutputs: Prepared 3 Outputs for Camera 0!
09-12 16:43:23.063 12498 12498 I PreviewView: Surface resized! Surface(name=null)/@0x14a5a9f (1920 x 1080 in format #4)
09-12 16:43:23.063 12498 12558 I CameraSession: Starting Camera Session...
09-12 16:43:23.072 12498 12558 I CameraManager: Camera 0: Opening...
09-12 16:43:23.078  6742  6805 I CameraService: CameraService::connect call (PID 12498 "com.mrousavy.camera.example", camera ID 0) and Camera API version 2
09-12 16:43:23.079  6742  6805 I Camera2ClientBase: Camera 0: Opened. Client: com.mrousavy.camera.example (PID 12498, UID 10002)
09-12 16:43:23.079  6742  6805 I CameraDeviceClient: CameraDeviceClient 0: Opened
09-12 16:43:23.079  6742  6805 I CameraService: makeClient: Camera2 API, override to portrait 0
09-12 16:43:23.081 12498 12498 D VisionCameraProxy: Finding view 45...
09-12 16:43:23.082 12498 12498 D VisionCameraProxy: Found view 45!
09-12 16:43:23.083  6742  6805 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
09-12 16:43:23.084   876  6255 I QCamera : <HAL><INFO> cameraDeviceOpen: 407: Open camera id 0 API version 768
09-12 16:43:23.088   876  6255 I QCamera : <HAL><INFO> QCamera3HardwareInterface: 530: AV timer enabled: 0
09-12 16:43:23.092   876  6255 I QCamera : <HAL><INFO> openCamera: 804: [KPI Perf]: E PROFILE_OPEN_CAMERA camera id 0
09-12 16:43:23.095  6742  6805 W CameraProviderManager: Camera provider legacy/0 says an unknown camera device@1.0/legacy/0 now has torch status 0. Curious.
09-12 16:43:23.095  6742  6805 I CameraProviderManager: Camera device device@3.3/legacy/0 torch status is now NOT_AVAILABLE
09-12 16:43:23.095  6742  6805 I CameraService: onTorchStatusChangedLocked: Torch status changed for cameraId=0, newStatus=0
09-12 16:43:23.096     0     0 I msm_pm_qos_update_request: update request 100
09-12 16:43:23.096     0     0 I msm_pm_qos_add_request: add requestmsm_pm_qos_update_request: update request -1
09-12 16:43:23.096     0     0 I msm_pm_qos_add_request: add request
09-12 16:43:23.096   876  6255 D QCamera : Debug log file is not enabled
09-12 16:43:23.096   905   925 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [2, 1]
09-12 16:43:23.096   905   925 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [2, 0]
09-12 16:43:23.096   905   925 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [2, 1]
09-12 16:43:23.096   905   925 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [2, 0]
09-12 16:43:23.096   905   925 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [2, 1]
09-12 16:43:23.096   905   925 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [2, 0]
09-12 16:43:23.102     0     0 I msm_csid_init: CSID_VERSION = 0x30050000
09-12 16:43:23.104     0     0 E         : msm_csid_irq CSID0_IRQ_STATUS_ADDR = 0x800
09-12 16:43:23.108     0     0 I MSM-CPP cpp_init_hardware: 1145 CPP HW Version: 0x60010000
09-12 16:43:23.108     0     0 E MSM-CPP cpp_init_hardware: 1163 stream_cnt:0
[... 423 lines cut ...]
### Camera Device

```shell
will provide this shortly
```

### Device

Xiaomi Mi 6

### VisionCamera Version

main

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

szmarczak, 2023-09-12:
Also got this crash:

```
09-12 17:09:08.153 16957 16957 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
09-12 17:09:08.153 16957 16957 F DEBUG   : LineageOS Version: '20.0-20230831-NIGHTLY-sagit'
09-12 17:09:08.153 16957 16957 F DEBUG   : Build fingerprint: 'Xiaomi/sagit/sagit:8.0.0/OPR1.170623.027/V9.2.3.0.OCAMIEK:user/release-keys'
09-12 17:09:08.153 16957 16957 F DEBUG   : Revision: '0'
09-12 17:09:08.153 16957 16957 F DEBUG   : ABI: 'arm64'
09-12 17:09:08.153 16957 16957 F DEBUG   : Timestamp: 2023-09-12 17:09:05.435281304+0200
09-12 17:09:08.153 16957 16957 F DEBUG   : Process uptime: 1026s
09-12 17:09:08.153 16957 16957 F DEBUG   : Cmdline: /system/bin/cameraserver
09-12 17:09:08.154 16957 16957 F DEBUG   : pid: 13762, tid: 16938, name: CameraServiceWa  >>> /system/bin/cameraserver <<<
09-12 17:09:08.154 16957 16957 F DEBUG   : uid: 1047
09-12 17:09:08.154 16957 16957 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
09-12 17:09:08.154 16957 16957 F DEBUG   :     x0  0000000000000000  x1  000000000000422a  x2  0000000000000006  x3  000000725812fad0
09-12 17:09:08.154 16957 16957 F DEBUG   :     x4  666d687164666668  x5  666d687164666668  x6  666d687164666668  x7  7f7f7f7f7f7f7f7f
09-12 17:09:08.154 16957 16957 F DEBUG   :     x8  00000000000000f0  x9  00000074fde23a00  x10 0000000000000001  x11 00000074fde61dd4
09-12 17:09:08.154 16957 16957 F DEBUG   :     x12 000000725812f5a0  x13 0000000000000036  x14 000000725812f728  x15 00000000000b20aa
09-12 17:09:08.154 16957 16957 F DEBUG   :     x16 00000074fdec6d58  x17 00000074fdea3c40  x18 00000072542a4000  x19 00000000000035c2
09-12 17:09:08.154 16957 16957 F DEBUG   :     x20 000000000000422a  x21 00000000ffffffff  x22 0000007505b4f0f9  x23 0000007258130000
09-12 17:09:08.154 16957 16957 F DEBUG   :     x24 000000728ba1ccf0  x25 000000725812fb84  x26 000000725812fff8  x27 00000000000fc000
09-12 17:09:08.154 16957 16957 F DEBUG   :     x28 00000000000fe000  x29 000000725812fb50
09-12 17:09:08.154 16957 16957 F DEBUG   :     lr  00000074fde53958  sp  000000725812fab0  pc  00000074fde53984  pst 0000000000000000
09-12 17:09:08.154 16957 16957 F DEBUG   : backtrace:
09-12 17:09:08.154 16957 16957 F DEBUG   :       #00 pc 0000000000051984  /apex/com.android.runtime/lib64/bionic/libc.so (abort+164) (BuildId: f73b4091a9c819a21763f5bf11b53c2b)
09-12 17:09:08.154 16957 16957 F DEBUG   :       #01 pc 0000000000114dcc  /system/lib64/libcameraservice.so (android::CameraServiceWatchdog::threadLoop()+348) (BuildId: d559a03a904e30747abfd8a951d6f210)
09-12 17:09:08.154 16957 16957 F DEBUG   :       #02 pc 0000000000012f18  /system/lib64/libutils.so (android::Thread::_threadLoop(void*)+416) (BuildId: a5ca39eaeeeb017be4afa358811e69fb)
09-12 17:09:08.154 16957 16957 F DEBUG   :       #03 pc 00000000000b6380  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+208) (BuildId: f73b4091a9c819a21763f5bf11b53c2b)
09-12 17:09:08.154 16957 16957 F DEBUG   :       #04 pc 00000000000530a8  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: f73b4091a9c819a21763f5bf11b53c2b)
```

szmarczak, 2023-09-12:
`expo-camera` also works as well, so this is definitely an issue with vision camera.

mrousavy, 2023-09-14:
Can you try removing those 25 lines: https://github.com/mrousavy/react-native-vision-camera/blob/4e96eb77e0755edf5a4351d61fa36d60415f1db6/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L436-L460

And see what happens?

szmarczak, 2023-09-15:
It still crashes. I tried removing `Zoom` and `Torch Mode` and it kept crashing as well.

Space6188, 2023-09-29:
try to delete zoom as prop from <Camera /> and use only animatedProps

mrousavy, 2023-09-30:
Hey, thanks for the detailed bug report. Is this related to #1801 ?

szmarczak, 2023-10-03:
The issue you mentioned looks exactly like this one, although I tried commenting out the code you mentioned and - with the code at that time - it still kept crashing. I'll test on the latest commit when I have time.

mrousavy, 2023-10-06:
Hey!

Quick update: I think there could be four problems here:

1. You are using a `limited`/`legacy` device, and use Preview (always on), `photo` and `video`/`frameProcessor` outputs (3 outputs) which are by default at maximum resolution. LIMITED/LEGACY devices might not support full-resolution streaming, see [the Android capture session documentation](https://developer.android.com/reference/android/hardware/camera2/CameraDevice#regular-capture) on supported configuration maps.
    We need `PRIV PREVIEW` + `YUV/PRIV RECORD` + `JPEG MAXIMUM`. anything lower, needs to be explicitly handled by me, but I can easily add that no problemo. I just need the device to test it on.
    [image]
    **What you can do:** Try disabling `photo`, `video` and `frameProcessor` and check if that works. If only PREVIEW is attached, everything should work.
2. We are using some setting in the Capture Request that is not supported (e.g. FPS, HDR, Zoom, ...) but we don't previously check if those settings are supported on the device. I can easily add some checks here, so that the Camera doesn't just swallow any errors and fails silently, but instead throws and says something like "60 FPS is not supported on this device". https://github.com/mrousavy/react-native-vision-camera/blob/fb6ebd9ee1b831ce1c2e0b42d123257f5f9c5459/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L461-L492
     **What you can do:** Try disabling each line in `getPreviewCaptureRequest` and see what happens. I've heard `zoom` is often the cause.
3. Some manufacturers (cough* cough Samsung) explicitly say they support something (e.g. 60 FPS), but then don't support it and just swallow errors and show blackscreens when trying to use it. This is the most ridiculous sh*t I've ever seen, but yea, that's how it is.
    **What you can do:** Also try other steps here and remove code by code to see what it is. Maybe also browse similar issues in the [CameraX issue tracker](https://issuetracker.google.com/hotlists/1551123) or StackOverflow to see if someone had similar issues, explicitly by searching for your device.
4. There could be some synchronization issue. I can try to make everything fully atomic so nothing ever goes out of sync with mutexes. This is pretty tricky, but I'll do it

mrousavy, 2023-10-24:
I **think** i just fixed this in #2049, and released it to npm (v3.6.0) 🎉 

Please try this and let me know if this is still an issue.

If you appreciate my work and dedication to fix your issues and make VisionCamera better, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. Thanks!

closed 2023-10-24 by mrousavy