margelo/react-native-vision-camera #1797: Android frameprocessor Image is already closed

jess-hwang, 2023-09-13:
### What's happening?

When turning on frameprocessor, the following error occurs.

### Reproduceable Code

```tsx
react-native 0.72.0
react-native-vision-camera 3.0.0
react-native-reanimated 3.4.2
react-native-worklets-core 0.2.0
```

### Relevant log output

```shell
Error: Exception in HostObject::get for prop 'width': java.lang.IllegalStateException: Image is already closed, js engine: hermes
Error: Exception in HostObject::get for prop 'width': java.lang.IllegalStateException: Image is already closed, js engine: hermes
Received unknown JNI type "class [F"! Cannot convert to jsi::Value.
```
```

### Camera Device

```shell
using format
{"autoFocusSystem": "contrast-detection", "fieldOfView": 104.12870642532393, "maxFps": 30, "maxISO": 3200, "minFps": 1,
```

### Device

Android 

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

jess-hwang, 2023-09-13:
solved 

I wrote wrong return type of plugin.ts

closed 2023-09-13 by jess-hwang

jess-hwang, 2023-09-14:
@vishva-simform  

```
export function plugin(frame:Frame): Frame {
  'worklet'
...
}

fixed

export function plugin(fame: Frame): String[] {
  'worklet'
...
}
```

vishva-simform, 2023-09-14:
@jess-hwang  
This is my code where I am facing the above error and the error goes infinite, Can you please help me with that?

Error: **Error: Exception in HostObject::get for prop 'width': java.lang.IllegalStateException: Image is already closed**

`import React, {useEffect, useRef} from 'react';
import {StyleSheet, Text} from 'react-native';

import {
  Camera,
  useCameraDevices,
  useFrameProcessor,
} from 'react-native-vision-camera';

const HomeScreen = () => {
  const [cameraAccess, setCameraAccess] = React.useState(false);
  const devices = useCameraDevices();
  const isBackCamera = devices.back;
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    console.log(frame, 'frame');
  }, []);
  const camera = useRef();

  const askPermission = async () => {
    try {
      const getPermission = await Camera.getCameraPermissionStatus();
      if (getPermission === 'denied') {
        const reqPrmission = await Camera.requestCameraPermission();
        if (reqPrmission === 'granted') {
          setCameraAccess(true);
          console.log('Access granted');
        }
      }
    } catch (e) {
      console.log(e);
    }
  };

  useEffect(() => {
    console.log('useEffect is called');
    askPermission();
  }, []);

  if (isBackCamera == null && !cameraAccess) {
    console.log('camera not found');
    return <Text>Waiting</Text>;
  }
  return (
    <Camera
      ref={camera}
      style={StyleSheet.absoluteFill}
      device={isBackCamera}
      isActive={true}
      frameProcessor={frameProcessor}
    />
  );
};

export default HomeScreen;
`

w4t3r-45, 2023-09-20:
i have the same issue , any solutions please ?

abdelrahman-muntaser, 2023-12-04:
@mrousavy  can you help with this

7emretelli, 2023-12-29:
> i have the same issue , any solutions please ?

For anyone who is still facing this issue, here is how I solved the problem:

1) Firstly keep if camera is active or not in a useState hook and use it in the Camera component:
```diff
      <Camera
        frameProcessor={frameProcessor}
        device={device}
+        isActive={isCameraActive}
       {...otherCameraProps}
      />
```

2) Whenever page change or camera is supposed to be closed, set isCameraActive to false.
3) In your frameProcessor function which uses "useFrameProcessor", open an array for dependencies and include "isCameraActive".
```diff
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    // your code here

+  }, [isCameraActive]);
```

4) Inside your frameProcessor function, after 'worklet', use condition to check if camera is active. If camera is not active, return without continue processing next lines.
```diff
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
+    if (!isCameraActive) {
+     return;
+   }
   // your code here
  }, [isCameraActive]);
```

jessiemichelleblack, 2024-01-25:
Happening on iOS device too.

Error: Cannot get `width`, frame is already closed!

7emretelli, 2024-01-28:
@mrousavy issue still continues, thanks.

vishaljak, 2024-02-23:
@mrousavy getting this issue on android, but the interesting part is that I am not using frame processors. I am just trying to capture an image. I am not sure what's wrong. This is the error exactly: `unknown/unknown: [unknown/unknown] Image is already closed`. I am trying to debug but no idea so far

FernandoAOborges, 2024-03-09:
"java.lang.IllegalStateException: Image is already closed
     FATAL EXCEPTION: mrousavy/VisionCamera.video"  

any news?