margelo/react-native-vision-camera #1801: 🐛 Android camera not working v3

lc-mm, 2023-09-14:
### What's happening?

On 3 different Android devices running the example app from the main branch results in the error `[session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]`

Screen is just blank and the app crashes upon pressing the buttons in the top right.

### Reproduceable Code

```tsx
Running example app in main branch
```

### Relevant log output

```shell
2023-09-13 22:14:23.211 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Running "VisionCameraExample" with {"rootTag":11}
2023-09-13 22:14:23.366 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering Navigator. Camera: undefined | Microphone: undefined
2023-09-13 22:14:23.417 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Loading react-native-worklets-core...
2023-09-13 22:14:23.419 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Worklets loaded successfully
2023-09-13 22:14:23.424 4050-4121/com.mrousavy.camera.example I/VisionCameraProxy: Initializing VisionCameraProxy...
2023-09-13 22:14:23.424 4050-4121/com.mrousavy.camera.example I/VisionCameraProxy: Creating Worklet Context...
2023-09-13 22:14:23.427 4050-4121/com.mrousavy.camera.example I/VisionCameraProxy: Worklet Context created!
2023-09-13 22:14:23.454 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering Navigator. Camera: granted | Microphone: undefined
2023-09-13 22:14:23.455 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering Navigator. Camera: granted | Microphone: granted
2023-09-13 22:14:23.704 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: re-rendering camera page without active camera
2023-09-13 22:14:23.732 4050-4122/com.mrousavy.camera.example W/unknown:ViewManagerPropertyUpdater: Could not find generated setter for class com.swmansion.rnscreens.ScreensShadowNode
2023-09-13 22:14:23.764 4050-4149/com.mrousavy.camera.example I/CameraManagerGlobal: Connecting to camera service
2023-09-13 22:14:23.764 4050-4050/com.mrousavy.camera.example I/ReactNative: [GESTURE HANDLER] Initialize gesture handler for root view com.facebook.react.ReactRootView{59a088c V.E...... .......D 0,0-1080,2358 #b}
2023-09-13 22:14:23.768 4050-4149/com.mrousavy.camera.example V/CameraExtensionManagerGlobal: Choosing the vendor camera extensions proxy package: com.google.android.apps.camera.services
2023-09-13 22:14:23.768 4050-4149/com.mrousavy.camera.example V/CameraExtensionManagerGlobal: Choosing the vendor camera extensions proxy service: com.google.android.apps.camera.services.extensions.service.PixelExtensions
2023-09-13 22:14:23.771 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: re-rendering camera page without active camera
2023-09-13 22:14:24.040 4050-4121/com.mrousavy.camera.example I/ReactNativeJS: Re-rendering camera page with active camera. Device: "BACK (0)" (4000x2000 photo / 3840x2160 video @ 60fps)
2023-09-13 22:14:24.079 4050-4142/com.mrousavy.camera.example I/CameraSession: Camera became available: 0
2023-09-13 22:14:24.079 4050-4142/com.mrousavy.camera.example I/CameraSession: Camera became available: 1
2023-09-13 22:14:24.079 4050-4050/com.mrousavy.camera.example I/CameraView: Props changed: [enableFrameProcessor, audio, cameraId, photo, orientation, video, isActive, fps, format, lowLightBoost, hdr]
2023-09-13 22:14:24.084 4050-4050/com.mrousavy.camera.example I/PreviewSize: Phone has a 1080 x 2240 screen.
2023-09-13 22:14:24.085 4050-4050/com.mrousavy.camera.example I/PreviewView: Using Preview Size 1920 x 1080.
2023-09-13 22:14:24.085 4050-4050/com.mrousavy.camera.example D/PreviewView: onMeasure(0, 0)
2023-09-13 22:14:24.085 4050-4050/com.mrousavy.camera.example D/PreviewView: Measured dimensions set: 0 x 0
2023-09-13 22:14:24.085 4050-4050/com.mrousavy.camera.example I/CameraView: Configuring Camera Device...
2023-09-13 22:14:24.086 4050-4050/com.mrousavy.camera.example I/CameraSession: Setting Format (fps: 60 | videoStabilization: null | hdr: false | lowLightBoost: false)...
2023-09-13 22:14:24.086 4050-4050/com.mrousavy.camera.example I/CameraSession: Setting isActive: false (isRunning: false)
2023-09-13 22:14:24.106 4050-4050/com.mrousavy.camera.example I/CameraSession: Setting isActive: true (isRunning: false)
2023-09-13 22:14:24.107 4050-4050/com.mrousavy.camera.example D/PreviewView: onMeasure(1080, 2358)
2023-09-13 22:14:24.107 4050-4050/com.mrousavy.camera.example D/PreviewView: Measured dimensions set: 1326 x 2358
2023-09-13 22:14:24.120 4050-4050/com.mrousavy.camera.example I/PreviewView: Surface created! Surface(name=null)/@0x201acd2
2023-09-13 22:14:24.120 4050-4050/com.mrousavy.camera.example I/CameraView: Configuring Camera Device...
2023-09-13 22:14:24.120 4050-4050/com.mrousavy.camera.example I/CameraSession: Configuring Session for Camera 0...
2023-09-13 22:14:24.122 4050-4050/com.mrousavy.camera.example I/CameraOutputs: Preparing Outputs for Camera 0...
2023-09-13 22:14:24.122 4050-4050/com.mrousavy.camera.example I/CameraOutputs: Adding native preview view output.
2023-09-13 22:14:24.124 4050-4050/com.mrousavy.camera.example I/PreviewSize: Phone has a 1080 x 2240 screen.
2023-09-13 22:14:24.127 4050-4050/com.mrousavy.camera.example I/CameraOutputs: Adding 4000x2000 photo output. (Format: 256)
2023-09-13 22:14:24.129 4050-4050/com.mrousavy.camera.example I/CameraOutputs: Adding 3840x2160 video output. (Format: 34)
2023-09-13 22:14:24.129 4050-4050/com.mrousavy.camera.example I/CameraOutputs: Prepared 3 Outputs for Camera 0!
2023-09-13 22:14:24.129 4050-4050/com.mrousavy.camera.example I/PreviewView: Surface resized! Surface(name=null)/@0x201acd2 (1920 x 1080 in format #4)
2023-09-13 22:14:24.130 4050-4142/com.mrousavy.camera.example I/CameraSession: Starting Camera Session...
2023-09-13 22:14:24.130 4050-4142/com.mrousavy.camera.example I/CameraManager: Camera 0: Opening...
2023-09-13 22:14:24.133 4050-4050/com.mrousavy.camera.example D/VisionCameraProxy: Finding view 45...
2023-09-13 22:14:24.133 4050-4050/com.mrousavy.camera.example D/VisionCameraProxy: Found view 45!
2023-09-13 22:14:24.134 1281-15551/? I/CameraService: CameraService::connect call (PID -1 "com.mrousavy.camera.example", camera ID 0) and Camera API version 2
2023-09-13 22:14:24.134 1281-15551/? I/Camera2ClientBase: Camera 0: Opened. Client: com.mrousavy.camera.example (PID 4050, UID 10292)
2023-09-13 22:14:24.147 4050-4142/com.mrousavy.camera.example I/CameraSession: Camera became un-available: 0
2023-09-13 22:14:24.147 4050-4142/com.mrousavy.camera.example I/CameraManager: Camera 0: Opened!
2023-09-13 22:14:24.149 4050-4142/com.mrousavy.camera.example I/CreateCaptureSession: Camera 0: Creating Capture Session #1000... Hardware Level: 1} | Outputs: [PREVIEW (1920 x 1080), PHOTO (4000 x 2000 in format #256), VIDEO (3840 x 2160 in format #34)]
2023-09-13 22:14:24.149 4050-4142/com.mrousavy.camera.example I/CreateCaptureSession: Using new API (>=28)
2023-09-13 22:14:24.244 4050-4142/com.mrousavy.camera.example I/CreateCaptureSession: Camera 0: Capture Session #1000 configured!
2023-09-13 22:14:24.244 4050-4142/com.mrousavy.camera.example I/CameraView: invokeOnInitialized()
2023-09-13 22:14:24.245 4050-4142/com.mrousavy.camera.example I/CameraSession: Adding output surface PREVIEW..
2023-09-13 22:14:24.245 4050-4142/com.mrousavy.camera.example I/CameraSession: Adding output surface VIDEO..
2023-09-13 22:14:24.245 4050-4142/com.mrousavy.camera.example I/CameraSession: Camera Session initialized! Starting repeating request..
[... 25 lines cut ...]
2023-09-13 22:14:27.759 4050-4142/com.mrousavy.camera.example I/CameraSession: Camera became available: 1
2023-09-13 22:14:27.759 4050-4142/com.mrousavy.camera.example I/CameraSession: Camera became available: 0
2023-09-13 22:14:34.610 4050-4063/com.mrousavy.camera.example I/.camera.exampl: Background concurrent copying GC freed 459370(19MB) AllocSpace objects, 5(100KB) LOS objects, 79% free, 6185KB/30MB, paused 154us,56us total 108.600ms
2023-09-13 22:14:34.613 1281-11857/? I/Camera2ClientBase: Closed Camera 0. Client was: com.mrousavy.camera.example (PID 4050, UID 10292)
```

### Device

Pixel 6a

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

lc-mm, 2023-09-14:
Camera Device:

```json
{
  "formats": [
    {
      "pixelFormats": [
        "rgb",
        "native",
        "yuv"
      ],
      "videoStabilizationModes": [
        "off",
        "standard",
        "off",
        "cinematic-extended"
      ],
      "autoFocusSystem": "contrast-detection",
      "supportsPhotoHDR": false,
      "photoHeight": 3024,
      "supportsVideoHDR": false,
      "maxISO": 7111,
      "minISO": 55,
      "minFps": 1,
      "videoWidth": 3840,
      "videoHeight": 2160,
      "photoWidth": 4032,
      "fieldOfView": 65.59413150556766,
      "maxFps": 60
    },
    {
      "pixelFormats": [
        "rgb",
        "native",
        "yuv"
      ],
      "videoStabilizationModes": [
        "off",
        "standard",
        "off",
        "cinematic-extended"
      ],
      "autoFocusSystem": "contrast-detection",
      "supportsPhotoHDR": false,
      "photoHeight": 3000,
      "supportsVideoHDR": false,
      "maxISO": 7111,
      "minISO": 55,
      "minFps": 1,
      "videoWidth": 3840,
      "videoHeight": 2160,
      "photoWidth": 4000,
      "fieldOfView": 65.59413150556766,
      "maxFps": 60
    },
    {
      "pixelFormats": [
        "rgb",
        "native",
        "yuv"
      ],
      "videoStabilizationModes": [
        "off",
        "standard",
        "off",
        "cinematic-extended"
      ],
      "autoFocusSystem": "contrast-detection",
      "supportsPhotoHDR": false,
      "photoHeight": 2268,
[... 6912 lines cut ...]
      "photoWidth": 1920,
      "fieldOfView": 65.59413150556766,
      "maxFps": 30
    }
  ],
  "hardwareLevel": "full",
  "maxZoom": 7,
  "minZoom": 0.6009132266044617,
  "supportsLowLightBoost": true,
  "sensorOrientation": "landscape-right",
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": true,
  "hasTorch": true,
  "devices": [
    "wide-angle-camera"
  ],
  "hasFlash": true,
  "name": "BACK (0)",
  "position": "back",
  "id": "0"
}
```

mrousavy, 2023-09-14:
Can you try removing those 25 lines: https://github.com/mrousavy/react-native-vision-camera/blob/4e96eb77e0755edf5a4351d61fa36d60415f1db6/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L436-L460

And see what happens?

satyabrata-dash5, 2023-09-14:
Same with me Camera not opening 
[image]

here is the log :
[image]

lc-mm, 2023-09-14:
> Can you try removing those 25 lines:
> 
> https://github.com/mrousavy/react-native-vision-camera/blob/4e96eb77e0755edf5a4351d61fa36d60415f1db6/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L436-L460
> 
> And see what happens?

The camera is now rendering after removing those lines

lc-mm, 2023-09-15:
The camera is working now, but while using a frame processor I am getting some stutters (in debug and release builds). Whenever the stutters happen I get a bunch of logs saying

```
VideoPipeline: ImageReader failed to acquire a new image!
```

The performance of the frame processor is fast before and after the stutters

Any ideas that I could try for this?

iketiunn, 2023-09-17:
> Can you try removing those 25 lines:
> 
> https://github.com/mrousavy/react-native-vision-camera/blob/4e96eb77e0755edf5a4351d61fa36d60415f1db6/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L436-L460
> 
> And see what happens?

This worked for me.

Was getting blank after enabling `video`.
After commenting on these lines and rebuilding. the camera preview and recording is working now.

mrousavy, 2023-09-19:
@iketiunn @lc-mm thanks for testing. Can you try to add each setting back one by one to find out which of those format settings is causing the Camera to fail? Thanks I appreciate your help, I don't have any device to test this on so I'm relying on you guys ❤️ 

@lc-mm regarding the pixelFormat, this is WIP I found an amazing solution here: https://github.com/mrousavy/react-native-vision-camera/pull/1789 (+ needs this to fix video recording: https://github.com/mrousavy/react-native-vision-camera/pull/1799)

@satyabrata-dash5 you don't have any Camera Devices. You're doing something wrong there. Check the docs

lc-mm, 2023-09-19:
Thanks @mrousavy I will let you know when I have a chance to narrow down which setting is causing the camera to fail.

For the slight performance issue - I was running on code with the changes from #1789 and I was seeing the intermittent pause with the message `VideoPipeline: ImageReader failed to acquire a new image!`

iketiunn, 2023-09-23:
> @iketiunn @lc-mm thanks for testing. Can you try to add each setting back one by one to find out which of those format settings is causing the Camera to fail? Thanks I appreciate your help, I don't [...]

I do the development on the Mi Mix 2s, but I also tested it on my day-to-day phone.

Tested device:
- Mi Mix 2s Android 10
Results:
- FPS passed ✅
- Video Stabilization passed ✅
- Night/HDR Mode passed ✅
- Zoom not passed 🚫
- Torch Mode passed ✅

Tested device:
- Pixel 5 Android 13
Results:
- FPS passed ✅
- Video Stabilization passed ✅
- Night/HDR Mode passed ✅
- Zoom not passed ✅
- Torch Mode passed ✅

```java
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.R) {
      captureRequest.set(CaptureRequest.CONTROL_ZOOM_RATIO, zoom)
    } else {
      val cameraCharacteristics = cameraManager.getCameraCharacteristics(cameraId!!)
      val size = cameraCharacteristics.get(CameraCharacteristics.SENSOR_INFO_ACTIVE_ARRAY_SIZE)!!
      captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
    }
```

The problem might be on the phone before Android 12.
I'm not an Android developer or a mobile developer. So this is what I can do so far.
Hope this gives you or others some clues.

deps:
```
"dependencies": {
    "@config-plugins/ffmpeg-kit-react-native": "^5.0.0",
    "@expo/vector-icons": "^13.0.0",
    "@react-navigation/native": "^6.0.2",
    "expo": "~49.0.10",
    "expo-build-properties": "^0.8.3",
    "expo-font": "~11.4.0",
    "expo-linking": "~5.0.2",
    "expo-router": "^2.0.5",
    "expo-splash-screen": "~0.20.5",
    "expo-status-bar": "~1.6.0",
    "expo-system-ui": "~2.4.0",
    "expo-web-browser": "~12.3.2",
    "ffmpeg-kit-react-native": "^5.1.0",
    "react": "18.2.0",
    "react-dom": "18.2.0",
    "react-native": "^0.72.4",
    "react-native-gesture-handler": "~2.12.0",
    "react-native-safe-area-context": "4.6.3",
    "react-native-screens": "~3.22.0",
    "react-native-tcp-socket": "^6.0.6",
    "react-native-vision-camera": "3",
    "react-native-web": "~0.19.6",
    "react-native-worklets-core": "^0.2.0"
  }
```

mrousavy, 2023-09-27:
Wait so zoom makes it break?

igperez-ar, 2023-09-29:
> Wait so zoom makes it break?

Apparently yes, same here https://github.com/mrousavy/react-native-vision-camera/issues/1858#issuecomment-1739959224

wajahat9876, 2023-10-06:
I have an issue that on my react native application MaskView causing my camera to not open on android device but all i have done is that update my MaskView and Expo Camera library and make a new Development Client after that my application is running Fine and issue resolved.

lc-mm, 2023-10-20:
@mrousavy I have had more time to look at this now, I have updated to version 3.5.1 and this is still an issue.

This issue is this [line](https://github.com/mrousavy/react-native-vision-camera/blob/23d173f6fc9cce4fe643d0c2fa296329b36ca4fe/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L459): `captureRequest.set(CaptureRequest.CONTROL_SCENE_MODE, sceneMode)`

When `sceneMode` is null because the prop `hdr` and `lowLightBoost` are both false. If I set either of those to true then the crash went away.

Here is some code that I used to stop the crash, not sure if it is the correct approach:

```
    captureRequest.set(
      CaptureRequest.CONTROL_SCENE_MODE,
      if (sceneMode != null) sceneMode else CaptureRequest.CONTROL_SCENE_MODE_DISABLED
    )
```

meabed, 2023-10-23:
I have same issue when i change the pixel format to yuv or rgb and assign frameprocessor.
it's black screen and in the logs it shows Unknown Camera Error 4 - camera disconnected 

https://github.com/mrousavy/react-native-vision-camera/issues/1947#issuecomment-1774243757

mrousavy, 2023-10-24:
I **think** I just fixed this in #2049, and released it to npm (v3.6.0) 🎉 

Please try the latest release and let me know if this is still an issue.

If you appreciate my work and dedication to fix your issues and make VisionCamera better, please [💖 consider sponsoring me on GitHub 💖](https://github.com/sponsors/mrousavy) to say thanks. Thanks!

lucksp, 2023-10-29:
I am seeing the `CameraSessionCannotBeConfiguredError` error, only on Android:

> ERROR  Your app just crashed. See the error below.
com.mrousavy.camera.CameraSessionCannotBeConfiguredError: [session/cannot-create-session] Failed to create a Camera Session for Camera 0! Outputs: [PREVIEW (1280 x 960), PHOTO (1280 x 960 in format #256)]
  com.mrousavy.camera.extensions.CameraDevice_createCaptureSessionKt$createCaptureSession$2$callback$1.onConfigureFailed(CameraDevice+createCaptureSession.kt:55)
  android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy.lambda$onConfigureFailed$1$android-hardware-camera2-impl-CallbackProxies$SessionStateCallbackProxy(CallbackProxies.java:64)
  android.hardware.camera2.impl.CallbackProxies$SessionStateCallbackProxy$$ExternalSyntheticLambda4.run(Unknown Source:4)
  android.os.Handler.handleCallback(Handler.java:942)
  android.os.Handler.dispatchMessage(Handler.java:99)
  android.os.Looper.loopOnce(Looper.java:201)
  android.os.Looper.loop(Looper.java:288)
  android.os.HandlerThread.run(HandlerThread.java:67)

This happens after I navigate to a new screen (modal) and then go back to the previous Screen View (regular screen view), which is where the camera view lives.  

As suggested, I have upgraded to the latest version `^3.6.4`

zjkuang, 2023-10-30:
With 3.6.4 the camera is not working steadily on Android, sometimes it is working but sometimes the camera view is dark. I got some error logs from Logcat:
```
CameraService::connect evicting conflicting client for camera ID 0
...
Failed to configure CameraSession! Error: [session/camera-not-ready] The Camera is not ready yet! Wait for the onInitialized() callback!, Config-Diff: Difference(deviceChanged=false, outputsChanged=false, sidePropsChanged=true)
```

mrousavy, 2023-10-31:
@zjkuang what happens if you change this line here:

https://github.com/mrousavy/react-native-vision-camera/blob/f4617a3918783ff188c8b63f0694b1ad94ce3fae/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L367

to

```kt
val captureSession = captureSession ?: return
```

?

zjkuang, 2023-10-31:
> @zjkuang what happens if you change this line here:
> 
> https://github.com/mrousavy/react-native-vision-camera/blob/f4617a3918783ff188c8b63f0694b1ad94ce3fae/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L367
> 
> to
> 
> ```kotlin
> val captureSession = captureSession ?: return
> ```
> 
> ?

Thanks @mrousavy . I tried this change but it didn't work, or perhaps I could say, it's getting worse. Without this change, **_occasionally_** the camera view could show up. (Thus it seemed to be a race condition I guess?) But with this change I tested quite a few times and the camera view is always black. Also the error logs seem to be a little different with this change:
```
2023-10-31 09:02:13.951  1100-5529  CameraService           cameraserver                         E  CameraService::connect evicting conflicting client for camera ID 0
2023-10-31 09:02:13.952  1100-5529  Camera2ClientBase       cameraserver                         D  Camera 0: start to disconnect
2023-10-31 09:02:13.952  1100-5529  Camera2ClientBase       cameraserver                         D  Camera 0: serializationLock acquired
2023-10-31 09:02:13.952  1100-5529  Camera2ClientBase       cameraserver                         D  Camera 0: Shutting down
2023-10-31 09:02:13.952  1100-5529  Camera2ClientBase       cameraserver                         D  Camera 0: start to cacheDump
2023-10-31 09:02:13.952  1100-5529  camera_metadata         cameraserver                         E  validate_camera_metadata_structure: metadata is null!
2023-10-31 09:02:13.952  1100-5529  Camera2-Metadata        cameraserver                         E  acquire: Failed to validate metadata structure 0x0
2023-10-31 09:02:13.952  1100-5529  camera_metadata         cameraserver                         E  validate_camera_metadata_structure: metadata is null!
2023-10-31 09:02:13.952  1100-5529  Camera2-Metadata        cameraserver                         E  acquire: Failed to validate metadata structure 0x0
...
```

SherryLinn, 2023-11-01:
[image]
I also exist here. What should I do? Wait online

mrousavy, 2023-11-07:
So it is a race condition I'd say

SherryLinn, 2023-11-07:
您好！您发送的邮件我已收到。谢谢！

MehPhisto, 2023-11-23:
I'm having the same issue using the back camera on Android v10 (Nokia 7.1) and react-native-vision-camera v3.6.8. Camera screen is black and after a few seconds i get the following error: 

`Camera.onError(session/camera-has-been-disconnected): [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR [session/camera-has-been-disconnected: [session/camera-has-been-disconnected] The given Camera device (id: 0) has been disconnected! Error: UNKNOWN_CAMERA_DEVICE_ERROR]`

The problem is probably still coming from the zoom, in:
https://github.com/mrousavy/react-native-vision-camera/blob/ccb04016a0fbe1188b7ff6798df5d64099f458fc/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L446
Deleting this line is solving the problem for me.

SherryLinn, 2023-11-23:
您好！您发送的邮件我已收到。谢谢！

mrousavy, 2023-11-24:
I wonder why zoom makes this break, are you passing a custom zoom value? What's going on here

MehPhisto, 2023-11-24:
Not at all, here is settings i'm using inside my test component.

````````  
export const BarcodeScanner = ({onCodeScanned, isActive = true, ...rest}: BarcodeScannerProps) => {
  ...
  const device = useCameraDevice('back');
  const codeScanner = useCodeScanner({
    codeTypes: ['ean-13'],
    onCodeScanned: (...args) => {
      // TODO: Implement workaround for regionOfInterest ?
  
      onCodeScanned(...args);
    },
    regionOfInterest: { // Not working on android
      x: deviceWidth*0.1,
      y: deviceWidth*0.1,
      width: deviceWidth*0.8,
      height: 200,
    }
  });

  ...
  
  return (
    ...
    <Camera
      {...rest} // only styles settings inside
      device={device}
      isActive={isActive}
      codeScanner={codeScanner}
    />
    ...
  );
};

abdalamichel, 2023-12-07:
> Not at all, here is settings i'm using inside my test component.
> 
> ```
> const {width: deviceWidth, height: deviceHeight} = Dimensions.get('window');
> 
> export const BarcodeScanner = ({onCodeScanned, isActive = true, ...rest}: BarcodeScannerProps) => {
>   ...
>   const device = useCameraDevice('back');
>   const codeScanner = useCodeScanner({
>     codeTypes: ['ean-13'],
>     onCodeScanned: (...args) => {
>       // TODO: Implement workaround for regionOfInterest ?
>   
>       onCodeScanned(...args);
>     },
>     // regionOfInterest: { // Not working on android
>       // x: deviceWidth*0.1,
>       // y: deviceWidth*0.1,
>       // width: deviceWidth*0.8,
>       // height: 200,
>     // }
>   });
> 
>   ...
>   
>   return (
>     ...
>     <Camera
>       {...rest} // only styles settings inside
>       device={device}
>       isActive={isActive}
>       codeScanner={codeScanner}
>     />
>     ...
>   );
> };
> ```

Same issue for me, it is related to codeScanner. Without codeScanner it works perfect

paulovictor237, 2023-12-18:
I'm going through the same problem.
Android 12 and below: presents the same error
Android 13 and above: works normally

However, 25% of my app's user base are on Android version 12, making the use of the library unfeasible.

spsaucier, 2023-12-18:
Most of my users who are having this problem are on Android 13.

fahmifan, 2023-12-20:
> Most of my users who are having this problem are on Android 13.

recently having this issue too, last week it's fine though ://

SherryLinn, 2024-01-13:
您好！您发送的邮件我已收到。谢谢！

mrousavy, 2024-01-15:
I'll add some special conditional checks to make sure this works

closed 2024-01-15 by mrousavy

Henriquehlr, 2024-02-26:
I'm facing the same issue on Android 12 when navigating between screens

version: 3.9.0