margelo/react-native-vision-camera #1802: [V2] iOS - App Crashes at Launch when built with Xcode 15

marcshilling, 2023-09-14:
### What's happening?

When built using Xcode 15 release candidate, the app is crashing at launch. Before the crash, the message `Swift class extensions and categories on Swift classes are not allowed to have +load methods` is printed to the console.

Relevant info - I'm also using `vision-camera-code-scanner@0.2.0` for barcode scanning.

It seems to be related to https://github.com/mrousavy/react-native-vision-camera/pull/1308. If I revert that change, the startup crash goes away, but then when trying to use the camera it lags and then crashes (probably the original issue)

My workaround for now is continuing to use Xcode 14 to build.

### Reproduceable Code

```tsx
N/A
```

### Relevant log output

```shell
Code Type:           X86-64 (Native)
Role:                Foreground
Parent Process:      launchd_sim [26423]
Coalition:           com.apple.CoreSimulator.SimDevice.249DDD8B-78A4-4847-98B3-29645FFF9567 [76346]
Responsible Process: SimulatorTrampoline [24850]

Date/Time:           2023-08-25 16:33:58.9305 -0400
Launch Time:         2023-08-25 16:33:57.5265 -0400
OS Version:          macOS 13.5.1 (22G90)
Release Type:        User
Report Version:      104

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Termination Reason: OBJC 1 

Triggered by Thread:  0

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x114d49a52 __abort_with_payload + 10
1   libsystem_kernel.dylib        	       0x114d6784e abort_with_payload_wrapper_internal + 82
2   libsystem_kernel.dylib        	       0x114d677fc abort_with_reason + 19
3   libobjc.A.dylib               	    0x7ff800057593 _objc_fatalv(unsigned long long, unsigned long long, char const*, __va_list_tag*) + 121
4   libobjc.A.dylib               	    0x7ff80005751a _objc_fatal(char const*, ...) + 114
5   libobjc.A.dylib               	    0x7ff80004166d load_images + 1383
6   dyld_sim                      	       0x11389509b dyld4::RuntimeState::notifyObjCInit(dyld4::Loader const*) + 359
7   dyld_sim                      	       0x11389a73c dyld4::Loader::runInitializersBottomUp(dyld4::RuntimeState&, dyld3::Array<dyld4::Loader const*>&) const + 182
8   dyld_sim                      	       0x11389d49a dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const::$_1::operator()() const + 92
9   dyld_sim                      	       0x11389a7dd dyld4::Loader::runInitializersBottomUpPlusUpwardLinks(dyld4::RuntimeState&) const + 93
10  dyld_sim                      	       0x1138b03df dyld4::APIs::runAllInitializersForMain() + 263
11  dyld_sim                      	       0x11388c10a dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 2535
12  dyld_sim                      	       0x11388b356 _dyld_sim_prepare + 854
13  dyld                          	       0x117f55424 dyld4::prepareSim(dyld4::RuntimeState&, char const*) + 1490
14  dyld                          	       0x117f53abc dyld4::prepare(dyld4::APIs&, dyld3::MachOAnalyzer const*) + 249
15  dyld                          	       0x117f533bd start + 1805
```

### Camera Device

```json
N/A
```

### Device

iPhone 14 Pro

### VisionCamera Version

2.15.4

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Eldhose08, 2023-09-19:
Same issue in Xcode 15

EldhoseKuriakose, 2023-09-19:
Same issue after updating to xcode 15

phatmovista, 2023-09-19:
So basically the breaking change from Xcode 13 -> 14 are now reverted with Xcode 14 -> 15? 😂

marcshilling, 2023-09-19:
@ms314006 Just keep using Xcode 14 for now. I recommend using https://github.com/XcodesOrg/xcodes to manage Xcode versions

@phatmovista Yes, but unfortunately reverting it is not an option because the app crashes when using the camera (the original reason the change was put in)

jenskuhrjorgensen, 2023-09-19:
@mrousavy this is hitting us too. Does V3 work with Xcode 15 or is this bug also present in V3?

QSuraj, 2023-09-20:
Facing the same issue. App crashes on launch.

dimisus, 2023-09-20:
Yep, same issue. Xcode 15 (15A240d).

boticas, 2023-09-20:
Same issue here. It works with Xcode 14, but can't build for iOS 17.

QSuraj, 2023-09-20:
This should be related to this https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1727199001

dimisus, 2023-09-20:
Downgraded to 14.3.1. It worked.

https://developer.apple.com/download/applications/

gabrielmaldi, 2023-09-20:
I'm on version 2.15.4 and had to replace
```objc
+(void)load
```
with
```objc
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()
```

in two lines in the file `./ios/Frame Processor/FrameProcessorPlugin.h`:

https://github.com/mrousavy/react-native-vision-camera/blob/9963f4dedc6fbef933fb99e0adca1343f4f5f972/ios/Frame%20Processor/FrameProcessorPlugin.h#L36

https://github.com/mrousavy/react-native-vision-camera/blob/9963f4dedc6fbef933fb99e0adca1343f4f5f972/ios/Frame%20Processor/FrameProcessorPlugin.h#L56

This avoids the initial crash. Haven't tried the camera which I assume will crash as stated in https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1727739161.

DanielAraldi, 2023-09-20:
> I'm on version 2.15.4 and had to replace
> 
> ```objc
> +(void)load
> ```
> 
> with
> 
> ```objc
> __attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()
> ```
> 
> in two lines in the file `./ios/Frame Processor/FrameProcessorPlugin.h`:
> 
> https://github.com/mrousavy/react-native-vision-camera/blob/9963f4dedc6fbef933fb99e0adca1343f4f5f972/ios/Frame%20Processor/FrameProcessorPlugin.h#L36
> 
> https://github.com/mrousavy/react-native-vision-camera/blob/9963f4dedc6fbef933fb99e0adca1343f4f5f972/ios/Frame%20Processor/FrameProcessorPlugin.h#L56
> 
> This avoids the initial crash. Haven't tried the camera which I assume will crash as stated in [#1307 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1727739161).

Thank you, that's worked for me!

**Note**: I am using RN 0.68.2 here.

marcshilling, 2023-09-20:
@gabrielmaldi @DanielAraldi be warned (I mentioned this in my original comment), the reversion fixes the startup crash but the original crash (using the camera with a frame processor) will pop back up, making the fix unusable

gabrielmaldi, 2023-09-20:
Yes, this is a temporary fix to be able to at least work on the rest of the app. Thanks a lot.

erissun, 2023-09-21:
> I'm on version 2.15.4 and had to replace
>
> ```objc
> +(void)load
> ```
>
> with
>
> ```objc
> __attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()
> ```
>
> in two lines in the file [...]

still got this issue on RN 0.69.3

QSuraj, 2023-09-21:
The quick fix for now is to revert this change -  https://github.com/mrousavy/react-native-vision-camera/pull/1308/files

Replacing `vision-camera-code-scanner` with [`react-native-camera-kit`](https://github.com/teslamotors/react-native-camera-kit).

That solved the issue for me.

2sem, 2023-09-21:
crash is resolved with this. thx
```
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()
```

however I can't use camera.
```
"react-native-vision-camera": "2.16.1",
"react-native": "0.68.7",
```

[image]

QSuraj, 2023-09-22:
This should solve the issue permanently - https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1731248952

QSuraj, 2023-09-22:
```

#define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
                                                                                    \
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \

```

Should be

```

#define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
                                                                                    \
+(void)load   \

```

2sem, 2023-09-22:
```
"react-native-vision-camera": "2.15.4",
"react-native": "0.68.7",
```
```
#define VISION_EXPORT_SWIFT_FRAME_PROCESSOR(name, objc_name) \
...
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)() \
```

[image]

QSuraj, 2023-09-22:
@2sem 
You need to change `VISION_EXPORT_SWIFT_FRAME_PROCESSOR` not `VISION_EXPORT_FRAME_PROCESSOR `

2sem, 2023-09-22:
My FrameProcessorPlugin
```
#define VISION_CONCAT2(A, B) A##B
#define VISION_CONCAT(A, B) VISION_CONCAT2(A, B)

/**
 * Use this Macro to register the given function as a Frame Processor.
 * * Make sure the given function is a C-style function with the following signature: static inline id callback(Frame* frame, NSArray* args)
 * * Make sure the given function's name is unique across other frame processor plugins
 * * Make sure your frame processor returns a Value that can be converted to JS
 * * Make sure to use this Macro in an @implementation, not @interface
 *
 * The JS function will have the same name as the given Objective-C function, but with a "__" prefix.
 * Make sure to add that function to the babel.config.js under reanimated's "globals" option, and add TypeScript type declarations.
 */
#define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
                                                                                    \
+(void)load                                                                         \
{                                                                                   \
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #frame_processor callback:^id(Frame* frame, NSArray<id>* args) { \
    return frame_processor(frame, args);                                           \
  }];                                                                               \
}

/**
 * Same as VISION_EXPORT_FRAME_PROCESSOR, but uses __attribute__((constructor)) for
 * registration. Useful for registering swift classes that forbids use of +(void)load.
 */
#define VISION_EXPORT_SWIFT_FRAME_PROCESSOR(name, objc_name) \
objc_name : NSObject<FrameProcessorPluginBase>                                      \
@end                                                                                \
                                                                                    \
@interface objc_name (FrameProcessorPlugin)                                         \
@end                                                                                \
@implementation objc_name (FrameProcessorPlugin)                                    \
                                                                                    \
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)() \
{                                                                                   \
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
    return [objc_name callback:frame withArgs:args];                               \
  }];                                                                               \
}
```

Thormeard, 2023-09-25:
Hi, got the same issue. (2.16.1)
Managed to fix crashing on startup with https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1731248952
But still got an error when trying to use frame processor: `frame-processor/unavailable: Frame Processors are not enabled.`

Anything else im missing?

My FrameProcessorPlugin.h
```
//
//  FrameProcessorPlugin.h
//  VisionCamera
//
//  Created by Marc Rousavy on 01.05.21.
//  Copyright © 2021 mrousavy. All rights reserved.
//

#ifndef FrameProcessorPlugin_h
#define FrameProcessorPlugin_h

#import <Foundation/Foundation.h>
#import "FrameProcessorPluginRegistry.h"
#import "Frame.h"

@protocol FrameProcessorPluginBase
+ (id) callback:(Frame*)frame withArgs:(NSArray<id>*)args;
@end

#define VISION_CONCAT2(A, B) A##B
#define VISION_CONCAT(A, B) VISION_CONCAT2(A, B)

/**
 * Use this Macro to register the given function as a Frame Processor.
 * * Make sure the given function is a C-style function with the following signature: static inline id callback(Frame* frame, NSArray* args)
 * * Make sure the given function's name is unique across other frame processor plugins
 * * Make sure your frame processor returns a Value that can be converted to JS
 * * Make sure to use this Macro in an @implementation, not @interface
 *
 * The JS function will have the same name as the given Objective-C function, but with a "__" prefix.
 * Make sure to add that function to the babel.config.js under reanimated's "globals" option, and add TypeScript type declarations.
 */
#define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
                                                                                    \
+(void)load                                                                         \
{                                                                                   \
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #frame_processor callback:^id(Frame* frame, NSArray<id>* args) { \
    return frame_processor(frame, args);                                           \
  }];                                                                               \
}

/**
 * Same as VISION_EXPORT_FRAME_PROCESSOR, but uses __attribute__((constructor)) for
 * registration. Useful for registering swift classes that forbids use of +(void)load.
 */
#define VISION_EXPORT_SWIFT_FRAME_PROCESSOR(name, objc_name)                        \
objc_name : NSObject<FrameProcessorPluginBase>                                      \
@end                                                                                \
                                                                                    \
@interface objc_name (FrameProcessorPlugin) <FrameProcessorPluginBase>              \
@end                                                                                \
@implementation objc_name (FrameProcessorPlugin)                                    \
                                                                                    \
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
{                                                                                   \
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
    return [objc_name callback:frame withArgs:args];                               \
  }];                                                                               \
}

#endif /* FrameProcessorPlugin_h */

```

frenberg, 2023-09-26:
> @Thormeard, it's final change, worked for me. But only to "react-native-vision-camera": "2.15.2", if I update "react-native-vision-camera" to 2.16.1 I got error(Frame processors not enabled) again, I don't know why .
> 
> ```
> //
> //  FrameProcessorPlugin.h
> //  VisionCamera
> //
> //  Created by Marc Rousavy on 01.05.21.
> //  Copyright © 2021 mrousavy. All rights reserved.
> //
> 
> #ifndef FrameProcessorPlugin_h
> #define FrameProcessorPlugin_h
> 
> #import <Foundation/Foundation.h>
> #import "FrameProcessorPluginRegistry.h"
> #import "Frame.h"
> 
> @protocol FrameProcessorPluginBase
> + (id) callback:(Frame*)frame withArgs:(NSArray<id>*)args;
> @end
> 
> 
> #define VISION_CONCAT2(A, B) A##B
> #define VISION_CONCAT(A, B) VISION_CONCAT2(A, B)
> 
> /**
>  * Use this Macro to register the given function as a Frame Processor.
>  * * Make sure the given function is a C-style function with the following signature: static inline id callback(Frame* frame, NSArray* args)
>  * * Make sure the given function's name is unique across other frame processor plugins
>  * * Make sure your frame processor returns a Value that can be converted to JS
>  * * Make sure to use this Macro in an @implementation, not @interface
>  *
>  * The JS function will have the same name as the given Objective-C function, but with a "__" prefix.
>  * Make sure to add that function to the babel.config.js under reanimated's "globals" option, and add TypeScript type declarations.
>  */
> #define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
>                                                                                     \
> +(void)load                                                                         \
> {                                                                                   \
>   [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #frame_processor callback:^id(Frame* frame, NSArray<id>* args) { \
>     return frame_processor(frame, args);                                           \
>   }];                                                                               \
> }
> 
> 
> /**
>  * Same as VISION_EXPORT_FRAME_PROCESSOR, but uses __attribute__((constructor)) for
>  * registration. Useful for registering swift classes that forbids use of +(void)load.
>  */
> #define VISION_EXPORT_SWIFT_FRAME_PROCESSOR(name, objc_name) \
> objc_name : NSObject<FrameProcessorPluginBase>                                      \
> @end                                                                                \
>                                                                                     \
> @interface objc_name (FrameProcessorPlugin) <FrameProcessorPluginBase>                                         \
> @end                                                                                \
> @implementation objc_name (FrameProcessorPlugin)                                    \
>                                                                                     \
> __attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)() \
> {                                                                                   \
>   [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
>     return [objc_name callback:frame withArgs:args];                               \
>   }];                                                                               \
> }
> 
> #endif /* FrameProcessorPlugin_h */
> ```
> 
> If you have patch-package you can apply this:
> 
[... 45 lines cut ...]
@@ -24,6 +24,13 @@ extension AVCaptureColorSpace {
     case "srgb":
       self = .sRGB
       return
+    case "appleLog":
+        if #available(iOS 17, *) {
+          self = .appleLog
+        } else {
+          throw EnumParserError.unsupportedOS(supportedOnOS: "17")
+        }
+        return
     default:
       throw EnumParserError.invalidValue
     }
@@ -37,6 +44,8 @@ extension AVCaptureColorSpace {
       return "p3-d65"
     case .sRGB:
       return "srgb"
+    case .appleLog:
+        return "appleLog"
     default:
       fatalError("AVCaptureDevice.Position has unknown state.")
     }

```

ms314006, 2023-09-27:
> @Thormeard, it's final change, worked for me. But only to "react-native-vision-camera": "2.15.2", if I update "react-native-vision-camera" to 2.16.1 I got error(Frame processors not enabled) again, [...]

This patch seems to be working for me! thanks!

@marcshilling You can try this patch.

Thormeard, 2023-09-27:
> @Thormeard, it's final change, worked for me. But only to "react-native-vision-camera": "2.15.2", if I update "react-native-vision-camera" to 2.16.1 I got error(Frame processors not enabled) again, [...]

Can confirm, fixes issue for me BUT only on **2.15.2** (not on latest 2.16.1). Thanks!

Xcode 15, ios 17, iPhone X

vincicat, 2023-09-28:
> case .appleLog:
> +        return "appleLog"

I can confirm the FrameProccessor patch work on XCode 15 with iOS 16 + RN0.70 + react-native-vision-camera@2.15.6 + react-native-reanimated@2.17.0, android need further testing but it should work.

* it seems that v2.16 have some change on FrameProcessor so we may need to stick to v2.15.6 for a while ... @mrousavy or other contributor can make a patch on 2.15 branch to fix XCode 15 and iPhone 15 issues first.

here is the diff for quick check
https://github.com/mrousavy/react-native-vision-camera/compare/v2.15.6...v2.16.0

mrousavy, 2023-09-30:
Hey - yea please submit a PR instead of creating random patches. Then I can merge and backport it.

closed 2023-09-30 by mrousavy

Shubham0850, 2023-10-02:
I"m still getting the issue.

ENV:
` "react-native-vision-camera": "2.15.4",`
`  "react-native": "0.71.7",`
macOS: `14.0`
xCode: `15`
iPhone 13 ( iOS 17.1 Beta ) - Physical Device

https://github.com/mrousavy/react-native-vision-camera/issues/1307 : This does solved the initial crashing issue, but got new error.
[image]

Here is my FrameProcessorPlugin.h
```
//
//  FrameProcessorPlugin.h
//  VisionCamera
//
//  Created by Marc Rousavy on 01.05.21.
//  Copyright © 2021 mrousavy. All rights reserved.
//

#ifndef FrameProcessorPlugin_h
#define FrameProcessorPlugin_h

#import <Foundation/Foundation.h>
#import "FrameProcessorPluginRegistry.h"
#import "Frame.h"

@protocol FrameProcessorPluginBase
+ (id) callback:(Frame*)frame withArgs:(NSArray<id>*)args;
@end

#define VISION_CONCAT2(A, B) A##B
#define VISION_CONCAT(A, B) VISION_CONCAT2(A, B)

/**
 * Use this Macro to register the given function as a Frame Processor.
 * * Make sure the given function is a C-style function with the following signature: static inline id callback(Frame* frame, NSArray* args)
 * * Make sure the given function's name is unique across other frame processor plugins
 * * Make sure your frame processor returns a Value that can be converted to JS
 * * Make sure to use this Macro in an @implementation, not @interface
 *
 * The JS function will have the same name as the given Objective-C function, but with a "__" prefix.
 * Make sure to add that function to the babel.config.js under reanimated's "globals" option, and add TypeScript type declarations.
 */
#define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
                                                                                    \
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
{                                                                                   \
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #frame_processor callback:^id(Frame* frame, NSArray<id>* args) { \
    return frame_processor(frame, args);                                           \
  }];                                                                               \
}

/**
 * Same as VISION_EXPORT_FRAME_PROCESSOR, but uses __attribute__((constructor)) for
 * registration. Useful for registering swift classes that forbids use of +(void)load.
 */
#define VISION_EXPORT_SWIFT_FRAME_PROCESSOR(name, objc_name) \
objc_name : NSObject<FrameProcessorPluginBase>                                      \
@end                                                                                \
                                                                                    \
@interface objc_name (FrameProcessorPlugin) <FrameProcessorPluginBase>              \
@end                                                                                \
@implementation objc_name (FrameProcessorPlugin)                                    \
                                                                                    \
__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
{                                                                                   \
  [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
    return [objc_name callback:frame withArgs:args];                               \
  }];                                                                               \
}

#endif /* FrameProcessorPlugin_h */
```

vogdev, 2023-10-02:
if you upgrade the macOS Sonoma you can't use xcode v14 anymore and this problem happens only on xcode v15 with xcode tools v15 🤷‍♂️

ryadav1495, 2023-10-03:
Any fix for this issue. 
I am using 
```
Xcode 15 ,
 "react-native-vision-camera": "2.15.4",
 iOS :17
```
App is crashing on my simulator after launch.

Halimazeez, 2023-10-03:
Since updating my mac os and installing xcode 15, and i can't revert back to xcode 14 where everything worked.

The patch above that removes +load makes vision camera work and the frame processing still works, however the plugin i was using is now saying that it cannot find the property "__ScanOCR" so i can't use it.

Any idea why the native call isn't being registered anymore? It was working before updating.

versions:
   "react-native-reanimated": "2.17.0",
    "react-native-vision-camera": "2.15.6",
    "vision-camera-ocr": "^1.0.0"

dimisus, 2023-10-05:
> if you upgrade the macOS to Sonoma you can't use xcode v14 anymore and this problem happens only on xcode v15 with xcode tools v15 🤷‍♂️

I have an M2 on Sonoma running XCode 14.3.1

1. download 14.x: https://developer.apple.com/download/applications/
2. open dmg and paste+rename the app to your Applications folder (like XCode_14)
3. `sudo xcode-select -s /Applications/XCode_14.app/` to use the v14 app by default

oortiz-2020104, 2023-10-06:
> I have an M2 on Sonoma running XCode 14.3.1
>
> 1. download 14.x: https://developer.apple.com/download/applications/
> 2. open dmg and paste+rename the app to your Applications folder (like XCode_14)
> 3. [...]

I just did everything you mentioned but I can't open the new Xcode I installed, but I can't open the version I downloaded, when I run the command `sudo xcode-select -s /Applications/XCode_14.app/` and then `sudo xcode-select -p` It shows that I am using the new version that I installed `> /Applications/Xcode_14.app/Contents/Developer`

I don't know if this is right or if I have to do something additional.

shilpa12233, 2023-10-07:
My app runs fine on simulator, RN 0.68.2 but crashes in react core pod RCTCxxBridge file - at _reactInstance.reset(new Instance); when I try running on device. iOS - 17, iphone 15, xcode 15

Mukthayar1, 2023-10-07:
I just moved to react native camera kit.

rafaelmotta, 2023-10-09:
gettins this issue also on xcode 15 + iOS 17+ vision camera 3.3.1.

devtyty, 2023-10-10:
This patch worked for me!

react-native: 0.71.13
react-native-camera-vision: 2.15.4
xcode: 15
macos: 14.0

`diff --git a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h
index 9408774..2aa04dc 100644
--- a/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
+++ b/node_modules/react-native-vision-camera/ios/Frame Processor/FrameProcessorPlugin.h	
@@ -33,7 +33,7 @@
  */
 #define VISION_EXPORT_FRAME_PROCESSOR(frame_processor)                              \
                                                                                     \
-+(void)load                                                                         \
+__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
 {                                                                                   \
   [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #frame_processor callback:^id(Frame* frame, NSArray<id>* args) { \
     return frame_processor(frame, args);                                           \
@@ -53,7 +53,7 @@ objc_name : NSObject<FrameProcessorPluginBase>
 @end                                                                                \
 @implementation objc_name (FrameProcessorPlugin)                                    \
                                                                                     \
-+(void)load                                                                          \
+__attribute__((constructor)) static void VISION_CONCAT(initialize_, objc_name)()    \
 {                                                                                   \
   [FrameProcessorPluginRegistry addFrameProcessorPlugin:@"__" @ #name callback:^id(Frame* frame, NSArray<id>* args) {    \
     return [objc_name callback:frame withArgs:args];                               \
`

suwu150, 2023-10-12:
> PD-6078 fix: Swift class extensions and categories on Swift classes are not allowed to have +load methods team-olulo/react-native-vision-camera#4

Yes, I also encountered the problem in 3.3.0 but couldn't solve it。

https://github.com/mrousavy/react-native-vision-camera/issues/1987

vitas61, 2024-01-10:
iphone 15 pro, iphone 15 pro max
xcode 15
v 2.15.6
I fixed the crash problem on startup and the camera also works
patch file:

```
diff --git a/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift b/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift
index 13a403b..6d0ece2 100644
--- a/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift
+++ b/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift
@@ -24,6 +24,13 @@ extension AVCaptureColorSpace {
     case "srgb":
       self = .sRGB
       return
+    case "appleLog":
+      if #available(iOS 17, *) {
+         self = .appleLog
+      } else {
+        throw EnumParserError.unsupportedOS(supportedOnOS: "17")
+      }
+      return
     default:
       throw EnumParserError.invalidValue
     }
@@ -37,8 +44,10 @@ extension AVCaptureColorSpace {
       return "p3-d65"
     case .sRGB:
       return "srgb"
+    case .appleLog:
+      return "appleLog"
     default:
-      fatalError("AVCaptureDevice.Position has unknown state.")
+      return "unknown"
     }
   }
 }
```

hrvdm, 2024-01-11:
This isn't a solution but if you want a workaround that doesn't totally kill quality this works for me on iOS & Android:

`useCameraFormat(camera, [{videoResolution: 'max'}])`

and

```
cameraRef?.current?.takePhoto({
      qualityPrioritization: 'balanced', // Maybe doesnt matter, havent tested
});
```

mrousavy, 2024-01-11:
`useCameraFormat` is from V3, I don't think V2 has this?

hrvdm, 2024-01-11:
Yeah I'm silly, I got here through another issue- I probably should have properly read this properly before commenting- My issue was crash when taking photos with `iso: max`

mikoprus, 2024-02-29:
Experienced this issue for V3 on XCode 15.0, updating to 15.2 solved the problem