margelo/react-native-vision-camera #1804: Frame Processors are not enabled 🐛

Jordan-Terzian, 2023-09-15:
### What's happening?

I'm attempting to make a pose detection screen on my app using Google MLKit. However, when I navigate to the screen, I'm getting a "Frame Processors are not enabled!" error despite "pod install" saying "[VisionCamera] react-native-worklets-core found, Frame Processors enabled!". When I minimize this error, It runs the camera without poses shown in frame. 

### Reproduceable Code

```tsx
import React, {useEffect} from 'react';
import 'react-native-reanimated';
import {Dimensions, Platform, StyleSheet, useWindowDimensions} from 'react-native';
import {Camera, useFrameProcessor, useCameraDevices} from 'react-native-vision-camera';
import {useSharedValue, useAnimatedStyle} from 'react-native-reanimated';
import Animated from 'react-native-reanimated';
import Svg, {Line} from 'react-native-svg';

const AnimatedLine = Animated.createAnimatedComponent(Line);

const usePosition = (pose, valueName1, valueName2) => {
  return useAnimatedStyle(
    () => ({
      x1: pose.value[valueName1].x,
      y1: pose.value[valueName1].y,
      x2: pose.value[valueName2].x,
      y2: pose.value[valueName2].y,
    }),
    [pose],
  );
};

export function objectDetect(frame) {
  'worklet';
  return __poseDetection(frame);
}

const defaultPose = {
  leftShoulder: {x: 0, y: 0},
  rightShoulder: {x: 0, y: 0},
  leftElbow: {x: 0, y: 0},
  rightElbow: {x: 0, y: 0},
  leftWrist: {x: 0, y: 0},
  rightWrist: {x: 0, y: 0},
  leftHip: {x: 0, y: 0},
  rightHip: {x: 0, y: 0},
  leftKnee: {x: 0, y: 0},
  rightKnee: {x: 0, y: 0},
  leftAnkle: {x: 0, y: 0},
  rightAnkle: {x: 0, y: 0},
};

function PoseDetectionScreen() {
  const pose = useSharedValue(defaultPose);

  const leftWristToElbowPosition = usePosition(pose, 'leftWrist', 'leftElbow');
  const leftElbowToShoulderPosition = usePosition(pose, 'leftElbow', 'leftShoulder');
  const leftShoulderToHipPosition = usePosition(pose, 'leftShoulder', 'leftHip');
  const leftHipToKneePosition = usePosition(pose, 'leftHip', 'leftKnee');
  const leftKneeToAnklePosition = usePosition(pose, 'leftKnee', 'leftAnkle');

  const rightWristToElbowPosition = usePosition(pose, 'rightWrist', 'rightElbow');
  const rightElbowToShoulderPosition = usePosition(pose, 'rightElbow', 'rightShoulder');
  const rightShoulderToHipPosition = usePosition(pose, 'rightShoulder', 'rightHip');
  const rightHipToKneePosition = usePosition(pose, 'rightHip', 'rightKnee');
  const rightKneeToAnklePosition = usePosition(pose, 'rightKnee', 'rightAnkle');

  const shoulderToShoulderPosition = usePosition(pose, 'leftShoulder', 'rightShoulder');
  const hipToHipPosition = usePosition(pose, 'leftHip', 'rightHip');

  const dimensions = useWindowDimensions();

  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    const poseObject = objectDetect(frame);

    const xFactor = dimensions.width / frame.width;
    const yFactor = dimensions.height / frame.height;

    const poseCopy = {
      leftShoulder: {x: 0, y: 0},
      rightShoulder: {x: 0, y: 0},
      leftElbow: {x: 0, y: 0},
      rightElbow: {x: 0, y: 0},
      leftWrist: {x: 0, y: 0},
      rightWrist: {x: 0, y: 0},
      leftHip: {x: 0, y: 0},
      rightHip: {x: 0, y: 0},
      leftKnee: {x: 0, y: 0},
      rightKnee: {x: 0, y: 0},
      leftAnkle: {x: 0, y: 0},
      rightAnkle: {x: 0, y: 0},
    };

    Object.keys(poseObject).forEach(v => {
      poseCopy[v] = {
        x: poseObject[v].x * xFactor,
        y: poseObject[v].y * yFactor,
      };
    });

    pose.value = poseCopy;
  }, []);

  const devices = useCameraDevices();
  const device = devices.back;

  useEffect(() => {
    const checkPermissions = async () => {
      await Camera.requestCameraPermission();
    };
    checkPermissions();
  }, []);

  if (device == null) {
    return null;
  }

  return (
    <>
      <Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        device={device}
        isActive={true}
        orientation="portrait"
        frameProcessorFps={15}
      />
      <Svg
        height={Dimensions.get('window').height}
        width={Dimensions.get('window').width}
        style={styles.linesContainer}>
        <AnimatedLine animatedProps={leftWristToElbowPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={leftElbowToShoulderPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={leftShoulderToHipPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={leftHipToKneePosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={leftKneeToAnklePosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={rightWristToElbowPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={rightElbowToShoulderPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={rightShoulderToHipPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={rightHipToKneePosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={rightKneeToAnklePosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={shoulderToShoulderPosition} stroke="red" strokeWidth="2" />
        <AnimatedLine animatedProps={hipToHipPosition} stroke="red" strokeWidth="2" />
      </Svg>
    </>
  );
}

export default PoseDetectionScreen;

const styles = StyleSheet.create({
  linesContainer: {
    position: 'absolute',
    top: 0,
    left: 0,
    height: Dimensions.get('window').height,
    width: Dimensions.get('window').width,
  },
});
```

### Relevant log output

```shell
2023-09-15 10:59:52.538700+1000 Dappa_App[11417:2619724] [native] VisionCamera.didSetProps(_:): Updating 12 prop(s)...
2023-09-15 10:59:52.538854+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureCaptureSession(): Configuring Session...
2023-09-15 10:59:52.538878+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureCaptureSession(): Initializing Camera with device com.apple.avfoundation.avcapturedevice.built-in_video:6...
2023-09-15 10:59:52.538893+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureCaptureSession(): Adding Video input...
2023-09-15 10:59:52.539275+1000 Dappa_App[11417:2619954] [javascript] system/frame-processors-unavailable: Frame Processors are not enabled!, js engine: hermes
2023-09-15 10:59:52.546047+1000 Dappa_App[11417:2619724] [native] Running surface LogBox ({
    initialProps =     {
    };
    rootTag = 11;
})
2023-09-15 10:59:52.606605+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureCaptureSession(): Adding Video Data output...
2023-09-15 10:59:52.607155+1000 Dappa_App[11417:2619945] [native] VisionCamera.invokeOnInitialized(): Camera initialized!
2023-09-15 10:59:52.607223+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureCaptureSession(): Session successfully configured!
2023-09-15 10:59:52.607739+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureFormat(): Configuring Format...
2023-09-15 10:59:52.607748+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureDevice(): Configuring Device...
2023-09-15 10:59:52.607785+1000 Dappa_App[11417:2619945] [native] VisionCamera.configureDevice(): Device successfully configured!
2023-09-15 10:59:52.608085+1000 Dappa_App[11417:2619945] [native] VisionCamera.didSetProps(_:): Starting Session...
2023-09-15 10:59:53.112611+1000 Dappa_App[11417:2619945] [native] VisionCamera.didSetProps(_:): Started Session!
```

### Camera Device

```json
N/A
```

### Device

iPhone 12 (16.6.1)

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mendesbarreto, 2023-09-15:
Hello @Jordan-Terzian,

I encountered the same issue, but I managed to resolve it by starting the project from scratch. You can take a look at the working project [HERE](https://github.com/mendesbarreto/ml-mobile-camera-template/tree/master).

Feel free to run it on your end and check if it encounters the same error.

My logs:
[image]

The project prints the frame and the pixel buffer.

I hope this helps.

basukiwinoto, 2023-09-16:
I had a similar issue resolved after installing `react-native-worklets-core` and applying this patch:
```
diff --git a/node_modules/react-native-worklets-core/cpp/WKTJsRuntimeFactory.h b/node_modules/react-native-worklets-core/cpp/WKTJsRuntimeFactory.h
index 31c6c4b..d127093 100644
--- a/node_modules/react-native-worklets-core/cpp/WKTJsRuntimeFactory.h
+++ b/node_modules/react-native-worklets-core/cpp/WKTJsRuntimeFactory.h
@@ -12,7 +12,7 @@
 #include <hermes/hermes.h>
 #else
 // JSC
-#include <jsc/JSCRuntime.h>
+#include <React-jsc/JSCRuntime.h>
 #endif
 
 namespace RNWorklet {
```

closed 2023-09-17 by Jordan-Terzian