margelo/react-native-vision-camera #1805: while recording video flip camera access without stop the video. is this feature available🐛

rajyuva1000, 2023-09-15:
### What's happening?

i want to switch camera while recording, but i am facing issue. 
 Recording failed! if {"cause": {"message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "stacktrace": "com.mrousavy.camera.InactiveSourceError: [capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.

### Reproduceable Code

```tsx
await curCamera.resumeRecording(); this feature not working as expected
```

### Relevant log output

```shell
ERROR  Recording failed! if {"cause": {"message": "[capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
One case is that if camera is closed due to lifecycle stopped, the active recording will be finalized with this error, and the output will be generated, containing the frames produced before camera closing. Attempting to start a new recording will be finalized immediately if the source remains inactive and no output will be generated.", "stacktrace": "com.mrousavy.camera.InactiveSourceError: [capture/inactive-source] The recording failed because the source becomes inactive and stops sending frames.
```

### Camera Device

```json
const onStartRecording = useCallback(async () => {
  setStartPauseRecord(false);
  console.log(cameraPosition, 'cameraPosition');
  try {
    if (camera.current == null) {
      throw new Error('Camera ref is null!');
    }
    
    var curCamera = camera.current;

    if (StartPauseRecord) {
      if (!isRecording) { // Check if not already recording
        curCamera.startRecording({
          onRecordingError: error => {
            console.error('Recording failed! if', error);
            onStoppedRecording();
            setIsRecording(false); // Update the recording state
          },
          onRecordingFinished: video => {
            console.log('Recording successfully finished!', ` ${video.path}`);
            onMediaCaptured(video);
            onStoppedRecording();
            setIsRecording(false); // Update the recording state
          },
        });
        startRec = !startRec;
        setStartPauseRecord(false);
        setIsRecording(true); // Update the recording state
      }
    } else {
      // Stop the recording when the camera position changes
      onStopRecording();
      if (isRecording) { // Check if recording is active
        await curCamera.stopRecording(); // Stop the current recording
      }
      // Switch camera position
      setCameraPosition(prevPosition => (prevPosition === 'back' ? 'front' : 'back'));
      // Start a new recording with the updated camera position
      if (!isRecording) { // Check if not already recording
        curCamera.startRecording({
          onRecordingError: error => {
            console.error('Recording failed! else', error);
            onStoppedRecording();
            setIsRecording(false); // Update the recording state
          },
          onRecordingFinished: video => {
            console.log('Recording successfully finished!', ` ${video.path}`);
            onMediaCaptured(video);
            onStoppedRecording();
            setIsRecording(false); // Update the recording state
          },
        });
        setIsRecording(true); // Update the recording state
      }
    }
    startCounter();
  } catch (e) {
    console.error('Failed to start recording!', e, 'camera');
    // Handle the error here, e.g., by showing an error message to the user
  }
}, [onMediaCaptured, onStoppedRecording, startCounter, FlashControl, StartPauseRecord, cameraPosition, isRecording]);
```

### Device

android

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [ ] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-19:
Either you're lying about something or you have mixed versions. V3 does not have the `inactive-source` error in code at all anymore. In V3 you can flip the camera devices while recording, I explicitly implemented that feature since many people requested it.

closed 2023-09-19 by mrousavy

rajyuva1000, 2023-09-21:
now upgraded to v3 version(3.0.0 ). but getting error when i click start function. error  look like,

WARN  Camera is not ready yet. Wait for the onInitialized() callback

ombogdanJoinToIt, 2023-10-19:
ios 17 The Camera is not ready yet! Wait for the onInitialized() callback!