margelo/react-native-vision-camera #1806: 🐛 Video showing black screen in Android

sriumalladi, 2023-09-15:
### What's happening?

I wanted to record video its always showing black screen , if i set Camera photo={true} it displays both front and back camera properly. but if i set Camera video={true} ist always showing black screen.

[unknown/unknown: [unknown/unknown] No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) 

### Reproduceable Code

```tsx
const devices = useCameraDevices();
  const device = devices.back;
  const HEIGHT = Dimensions.get('window').height;
  const WIDTH = Dimensions.get('window').width;
  const onError = error => {
    console.error(error);
  };

React.useEffect(() => {
    if (device) {
      Camera.requestCameraPermission()
        .then(result => {
          console.log('requestCameraPermission: ', result);
        })
        .catch(err => {
          console.log('requestCameraPermission Err: ', err);
        });
      Camera.getMicrophonePermissionStatus()
        .then(result => {
          console.log('requestMicrophonePermission: ', result);
        })
        .catch(err => {
          console.log('requestMicrophonePermission Err: ', err);
        });
    }
  }, [device]);

 if (device == null) return <ActivityIndicator />;

return (
    <View style={{flex: 1}}>
      <Camera
        device={device}
        isActive={true}
        video={true}
        onError={onError}
        style={{position: 'absolute', width: WIDTH, height: HEIGHT}}
      />
    </View>
  );
```

### Relevant log output

```shell
09-15 16:24:33.242 26505 20131 E Ads     :      at java.lang.Thread.run(Thread.java:923)
09-15 16:24:33.246 26505 20261 E Ads     : Couldn't create RTB adapter :
09-15 16:24:33.246 26505 20261 E Ads     : android.os.RemoteException
09-15 16:24:33.246 26505 20261 E Ads     :      at com.google.android.gms.internal.ads.zzbxe.zzc(SourceFile:34)
09-15 16:24:33.246 26505 20261 E Ads     :      at com.google.android.gms.internal.ads.zzbxg.zzbQ(SourceFile:34)
09-15 16:24:33.246 26505 20261 E Ads     :      at com.google.android.gms.internal.ads.zzaok.onTransact(SourceFile:21)        
09-15 16:24:33.246 26505 20261 E Ads     :      at android.os.Binder.transact(Binder.java:1043)
09-15 16:24:33.246 26505 20261 E Ads     :      at m.aif.aZ(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:8)
09-15 16:24:33.246 26505 20261 E Ads     :      at com.google.android.gms.ads.internal.mediation.client.b.c(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:9)
09-15 16:24:33.246 26505 20261 E Ads     :      at com.google.android.gms.ads.nonagon.adapter.e.b(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:5)
09-15 16:24:33.246 26505 20261 E Ads     :      at com.google.android.gms.ads.nonagon.signals.ek.a(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:64)
09-15 16:24:33.246 26505 20261 E Ads     :      at m.byn.a(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:3)
09-15 16:24:33.246 26505 20261 E Ads     :      at m.bxw.run(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:21)
09-15 16:24:33.246 26505 20261 E Ads     :      at m.byp.run(:com.google.android.gms.policy_ads_fdr_dynamite@233012805@233012802057.555274289.555274289:5)
09-15 16:24:33.246 26505 20261 E Ads     :      at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
09-15 16:24:33.246 26505 20261 E Ads     :      at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
[... 350 lines cut ...]
### Camera Device

```json
"photoWidth": 1440, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 960, "photoWidth": 1280, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 720, "photoWidth": 1280, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 1080, "photoWidth": 1080, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 540, "photoWidth": 960, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 600, "photoWidth": 800, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 480, "photoWidth": 800, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 720, "photoWidth": 720, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 480, "photoWidth": 720, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 480, "photoWidth": 640, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 288, "photoWidth": 352, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 240, "photoWidth": 320, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 144, "photoWidth": 176, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 720, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 6000, "photoWidth": 8000, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "contrast-detection", "fieldOfView": 68.04700270068098, "maxFps": 30, "maxISO": 3200, "minFps": 1, "minISO": 50, "photoHeight": 3000, "photoWidth": 4000, "pixelFormats": [Array], "supportsPhotoHDR": false, "supportsVideoHDR": false, "videoHeight": 480, "videoStabilizationModes": [Array], "videoWidth": 720}, {"autoFocusSystem": "con [... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

Romick2005, 2023-09-19:
Works fine on latest react-native-vision-camera@2.16.1

yegorius, 2023-09-19:
Having the same issue on OnePlus 8T. Tried removing the lines suggested in #1801 to no success. The culprit is here:
```
Failed to configure session: No implementation found for com.facebook.jni.HybridData com.mrousavy.camera.core.VideoPipeline.initHybrid(int, int) (tried Java_com_mrousavy_camera_core_VideoPipeline_initHybrid and Java_com_mrousavy_camera_core_VideoPipeline_initHybrid__II)
```

yegorius, 2023-09-20:
I have identified the problem.
The implementation of `VideoPipeline.initHybrid(int, int)` is written in C++, thus `System.loadLibrary("VisionCamera")` must be called. But if `react-native-worklets-core` is not present `System.loadLibrary` won't be called.
The execution path is like this:
```
// FrameProcessorPlugins.ts
if (hasWorklets) {
    const result = CameraModule.installFrameProcessorBindings();

// CameraViewModule.kt
fun installFrameProcessorBindings(): Boolean {
    val proxy = VisionCameraProxy(reactApplicationContext)

// VisionCameraProxy.kt
class VisionCameraProxy(context: ReactApplicationContext) {
  init {
    System.loadLibrary("VisionCamera")
```
You can see here that everything depends on `hasWorklets` variable.
Anyway, video recording works if I add `react-native-worklets-core` to the project.

fontesrp, 2023-09-20:
Absolute savior, @yegorius. Thank you!

mrousavy, 2023-09-30:
Hey - this should be fixed with the latest update in VisionCamera V3 (🥳) - if not, please create a new issue.

If your issue has been fixed, consider [sponsoring me on GitHub](https://github.com/sponsors/mrousavy) to say thanks 💖

closed 2023-09-30 by mrousavy