margelo/react-native-vision-camera #1807: How can i implement QR code Scanner with vision camera❓

Akshaybagai52, 2023-09-16:
### Question

How can I implement QR code scanner with vision camera. I tried using vision-camera-code-scanner but it is not working whenever i install vision-camera-code-scanner library then my build is getting failed. Is there anyone who can help me to get this solution done

### What I tried

I also tried exploring react-native-camera but it is deprecated now

### VisionCamera Version

latest

### Additional information

- [ ] I am using Expo
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for similar questions [in the issues page](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22❓+question%22) as well as [in the discussions page](https://github.com/mrousavy/react-native-vision-camera/discussions) and found none.

imransilvake, 2023-09-17:
Same issue

xulihang, 2023-09-18:
vision-camera-code-scanner or vision-camera-dynamsoft-barcode-reader are not adapted for vision camera v3. You have to use vision camera v2.

jimitshah-dp, 2023-09-18:
If you go through this thread, @mrousavy Has said he will take this up in sometime, but this is going to be part of V3 in someform.
https://github.com/mrousavy/react-native-vision-camera/issues/1723#issue-1867951881

itzFaiz, 2023-09-18:
which react, react native, and reanimated versions are compatible with V2. I have been stuck on this issue from past 2-3 weeks and i haven't found any solution.

xulihang, 2023-09-18:
react native <= 0.71
reanimated v2

This should work with vision camera v2.

If vision camera v2 can support reanimated v3, it should work on react native 0.72+. Check this pr: https://github.com/mrousavy/react-native-vision-camera/pull/1759

imransilvake, 2023-09-18:
I only needed QR code scan and I found another lib from Tesla which works really great for me.
https://github.com/teslamotors/react-native-camera-kit

Akshaybagai52, 2023-09-19:
@imransilvake Thanks it also worked for me in 0.72.4

mrousavy, 2023-09-30:
Hey - check out the Frame Processor Plugins docs.

closed 2023-09-30 by mrousavy

Productivix, 2023-10-05:
hi, may be better to complete the online doc on [code-scanning](https://www.react-native-vision-camera.com/docs/guides/code-scanning )?

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.

mrousavy, 2023-10-06:
> hi, may be better to complete the online doc on [code-scanning](https://www.react-native-vision-camera.com/docs/guides/code-scanning)?

? 

those docs are complete. even with extra info there.

pavalep, 2023-10-27:
appreciating the awesome work by react-native-vision-camera developers and contributors.
https://github.com/teslamotors/react-native-camera-kit for all those who are struggling with qrcode can use this by tesla

mrousavy, 2023-10-31:
@pavalep what? VisionCamera has a QR / Barcode scanner.

msaidm, 2024-02-07:
> Hey!
>
> JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍 Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀
>
> If you [...]

Thank you!

Do you know how to stop the camera after it scans and it finds the correct condition?

onst handleCodeScanned = useCallback(async (codes) => {
    const scannedCode = codes[0].value.trim();
    console.log(`Scanned ${scannedCode}`);
    if (!scannedCorrectCode && appointmentId === scannedCode) {
      setIsActive(false);
      setIsModalVisible(false);
      console.log(`Scanned ${scannedCode} (correct appointment ID)`);
      await handleUserScannedTheCorrectAppointmentId();
      setScannedCorrectCode(true); // Update state to prevent further executions
    } else {
      handleUserScannedTheWrongAppointmentId();
    }
  }, []); // Dependency array ensures that the callback is memoized properly

  const codeScanner = useCodeScanner({
    codeTypes: ['qr', 'ean-13'],
    onCodeScanned: handleCodeScanned
  });