margelo/react-native-vision-camera #1809: ✨Support focusing manually by specifying focus depth

bakoe, 2023-09-17:
### What feature or enhancement are you suggesting?

Currently, VisionCamera only supports **focusing manually** by specifying **coordinates of a subject** where the camera should focus to (using the `<Camera>`’s [focus method](https://www.react-native-vision-camera.com/docs/api/classes/Camera#focus)) 

In addition, both Android and iOS allow focusing manually by **specifying a fixed depth** (i.e., a distance from the camera).

* iOS: [`setFocusModeLocked(lensPosition)`](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1624617-setfocusmodelocked)
* Android: [`LENS_FOCUS_DISTANCE`](https://developer.android.com/reference/kotlin/android/hardware/camera2/CaptureRequest#lens_focus_distance)

It seems like RNCamera used a prop called `focusDepth` for this (instead of, e.g., exposing a method like `focusToDepth(depth)`). 

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

As far as I know, there are no workarounds available to focus by a depth/distance (instead of focusing on a subject). 

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-09-20:
Hey! Good feature request 👍

kinghoddy, 2025-06-21:
Any update on this. ?