margelo/react-native-vision-camera #1814: V2.16.1 and react-native 0.72.4 android Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'. 🔧

webhibegit, 2023-09-19:
### How were you trying to build the app?

npx react-native run-android

### Full build logs

```tsx
FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-vision-camera:buildCMakeDebug[arm64-v8a]'.
> com.android.ide.common.process.ProcessException: ninja: Entering directory `/home/ubuntu/Documents/ReactNative/CameraTest/node_modules/react-native-vision-camera/android/.cxx/Debug/3r1h2y4e/arm64-v8a'
  [1/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o
  [2/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o
  [3/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o
  [4/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o
  [5/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o
  [6/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o
  [7/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o
  [8/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o
  [9/10] Building CXX object CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o
  [10/10] Linking CXX shared library ../../../../build/intermediates/cxx/Debug/3r1h2y4e/obj/arm64-v8a/libVisionCamera.so
  FAILED: ../../../../build/intermediates/cxx/Debug/3r1h2y4e/obj/arm64-v8a/libVisionCamera.so 
  : && /home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/bin/clang++ --target=aarch64-none-linux-android21 --sysroot=/home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot -fPIC -fexceptions -frtti -std=c++1y -DONANDROID -DFOR_HERMES=1 -g  -fno-limit-debug-info  -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Qunused-arguments -Wl,--no-undefined -shared -Wl,-soname,libVisionCamera.so -o ../../../../build/intermediates/cxx/Debug/3r1h2y4e/obj/arm64-v8a/libVisionCamera.so CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCamera.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/JSIJNIConversion.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameHostObject.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/CameraView.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/VisionCameraScheduler.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JFrameProcessorPlugin.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JImageProxy.cpp.o CMakeFiles/VisionCamera.dir/src/main/cpp/java-bindings/JHashMap.cpp.o  /home/ubuntu/.gradle/caches/transforms-3/0eba9a168eb8819dccb7829f2ba40fb2/transformed/jetified-hermes-android-0.72.4-debug/prefab/modules/libhermes/libs/android.arm64-v8a/libhermes.so  /home/ubuntu/.gradle/caches/transforms-3/6fe00d6361b0f0c55549aec89a7f6742/transformed/jetified-react-android-0.72.4-debug/prefab/modules/folly_runtime/libs/android.arm64-v8a/libfolly_runtime.so  /home/ubuntu/.gradle/caches/transforms-3/6fe00d6361b0f0c55549aec89a7f6742/transformed/jetified-react-android-0.72.4-debug/prefab/modules/glog/libs/android.arm64-v8a/libglog.so  /home/ubuntu/.gradle/caches/transforms-3/6fe00d6361b0f0c55549aec89a7f6742/transformed/jetified-react-android-0.72.4-debug/prefab/modules/jsi/libs/android.arm64-v8a/libjsi.so  /home/ubuntu/.gradle/caches/transforms-3/6fe00d6361b0f0c55549aec89a7f6742/transformed/jetified-react-android-0.72.4-debug/prefab/modules/reactnativejni/libs/android.arm64-v8a/libreactnativejni.so  /home/ubuntu/.gradle/caches/transforms-3/34c2120d0d3cd8dc97826fc02c9e8fc4/transformed/jetified-fbjni-0.3.0/prefab/modules/fbjni/libs/android.arm64-v8a/libfbjni.so  /home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/lib/aarch64-linux-android/21/liblog.so  -landroid   -latomic -lm && :
  ld: error: undefined symbol: reanimated::extractWorkletRuntime(facebook::jsi::Runtime&, facebook::jsi::Value const&)
  >>> referenced by FrameProcessorRuntimeManager.cpp:105 (/home/ubuntu/Documents/ReactNative/CameraTest/node_modules/react-native-vision-camera/android/src/main/cpp/FrameProcessorRuntimeManager.cpp:105)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o:(vision::FrameProcessorRuntimeManager::setFrameProcessor(facebook::jsi::Runtime&, int, facebook::jsi::Value const&, facebook::jsi::Value const&))
  
  ld: error: undefined symbol: reanimated::extractShareableOrThrow(facebook::jsi::Runtime&, facebook::jsi::Value const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&)
  >>> referenced by Shareables.h:163 (/home/ubuntu/Documents/ReactNative/CameraTest/node_modules/react-native-reanimated/android/build/prefab-headers/reanimated/Shareables.h:163)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o:(std::__ndk1::shared_ptr<reanimated::ShareableWorklet> reanimated::extractShareableOrThrow<reanimated::ShareableWorklet>(facebook::jsi::Runtime&, facebook::jsi::Value const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const&))
  
  ld: error: undefined symbol: typeinfo for reanimated::Shareable
  >>> referenced by memory:0 (/home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o:(std::__ndk1::enable_if<(!(is_array<reanimated::ShareableWorklet>::value)) && (!(is_array<reanimated::Shareable>::value)), std::__ndk1::shared_ptr<reanimated::ShareableWorklet> >::type std::__ndk1::dynamic_pointer_cast<reanimated::ShareableWorklet, reanimated::Shareable>(std::__ndk1::shared_ptr<reanimated::Shareable> const&))
  >>> referenced by memory:0 (/home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o:(std::__ndk1::enable_if<(!(is_array<reanimated::ShareableWorklet>::value)) && (!(is_array<reanimated::Shareable>::value)), std::__ndk1::shared_ptr<reanimated::ShareableWorklet> >::type std::__ndk1::dynamic_pointer_cast<reanimated::ShareableWorklet, reanimated::Shareable>(std::__ndk1::shared_ptr<reanimated::Shareable> const&))
  
  ld: error: undefined symbol: typeinfo for reanimated::ShareableWorklet
  >>> referenced by memory:0 (/home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o:(std::__ndk1::enable_if<(!(is_array<reanimated::ShareableWorklet>::value)) && (!(is_array<reanimated::Shareable>::value)), std::__ndk1::shared_ptr<reanimated::ShareableWorklet> >::type std::__ndk1::dynamic_pointer_cast<reanimated::ShareableWorklet, reanimated::Shareable>(std::__ndk1::shared_ptr<reanimated::Shareable> const&))
  >>> referenced by memory:0 (/home/ubuntu/Android/Sdk/ndk/23.1.7779620/toolchains/llvm/prebuilt/linux-x86_64/sysroot/usr/include/c++/v1/memory:0)
  >>>               CMakeFiles/VisionCamera.dir/src/main/cpp/FrameProcessorRuntimeManager.cpp.o:(std::__ndk1::enable_if<(!(is_array<reanimated::ShareableWorklet>::value)) && (!(is_array<reanimated::Shareable>::value)), std::__ndk1::shared_ptr<reanimated::ShareableWorklet> >::type std::__ndk1::dynamic_pointer_cast<reanimated::ShareableWorklet, reanimated::Shareable>(std::__ndk1::shared_ptr<reanimated::Shareable> const&))
  clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.
  
  C++ build system [build] failed while executing:
      /home/ubuntu/Android/Sdk/cmake/3.22.1/bin/ninja \
        -C \
        /home/ubuntu/Documents/ReactNative/CameraTest/node_modules/react-native-vision-camera/android/.cxx/Debug/3r1h2y4e/arm64-v8a \
        VisionCamera
    from /home/ubuntu/Documents/ReactNative/CameraTest/node_modules/react-native-vision-camera/android

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 13s
```

[... cut ...]
### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

tezqa, 2023-09-19:
I'm just stuck on the same error right now :(

Bayartogtokh, 2023-09-19:
downgrade version to 2.16.1 i have a same version as 0.72.4

tezqa, 2023-09-19:
> downgrade version to 2.16.1 i have a same version as 0.72.4

2.16.1?
@webhibegit is already using this version, and I'm using the 2.16.0.

DavideSegullo, 2023-09-19:
I have the same error too:

- react-native-vision-camera 2.16.1
- react-native 0.72.3
- react-native-reanimated 3.5.2

jonixj, 2023-09-19:
Same here on 3.0

WanderingBit, 2023-09-19:
I tried building the project with both CMake versions 3.22.1 and 3.18.1 individually, but the problem persisted. The error did change with version 3.22.1. However, I was able to build using CMake version 3.18 when the frame processor was disabled.

webhibegit, 2023-09-20:
Can we use react-native-worklets-core in V2.16.1? In v3 [this](https://github.com/mrousavy/react-native-vision-camera/issues/1775#issuecomment-1710035361) solution is working. So could we use react-native-worklets-core in v2.16.1 also?

tezqa, 2023-09-20:
Surprisingly, this morning, no more errors with v2.6.1. I couldn't use v3 since I also use `vision-camera-ocr` which is not compatible with `react-native-camera-vision@3.0.0`.

Not sure about what solved the issue, since I've done a lot of things: React native update to v0.72.4, Android studio update to Giraffe (I can't believe I'm writing such a version name) + Invalidate cache + clean + rebuild project. But now it works. I still have to test my entire app to check if everything works correctly. I'll keep you updated.

JmePaz, 2023-09-21:
I solved this issue by changing the latest version of react-native-reanimated to just `"3.5.0"`. 

My full dependencies:
`
"dependencies": {

-     "react": "18.2.0",
-     "react-native": "0.72.4",
-     "react-native-vision-camera": "^2.16.1",
-     "react-native-worklets-core": "^0.2.0",
-     "string-hash-64": "^1.0.3",
-     "react-native-reanimated": "3.5.0"

  }
`

tezqa, 2023-09-21:
I confirm I'm using `react-native-reanimated@3.5.0` but it was already the case when I had issues.

DavideSegullo, 2023-09-21:
> I have the same error too:
> 
> * react-native-vision-camera 2.16.1
> * react-native 0.72.3
> * react-native-reanimated 3.5.2

It works fine on react-native-reanimated 3.5.1

stemyke, 2023-09-21:
@mrousavy Sorry for pinging you here, I tried the versions today whet @DavideSegullo suggested, but I have an other problem. It says:

> [CXX1214] D:\Work\rnea\node_modules\react-native-vision-camera\android\CMakeLists.txt debug|x86 : User has minSdkVersion 21 but library was built for 26 [//react-native-reanimated/reanimated]
Affected Modules: react-native-vision-camera

Do you have any idea what could be the problem? I checked everything where minSdk can be set and I am sure I set 26 everywhere possible. After cleaning rebuilding the project it still does not work.

DavideSegullo, 2023-09-21:
> @mrousavy Sorry for pinging you here, I tried the versions today whet @DavideSegullo suggested, but I have an other problem. It says:
>
>
> Do you have any idea what could be the problem? I checked [...]

can you reproduce it in a repo?

Skizu, 2023-09-21:
> @mrousavy Sorry for pinging you here, I tried the versions today whet @DavideSegullo suggested, but I have an other problem. It says:
>
>
> Do you have any idea what could be the problem? I checked [...]

This is a result of `react-native-vision-camera` having a hard reference to `minSdkVersion 21` instead of using something like `minSdkVersion getExtOrIntegerDefault('minSdkVersion')` akin to `react-native-reanimated`. As such I assume your applications `minSdkVersion` is 26.

stemyke, 2023-09-21:
@Skizu Yes my minSdkVersion is set to 26. Can I change it somehow for me in vision-camera code?

Skizu, 2023-09-21:
You can make a patch for `react-native-vision-camera`'s `build.gradle` where you replace the line `minSdkVersion 21` with `minSdkVersion getExtOrIntegerDefault('minSdkVersion')`. After this your project, `react-native-vision-camera`, and `react-native-reanimated` will use the same versions.

Currently `react-native-vision-camera` does not have a fix for this or for building for anything except hermes.

stemyke, 2023-09-21:
@Skizu Thank you very much it is finally building with this patch 🙂

stemyke, 2023-09-21:
@Skizu It looks like reanimated team released again a new version that you are using a few hours ago. It seems it again has the original problem by not finding the extractWorklets method, so I stick my package.json to "react-native-reanimated": "3.5.3". I am using Hermes by the way yes.

WanderingBit, 2023-09-21:
> @Skizu For now I am stuck again with an other problem. If I attach a frame processor to my camera then the app is frozen complately. Even if it just contains a simple console.log
> 
> ```js
> const frameProcessor = useFrameProcessor(
>         frame => {
>             'worklet';
>             console.log(frame.toString());
>         },
>         [],
>     );
>     
> ```

Of course, you shouldn't parse the frame as a string at the maximum refresh rate. That's completely normal.

stemyke, 2023-09-21:
@WanderingBit Why would it be normal? The toString() method generally just printing out the frame's width and height. Even if I set frameProcessorFps to 1 which is by Marcs saying is the default value actually it still does not work.

JmePaz, 2023-09-21:
My solution did not unfortunately work. I am confused. Because sometimes it build successful unexpectedly (with frame processor enabled), and most of the time it does not work and produced this kind of error. TBH I do not know the reason why this happen. tried a bunch of things, hopefully this will be resolved soon.

rizshivalli, 2023-09-21:
I am also facing the same issue, Sometimes it builds but after `./gradlew clean` it gives the `:react-native-vision-camera:buildCMakeDebug[arm64-v8a]` error and does not build. hopefully this will be resolved soon

KarlosQ, 2023-09-22:
Create that patch and use patch-package:

```
diff --git a/node_modules/react-native-vision-camera/android/build.gradle b/node_modules/react-native-vision-camera/android/build.gradle
index d308e15..e1d9dd9 100644
--- a/node_modules/react-native-vision-camera/android/build.gradle
+++ b/node_modules/react-native-vision-camera/android/build.gradle
@@ -135,7 +135,7 @@ android {
   }
 
   defaultConfig {
-    minSdkVersion 21
+    minSdkVersion getExtOrIntegerDefault('minSdkVersion')
     targetSdkVersion getExtOrIntegerDefault('targetSdkVersion')
 
     if (ENABLE_FRAME_PROCESSORS) {
``` 

I will create a PR for fix this

JmePaz, 2023-09-22:
Hello @KarlosQ I am new to this, how do you create a patch and use that patch package?

metrix-hu, 2023-09-22:
@JmePaz You just have to create a patches folder copy the mentioned contents in it to a file named: react-native-vision-camera+2.16.1

[react-native-vision-camera+2.16.1.patch](https://github.com/mrousavy/react-native-vision-camera/files/12699528/react-native-vision-camera%2B2.16.1.patch)

Install patch-package as a dev dependency with yarn or npm whichever one you use.

Add this one to your package.json inside the scripts object:

"postinstall": "patch-package"

Run npm install or yarn install, and then the patch will be applied.

JmePaz, 2023-09-22:
Unfortunately it does not works for me :( .

metrix-hu, 2023-09-22:
@JmePaz If you are experiencing the original error, not what @stemyke mentioned, then this patch would not help you. We are still facing the original error too after cleaning the project. We always change the reanimated version to something else in package.json do an install again, and hit play again in Android Studio. This way the error usually goes away but not always unfortunately. Probable some path is missing from CMakeLists in VisionCamera or could not be resolved automatically in certain scenarios.

JmePaz, 2023-09-22:
@metrix-hu True, Unfortunately I am trying everything I can HUHUHHUHH straight for two days but nothing works. It looks like I had to give up using this module/package for now. Any recommendation for alternative package/module. I am building a qr scanner app. (Any recommendation would be appreciated thank you).

metrix-hu, 2023-09-22:
@JmePaz If you are not upgraded yet to latest version of "react-native" 0.72.x, then I might have a solution for you. There were some discussions where I had specific versions for vision-camera and vision-camera-code-scanner and reanimated v2 that worked perfectly for me but unfortunately I cant find it right now. There is an other discussion going around qr code scanning and somebody had an alternative: https://github.com/mrousavy/react-native-vision-camera/issues/1807#issuecomment-1722861098

stealkiller06, 2023-09-22:
For me It's working on IOS and Android using this:
```
   "react-native": "0.72.4",
  "react-native-reanimated": "3.5.2",
   "react-native-vision-camera": "2.16.1",
   "vision-camera-code-scanner": "^0.2.0",
```

JmePaz, 2023-09-22:
@stealkiller06  May I ask if After cleaning the gradle, does it still work?

rizshivalli, 2023-09-22:
For me, It is working with below versions

` "react-native-vision-camera": "2.15.4",
    "react-native-reanimated": "3.5.0",
    "react-native": "0.72.4",
`

also i had to add `kotlinVersion = "1.6.20"` and  `minSdkVersion = 26` to android/build.gradle

FouadMagdy01, 2023-09-24:
i was facing the exact error and these steps helped me getting a successful build, i don't know  if it can help you also, but you can try

check this answer in stackoverflow  https://stackoverflow.com/a/35272475 

in the gradle version, choose version 8.1.1

again, this is not guarnteed to be the reason for this error, but you can try it

Skizu, 2023-09-28:
> For me It's working on IOS and Android using this:
> 
> ```
>    "react-native": "0.72.4",
>    "react-native-reanimated": "3.5.2",
>    "react-native-vision-camera": "2.16.1",
>    "vision-camera-code-scanner": "^0.2.0",
> ```

@stealkiller06 This is using Hermes correct?

mrousavy, 2023-09-30:
Hey! I think this should be fixed in the latest release, no?

closed 2023-09-30 by mrousavy

Skizu, 2023-10-01:
This is still an active issue. Using react native 0.72 with JSC does not compile.Sent from my iPhone

mmo-uy, 2023-10-02:
> For me It's working on IOS and Android using this:
> 
> ```
>    "react-native": "0.72.4",
>   "react-native-reanimated": "3.5.2",
>    "react-native-vision-camera": "2.16.1",
>    "vision-camera-code-scanner": "^0.2.0",
> ```
As @stealkiller06 says [here](https://github.com/mrousavy/react-native-vision-camera/issues/1814#issuecomment-1731613042) I confirm even using "react-native": "0.72.5", this solution works (tested in Android only)

Skizu, 2023-10-02:
I can confirm it does **not** work with a `minSdkVersion 24` and using `hermesEnabled=false` with `newArchEnabled=false`

> As @stealkiller06 says [here](https://github.com/mrousavy/react-native-vision-camera/issues/1814#issuecomment-1731613042) I confirm even using "react-native": "0.72.5", this solution works (tested [...]

pepperav, 2023-10-02:
I have same issue after updating react-native-vision-camera from 3.1.0 to 3.2.2.

**package.json**
```
"dependencies": {
    "@types/react": "~18.2.14",
    "expo": "~49.0.12",
    "expo-dev-client": "~2.4.11",
    "expo-splash-screen": "~0.20.5",
    "expo-status-bar": "~1.6.0",
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-vision-camera": "^3.1.0",
    "react-native-worklets-core": "^0.2.0",
    "typescript": "^5.1.3"
  }
```

iliapnmrv, 2023-10-03:
> For me It's working on IOS and Android using this:
> 
> ```
>    "react-native": "0.72.4",
>   "react-native-reanimated": "3.5.2",
>    "react-native-vision-camera": "2.16.1",
>    "vision-camera-code-scanner": "^0.2.0",
> ```

I can confirm that it works but after `./gradlew clean` issue persists. Any solutions?

RNVC v3 does not work with `vision-camera-code-scanner v2`

denisbevilacqua, 2023-10-03:
This issue was resolved for me by deleting the `.gradle` folder from `<NameOfProject>/android` and then running it again with `yarn android`. Could you please try doing the same, @iliapnmrv ?

iliapnmrv, 2023-10-03:
@denisbevilacqua Thank you so much, you saved time! `.gradle` folder was the reason

JmePaz, 2023-11-03:
@denisbevilacqua Thank you! sorry for the late reply

pasquale95, 2023-11-09:
Hi everyone,
does anyone know what is the reason of this weird behaviour? As soon as I run `./gradlew clean`, `./gradlew assembleDebug` stops working.
I am using the following setup:
```
"react-native": "0.72.6",
"react-native-reanimated": "3.5.4",
"react-native-vision-camera": "2.16.5",
"vision-camera-code-scanner": "^0.2.0",
```
To solve the issue I must delete `.gradle` and rerun `./gradlew assembleDebug` command.
I must add that the same exact procedure must be done also to run `./gradlew assembleRelease` successfully.

LuizCristino, 2023-11-10:
> I can confirm that it works but after `./gradlew clean` issue persists. Any solutions?
>
> RNVC v3 does not work with `vision-camera-code-scanner v2`

If you upgrade to RNVC v3 you don't need `vision-camera-code-scanner` no more. You can follow this guide on their own docs. Also don't forget to install the worklets (this package [react-native-worklets-core](https://github.com/margelo/react-native-worklets-core)).
https://react-native-vision-camera.com/docs/guides/code-scanning

mrousavy, 2023-11-20:
You don't need Worklets for the simple CodeScanning.

AamenaSaiyed, 2024-01-19:
Hey I am facing that header issue but not find any issue can anyone help me over it??
[image]

[image]

[image]

This is from android studio build log box

Pablo-DiazTassara, 2024-01-31:
Hi, what worked for me was updating react-native-worklets-core to 0.3.0.
Delete the node_module folder, the yarn.lock and install everything again. Then I cleared metro cache and run gradle clean.

"react-native-vision-camera": "^3.8.2",
"react-native-worklets-core": "^0.3.0",
"react": "18.2.0",
"react-native": "^0.72.0",

buildToolsVersion="33.0.0"
minSdkVersion = 26
compileSdkVersion = 33
targetSdkVersion = 34
ndkVersion="20.1.5948944"

mrousavy, 2024-01-31:
Great to hear! 👍

facucalle92, 2024-01-31:
Maybe for someone who is not being able to solve it, you can try reinstalling the NDK from Android Studio.
I got the solution for a similar issue here: https://github.com/software-mansion/react-native-reanimated/issues/4241#issuecomment-1590742192

NoahSimajji, 2024-04-30:
Hi there, any solution for this? So i have to remove the .gradle every time after i ./gradlew clean?

francesco-clementi-92, 2024-06-21:
I have the same problem and deleting the .gradle fix it, but I need to do it everytime. How could help also on debugging the problem, but I don't even know where to start. Any help is appreciated.
This is my setup:
```
"react-native-vision-camera": "4.3.2",
"react-native-worklets-core": "^1.3.3",
"vision-camera-ocr-plugin": "^3.0.9"
```

jenskuhrjorgensen, 2024-08-09:
Whenever this error pops up, for me the fix so far is to delete and reinstall `react-native-reanimated` i.e.
```
rm -rf node_modules/react-native-reanimated && npm install
```