margelo/react-native-vision-camera #1815: ultra-wide-angle cameras not detected on Android OnePlus phones

mdottavi, 2023-09-19:
### What's happening?

Hi, we're integrating react-native-vision-camera in our project.
We tested the example app on several Android phones and found the following issues:

- OnePlus 10 Pro  & OnePlus Nord2 (Android 13):
  example app starts but shows only 1 Front Cam + 1 Back wide-angle camera
  but the phone includes 3 Back Cameras (including 1 ultra-wide-angle- camera 0.6 zoom)

We added in the example app code the dump of Camera.getAvailableCamerDevices and get only the following output:
 LOG  wide-angle-camera name:BACK (0) position:back flash:true Torch:true minZoom:1 maxZoom:10 neutralZoom:1
 LOG  wide-angle-camera name:FRONT (1) position:front flash:false Torch:false minZoom:1 maxZoom:4 neutralZoom:1
 

### Reproduceable Code

```tsx
we added to the CameraPage.tsx the following code:

  useEffect(()=>{ 
    async function listCameras() {
      const TestDevices=await Camera.getAvailableCameraDevices();
      TestDevices.forEach (d => {
        console.log(d.devices + " name:"+ d.name+ " position:"+ d.position +" flash:" + d.hasFlash + " Torch:" + d.hasTorch + " minZoom:" + d.minZoom + " maxZoom:" + d.maxZoom + " neutralZoom:" + d.neutralZoom);
      });
    }
    listCameras();
  
  }, []);
```

### Relevant log output

```shell
LOG  Running "VisionCameraExample" with {"rootTag":11}
 LOG  Re-rendering Navigator. Camera: undefined | Microphone: undefined
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Re-rendering Navigator. Camera: granted | Microphone: undefined
 LOG  Re-rendering Navigator. Camera: granted | Microphone: granted
 LOG  re-rendering camera page without active camera
 LOG  re-rendering camera page without active camera
 LOG  Running "VisionCameraExample" with {"rootTag":11}
 LOG  Re-rendering Navigator. Camera: undefined | Microphone: undefined
 LOG  Loading react-native-worklets-core...
 LOG  Worklets loaded successfully
 LOG  Re-rendering Navigator. Camera: denied | Microphone: undefined
 LOG  Re-rendering camera page with active camera. Device: "BACK (0)" (4096x2304 photo / 3840x2176 video @ 30fps)
 LOG  wide-angle-camera name:BACK (0) position:back flash:true Torch:true minZoom:1 maxZoom:10 neutralZoom:1
 LOG  wide-angle-camera name:FRONT (1) position:front flash:false Torch:false minZoom:1 maxZoom:4 neutralZoom:1
 LOG  Re-rendering Navigator. Camera: denied | Microphone: denied
 LOG  Camera initialized!
 LOG  Re-rendering camera page with active camera. Device: "BACK (0)" (4096x2304 photo / 3840x2176 video @ 30fps)
 LOG  Re-rendering camera page with inactive camera. Device: "BACK (0)" (4096x2304 photo / 3840x2176 video @ 30fps)
```

### Camera Device

```json
LOG  wide-angle-camera name:FRONT (1) position:front flash:false Torch:false minZoom:1 maxZoom:4 neutralZoom:1
 LOG  {
  "hardwareLevel": "level-3",
  "maxZoom": 10,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-right",
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "hasTorch": true,
  "devices": [
    "wide-angle-camera"
  ],
  "hasFlash": true,
  "name": "BACK (0)",
  "position": "back",
  "id": "0"
}
 LOG  wide-angle-camera name:FRONT (1) position:front flash:false Torch:false minZoom:1 maxZoom:4 neutralZoom:1
 LOG  {
  "hardwareLevel": "full",
  "maxZoom": 4,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "sensorOrientation": "landscape-left",
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "hasTorch": false,
  "devices": [
    "wide-angle-camera"
  ],
  "hasFlash": false,
  "name": "FRONT (1)",
  "position": "front",
  "id": "1"
}
```

### Device

OnePlus 10 Pro and OnePlus Nord2

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

- [X] I can reproduce the issue in the VisionCamera Example app.

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-19:
Hey!

To me this sounds like Android just doesn't provide those Cameras to third party apps. Classic Android.

This is the line where I get all Camera devices on the phone. This is the same code you would use in a native Android app. So if the devices are not part of the `cameraIdList`, you can't use them in your app - not in a native Android app, not in VisionCamera.

You can check if that's the case by putting a break-point in this line and seeing what items are part of the cameraIdList:

https://github.com/mrousavy/react-native-vision-camera/blob/18c7034b5006986a7df6da9bf799dcc745244932/package/android/src/main/java/com/mrousavy/camera/CameraViewModule.kt#L147

And you can also **double-check** this by running the official native Android Camera Example Apps by Google/Android on your phone and try to find if the camera devices exist on your phone: https://github.com/android/camera-samples

Again, if my suspicion is true, then there's nothing I can do. This is Android's/One-Plus's fault.

mrousavy, 2023-09-19:
Also related; https://community.oneplus.com/thread/1032038

[image]

closed 2023-09-27 by mrousavy