margelo/react-native-vision-camera #1818: ✨Please enable the snippet in build.gradle again.

algopert, 2023-09-19:
### What feature or enhancement are you suggesting?

Hi
Hope you are well.

Today I faced an issue with **build.gradle**.
It was because you disabled the snippet related to `ENABLE_FRAM_RPOCESSORS `

```
// plugin.js file only exists since REA v2.
// def hasReanimated2 = file("${nodeModules}/react-native-reanimated/plugin.js").exists()
// def disableFrameProcessors = rootProject.ext.has("disableFrameProcessors") ? rootProject.ext.get("disableFrameProcessors").asBoolean() : false
// def ENABLE_FRAME_PROCESSORS = hasReanimated2 && !disableFrameProcessors
def ENABLE_FRAME_PROCESSORS = true
```

Please restore to original like below.
```
// plugin.js file only exists since REA v2.
def hasReanimated2 = file("${nodeModules}/react-native-reanimated/plugin.js").exists()
def disableFrameProcessors = rootProject.ext.has("disableFrameProcessors") ? rootProject.ext.get("disableFrameProcessors").asBoolean() : false
def ENABLE_FRAME_PROCESSORS = hasReanimated2 && !disableFrameProcessors
```

Thanks.

### What Platforms whould this feature/enhancement affect?

Android

### Alternatives/Workarounds

I am currently developing the capture app.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

mrousavy, 2023-09-20:
Hi. Please send a PR with the changes that targets the `v2` branch and I can merge that.

mrousavy, 2023-09-30:
Hey, this should be fixed in latest

closed 2023-09-30 by mrousavy