margelo/react-native-vision-camera #1819: 🐛 CaptureRequest contains unconfigured Input/Output Surface

OpachiEg, 2023-09-20:
### What's happening?

The CaptureRequest contains unconfigured Input/Output Surface error occurs when you change device, and also when you close the application and then open it again. The error appears when the video property is set.

### Reproduceable Code

```tsx
const [cameraType, setCameraType] = useState('back');
    const devices = useCameraDevices("wide-angle-camera")[cameraType];
    const [showCamera, setShowCamera] = useState(false);
    const camera = useRef(null);

    useEffect(() => {
        requestPermissions()
    }, []);

    const requestPermissions = async () => {
        const cameraPermission = await Camera.requestCameraPermission();
        const microphonePermission = await Camera.requestMicrophonePermission();
        if (cameraPermission === "granted" && microphonePermission === "granted") {
            setShowCamera(true);
        }
    };
   
    const changeCameraType = () => {
         if (cameraType === 'back') setCameraType('front');
         else setCameraType('back');
    };

    return (
        <View style={...}>
            {showCamera && devices &&
            <Camera video={true} device={devices} isActive={true} ref={camera} style={StyleSheet.absoluteFill}/>}
        </View>
    );
```

### Relevant log output

```shell
java.lang.IllegalArgumentException: CaptureRequest contains unconfigured Input/Output Surface!
  android.hardware.camera2.CaptureRequest.convertSurfaceToStreamId(CaptureRequest.java:739)
  android.hardware.camera2.impl.CameraDeviceImpl.submitCaptureRequest(CameraDeviceImpl.java:1215)
  android.hardware.camera2.impl.CameraDeviceImpl.setRepeatingRequest(CameraDeviceImpl.java:1263)
  android.hardware.camera2.impl.CameraCaptureSessionImpl.setRepeatingRequest(CameraCaptureSessionImpl.java:313)
  com.mrousavy.camera.core.CameraSession.updateRepeatingRequest(CameraSession.kt:548)
  com.mrousavy.camera.core.CameraSession.startRunning(CameraSession.kt:526)
  com.mrousavy.camera.core.CameraSession.access$startRunning(CameraSession.kt:51)
  com.mrousavy.camera.core.CameraSession$configureSession$1.invokeSuspend(CameraSession.kt:157)
  kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
  kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
  android.os.Handler.handleCallback(Handler.java:938)
  android.os.Handler.dispatchMessage(Handler.java:99)
  android.os.Looper.loopOnce(Looper.java:201)
  android.os.Looper.loop(Looper.java:288)
  android.os.HandlerThread.run(HandlerThread.java:67)
```

### Camera Device

```json
{
  "hardwareLevel": "limited",
  "pixelFormats": [
    "JPEG",
    "PRIVATE",
    "YUV_420_888"
  ],
  "maxZoom": 4,
  "minZoom": 1,
  "supportsLowLightBoost": true,
  "sensorOrientation": "landscape-right",
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": false,
  "isMultiCam": false,
  "hasTorch": true,
  "devices": [
    "wide-angle-camera"
  ],
  "hasFlash": true,
  "name": "BACK (0)",
  "position": "back",
  "id": "0"
}
```

### Device

Realme C31 (Android 12)

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

bglgwyng, 2023-09-25:
It's duplicated. #1773

OpachiEg, 2023-09-26:
Did you fix it?

bglgwyng, 2023-09-27:
No 😢

mrousavy, 2023-09-30:
Duplicate of #1773 yep

mrousavy, 2023-09-30:
@bglgwyng can I somehow give you permission to close duplicate issues here? is that something you want to do?

closed 2023-09-30 by mrousavy

bglgwyng, 2023-09-30:
Sure! I'm glad I can offer even a small help.