margelo/react-native-vision-camera #1820: Inconsistent `Segfault` Exceptions During Continuous OCR Processing V2.15.6🐛

mihailpozarski, 2023-09-20:
### What's happening?

I don't really know if this is a bug or an implementation mistake... sorry if it's the second

**Description:**
I'm encountering intermittent Segfault exceptions while running continuous OCR processing in my OCR Access control app. This app utilizes react-native-vision-camera and vision-camera-ocr to perform OCR on national identity cards. Here's the context of the issue:

**Context:**

- The app continuously performs OCR on national identity cards using react-native-vision-camera and vision-camera-ocr.
- The app is designed to stop image processing if it fails to recognize text for longer than 20 seconds. When this occurs, the app unmounts the camera and switches to an idle screen to conserve resources.
- In idle mode, the app only monitors the light sensor module values of the device to detect changes in shadows. If the change exceeds 10%, the camera is reactivated, and OCR processing resumes.

**Issue Details:**

The problem arises when the app performs long non-stop OCR runs (500+ national identity card readings).

The typical processing time for a user's OCR operation is as follows:
- `0.3-0.8 seconds` for `vision-camera-ocr` to extract text from the card.
- `0.01 seconds` to validate the national identity number (for Chile).
- `0.4-0.8 seconds` to check if the user is permitted access.
- `1.0 seconds` to display the result.
After obtaining a valid national identity number, the app discards all subsequent frames until the result stop to display, and the process restarts. The issue occurs when conducting extended OCR runs, and it results in sporadic Segfault exceptions that crash the app. These exceptions are reported in Sentry.io, but they lack consistency of when it happens and why.

### Reproduceable Code

```tsx
// MY MAIN COMPONENT

import {runOnJS, useDerivedValue, useSharedValue} from 'react-native-reanimated';
import React, {useState, useEffect} from 'react';
import CameraContainer from '../../components/CameraContainer';
import {View} from 'react-native';
import {extractRutFromText} from '../../services/rutExtractionService';
import styles from './styles';
import {useSelector, useDispatch} from 'react-redux';
import { setProcessingAccess } from '../../store/slices/appStatusSlice';
import { useFrameProcessor } from 'react-native-vision-camera';
import { scanOCR } from 'vision-camera-ocr';
import {saveAccessLog} from '../../actions';
import {
  getClosestWindow,
} from '../../services/accessLogsService';
import AccessResult from '../../components/AccessResult';
import dayjs from 'dayjs';

const MainArea = () => {
  const dispatch = useDispatch();

  const {windows} = useSelector(state => state.windows); // redux saved windows that are the time frame where the user can access
  const config = useSelector(state => state.globalConfiguration); // redux saved global configuration for the app
  const {processingAccess} = useSelector(state => state.appStatus); // redux app status to know if the app is processing an access or not
  const [rut, setRut] = useState(''); // National identity number to be processed
  const [accessInfo, setAccessInfo] = useState({
    window: null,
    now: null,
  }); // Access information to be displayed in the UI

  const sharedText = useSharedValue(''); // shared value where the text from the camera is stored

  // Frame processor to process the text from the camera
  const frameProcessor = useFrameProcessor(frame => {
    'worklet';
    try {
      if(processingAccess) return;

      const result = scanOCR(frame).result;
      sharedText.value = result.text;
    } catch (error) {
      console.log('FrameProcessorError', error);
[... 422 lines cut ...]
  ],
  "position": "back",
  "id": "0"
}
```

### Device

Samsung Active Tab 3 (SM-T575)

### VisionCamera Version

2.15.6

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-30:
Hey - what does the native plugin look like? Not sure where that's coming from

mrousavy, 2023-09-30:
Can you reproduce this in V3?

bglgwyng, 2024-04-02:
@mrousavy I think we need to reopen this issue. I found a similar behavior in V4.

[crash logs](https://gist.github.com/bglgwyng/dc17a68aeb4ccc3b88ff563126531cb1)
[tombstone.zip](https://github.com/mrousavy/react-native-vision-camera/files/14830202/tombstone.zip)

mrousavy, 2024-04-02:
```
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #00 pc 0000000000053690  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!libVisionCamera.so (std::__ndk1::__shared_ptr_emplace<facebook::jsi::Function, std::__ndk1::allocator<facebook::jsi::Function> >::__on_zero_shared()+40) (BuildId: 2ef446e9d5c3437bd251bc77fa755108fa8bffe8)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #01 pc 0000000000052a0c  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!libVisionCamera.so (std::__ndk1::shared_ptr<facebook::jsi::Function>::~shared_ptr()+64) (BuildId: 2ef446e9d5c3437bd251bc77fa755108fa8bffe8)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #02 pc 0000000000121180  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #03 pc 000000000014dad0  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #04 pc 000000000014dc4c  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #05 pc 000000000014e770  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #06 pc 000000000014d528  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.274 23503-23503 DEBUG                   crash_dump64                         A        #07 pc 000000000010c270  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.275 23503-23503 DEBUG                   crash_dump64                         A        #08 pc 0000000000101d20  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (std::__ndk1::function<facebook::jsi::Value (facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)>::~function()+20) (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
2024-04-02 10:02:36.275 23503-23503 DEBUG                   crash_dump64                         A        #09 pc 00000000001195ec  /data/app/~~D9p2t0ckgWZPmbjjrQrFIA==/com.gwyng.shootee-GBSdWDkhR2AeS4bIW1EH6Q==/base.apk!librnworklets.so (RNWorklet::JsiWorkletApi::createRunInJsFn(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)::'lambda'(facebook::jsi::Runtime&, facebook::jsi::Value const&, facebook::jsi::Value const*, unsigned long)::~()+20) (BuildId: 56fa855bc793fcb6dd9ad036e50bf9fb14bec2f9)
```

Some quick analysis from your crash logs @bglgwyng (bottom to top):
1. `createRunInJsFn` gets called, so you are creating a shareable function for the Worklet runtime
2. The `jsi::Function` gets allocated and works fine
3. At the very top, `on_zero_shared` is called, which means there are no more references to that shared pointer (either the `jsi::Function` itself, or some other value like the Worklet, or even the `FrameHostObject`).
4. Inside `on_zero_shared`, the `shared_ptr` is trying to delete the memory, but fails, hence the crash.

Do you have more logs of this? Do you know for a fact that this happens in the middle of the application, or does thhis happen when the user is closing the app for example?

mrousavy, 2024-04-02:
@chrfalch maybe this is a bug in RN Worklets 🤔

mrousavy, 2024-04-22:
Hey - I think this issue has been fixed in [VisionCamera 4.0.0](https://github.com/mrousavy/react-native-vision-camera/releases/tag/v4.0.0). 🥳 

Please try V4 and let me know if you still experience this issue;
- if not, please consider [💖 **sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy/) to support the development of VisionCamera and thank me for my time spent on fixing bugs and building new features.
- if you still see this issue, please comment and we can re-open this. But please update your native logs with the native (Xcode/Android Studio) logs from running VisionCamera V4 so I can investigate this.

closed 2024-04-22 by mrousavy