margelo/react-native-vision-camera #1821: 🐛

mnecchi, 2023-09-20:
### What's happening?

Just created a new RN app using `react-native init`, installed `react-native-reanimated` and `react-native-vision-camera` and installed the pods for the new architecture.
What I get when running on a real device is this:
[image]

### Reproduceable Code

```tsx
import {StyleSheet} from 'react-native';
import {
  Camera,
  useCameraDevices,
  CameraPermissionStatus,
} from 'react-native-vision-camera';
import {useEffect, useState} from 'react';

const App = () => {
  const [cameraStatus, setCameraStatus] = useState<CameraPermissionStatus>();

  const device = useCameraDevices();

  useEffect(() => {
    Camera.getCameraPermissionStatus().then(setCameraStatus);
  }, []);

  useEffect(() => {
    if (!cameraStatus || cameraStatus === 'granted') {
      return;
    }

    Camera.requestCameraPermission().then(setCameraStatus);
  }, [cameraStatus]);

  if (!device?.front) {
    return null;
  }

  return (
    <Camera isActive style={StyleSheet.absoluteFill} device={device.front} />
  );
};

export default App;
```

### Relevant log output

```shell
flipper: FlipperClient::addPlugin Inspector
flipper: FlipperClient::addPlugin Preferences
flipper: FlipperClient::addPlugin React
flipper: FlipperClient::addPlugin Network
2023-09-20 15:48:30.040422+0100 TestApp[1655:86980] [connection] nw_socket_handle_socket_event [C1.1.1:1] Socket SO_ERROR [54: Connection reset by peer]
2023-09-20 15:48:30.173896+0100 TestApp[1655:86992] [javascript] Running "TestApp" with {"fabric":true,"initialProps":{"concurrentRoot":true},"rootTag":1}
2023-09-20 15:48:30.400029+0100 TestApp[1655:86980] [connection] nw_socket_handle_socket_event [C5:1] Socket SO_ERROR [61: Connection refused]
2023-09-20 15:48:30.400748+0100 TestApp[1655:86971] [connection] nw_connection_get_connected_socket_block_invoke [C5] Client called nw_connection_get_connected_socket on unconnected nw_connection
2023-09-20 15:48:30.400847+0100 TestApp[1655:86971] TCP Conn 0x280f06ee0 Failed : error 0:61 [61]
2023-09-20 15:48:30.401203+0100 TestApp[1655:86971] [connection] nw_connection_copy_connected_local_endpoint_block_invoke [C5] Client called nw_connection_copy_connected_local_endpoint on unconnected nw_connection
2023-09-20 15:48:30.401300+0100 TestApp[1655:86971] [connection] nw_connection_copy_connected_remote_endpoint_block_invoke [C5] Client called nw_connection_copy_connected_remote_endpoint on unconnected nw_connection
2023-09-20 15:48:33.111779+0100 TestApp[1655:86972] SocketStream read error [0x280f48e60]: 1 54
2023-09-20 15:48:36.179034+0100 TestApp[1655:86972] SocketStream read error [0x280f7c5a0]: 1 54
```

### Camera Device

```json
{
    "maxZoom": 128.875,
    "hasFlash": true,
[... 1666 lines cut ...]
      }
    ],
    "hasTorch": false
  }
```

### Device

iPhone 14 (iOS 16.6.1)

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-27:
Yea, VisionCamera does not support the new arch yet since I don't use new arch in any of my apps yet. If you want this implemented consider paying me to do it, or submit a PR yourself :)

mnecchi, 2023-09-28:
> Yea, VisionCamera does not support the new arch yet since I don't use new arch in any of my apps yet. If you want this implemented consider paying me to do it, or submit a PR yourself :)

Fair enough 😂

closed 2023-09-30 by mrousavy