margelo/react-native-vision-camera #1833: 🐛

fvalles, 2023-09-22:
### What's happening?

When using react native vision camera frameProcessor property in order to scan QR codes (combined with `vision-camera-code-scanner`), I get the following android error when navigating to the screen that renders the camera view (app builds correctly):

`java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)`

and then the app dies.

Fun fact is that without using the frameProcessor prop, everything works fine.

This error happens without installing the library `react-native-worklets-core` and adding to babel.config.js: `['react-native-worklets-core/plugin']`

---

On the other hand, If I add  `react-native-worklets-core` and its babel config line, I get a different error and I can't even build the app:

 ```
What went wrong:
Could not determine the dependencies of task ':react-native-worklets-core:compileDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-worklets-core:debugCompileClasspath'.
   > Could not find com.facebook.react:react-android:.
     Required by:
         project :react-native-worklets-core
   > Could not find com.facebook.react:hermes-android:.
     Required by:
         project :react-native-worklets-core
```

---

I have looked at this two similar bug reports and tried everything explained there but the issue persists and I am not finding a solution..
https://github.com/mrousavy/react-native-vision-camera/issues/1463
https://github.com/mrousavy/react-native-vision-camera/issues/1097

### Reproduceable Code

```tsx
import { useIsFocused } from '@react-navigation/native'
import { BarcodeFormat, Barcode, scanBarcodes } from 'vision-camera-code-scanner'

import React, { useEffect, useState } from 'react'
import { StyleSheet } from 'react-native'
import { runOnJS } from 'react-native-reanimated'
import { useCameraDevices, useFrameProcessor } from 'react-native-vision-camera'
import { Camera } from 'react-native-vision-camera'

import { LoaderComponent } from '#components/loader'

export const QRReader = () => {
  const [hasPermission, setHasPermission] = React.useState(false)
  const devices = useCameraDevices()
  const device = devices.back
  const [barcodes, setBarcodes] = useState<Barcode[]>([])
  const isFocused = useIsFocused()

  console.log('hasPermission', hasPermission)

  useEffect(() => {
    console.log('barcodes', barcodes)
  }, [barcodes])

  const frameProcessor = useFrameProcessor((frame) => {
    'worklet'
    const detectedBarcodes = scanBarcodes(frame, [BarcodeFormat.QR_CODE])
    runOnJS(setBarcodes)(detectedBarcodes)
  }, [])

  useEffect(() => {
    const setCameraPermission = async () => {
      const status = await Camera.requestCameraPermission()
      setHasPermission(status === 'authorized')

      const cameraPermission = await Camera.getCameraPermissionStatus()
      console.log('cameraPermission', cameraPermission)
    }

    setCameraPermission()
  }, [])

  if (device == null) {
    return <LoaderComponent />
  }

  return (
    <Camera
      style={StyleSheet.absoluteFill}
      device={device}
      isActive={isFocused}
      frameProcessor={frameProcessor}
      frameProcessorFps={5}
    />
  )
}
```

### Relevant log output

```shell
java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)

FATAL EXCEPTION: pool-28-thread-1
Process: myApp.dev, PID: 22643
java.lang.UnsatisfiedLinkError: No implementation found for void com.mrousavy.camera.CameraView.frameProcessorCallback(androidx.camera.core.ImageProxy) (tried Java_com_mrousavy_camera_CameraView_frameProcessorCallback and Java_com_mrousavy_camera_CameraView_frameProcessorCallback__Landroidx_camera_core_ImageProxy_2)
	at com.mrousavy.camera.CameraView.frameProcessorCallback(Native Method)
	at com.mrousavy.camera.CameraView.configureSession$lambda-7$lambda-6(CameraView.kt:491)
	at com.mrousavy.camera.CameraView.$r8$lambda$cqtIchEZdTZaV3R0UUrDpVbB1Es(Unknown Source:0)
	at com.mrousavy.camera.CameraView$$ExternalSyntheticLambda1.analyze(Unknown Source:2)
	at androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$2(ImageAnalysis.java:476)
	at androidx.camera.core.ImageAnalysis$$ExternalSyntheticLambda0.analyze(Unknown Source:2)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer.lambda$analyzeImage$0$androidx-camera-core-ImageAnalysisAbstractAnalyzer(ImageAnalysisAbstractAnalyzer.java:285)
	at androidx.camera.core.ImageAnalysisAbstractAnalyzer$$ExternalSyntheticLambda1.run(Unknown Source:14)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1137)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:637)
	at java.lang.Thread.run(Thread.java:1012)
```

### Camera Device

```json
{
  "formats": [],
  "maxZoom": 10,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "supportsDepthCapture": false,
  "neutralZoom": 1,
  "supportsFocus": true,
  "supportsRawCapture": false,
  "hasFlash": false,
  "name": "back (0)",
  "supportsParallelVideoProcessing": false,
  "isMultiCam": false,
  "hasTorch": false,
  "devices": [
    "telephoto-camera"
  ],
  "position": "back",
  "id": "0"
}
```

### Device

Android Pixel 4 api 31

### VisionCamera Version

^2.16.1

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

PranatoshRoy, 2023-09-22:
I have the same issue while using vision-camera-code-scanner. need help

QSuraj, 2023-09-24:
Making following changes should fix the issue - https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1731248952

fvalles, 2023-09-24:
> Making following changes should fix the issue - [#1307 (comment)](https://github.com/mrousavy/react-native-vision-camera/issues/1307#issuecomment-1731248952)

The thread you mentioned is for ios and mine is an issue with android devices

JorgeQuevedoC, 2023-09-27:
I was facing the exact same problem, curious enough. I tried with specific package versions and it worked:

"react-native-vision-camera": "2.15.6"
"react-native-reanimated": "^2.14.4"
"vision-camera-code-scanner": "github:jorgebrunetto/vision-camera-code-scanner"

You can try and see if this works for you too

fvalles, 2023-09-27:
> I was facing the exact same problem, curious enough. I tried with specific package versions and it worked:
> 
> "react-native-vision-camera": "2.15.6" "react-native-reanimated": "2.17.0" "vision-camera-code-scanner": "github:jorgebrunetto/vision-camera-code-scanner"
> 
> You can try and see if this works for you too

Ey Jorge! And what is the  "vision-camera-code-scanner” version you are using? That field is empty in your previous response

JorgeQuevedoC, 2023-09-27:
It is a fork that the team I am working with was using. Probably using the one you have right now won't impact. The breaking change is in "react-native-vision-camera" from 2.15.6 to 2.16.1 because they changed from CameraX to Camera2 
[image]

fvalles, 2023-09-28:
> It is a fork that the team I am working with was using. Probably using the one you have right now won't impact. The breaking change is in "react-native-vision-camera" from 2.15.6 to 2.16.1 because they changed from CameraX to Camera2 [image]

you just rock it man! With those libraries versions everything is working well in my android device!! 😁

Thanks a lot for your help! 🫶

Now I will check if everything works on the iOS side but I am confident that it will!

drastus, 2023-09-29:
> I was facing the exact same problem, curious enough. I tried with specific package versions and it worked:
>
> "react-native-vision-camera": "2.15.6" "react-native-reanimated": "2.17.0" [...]

@JorgeQuevedoC what version of React Native are you using?

fvalles, 2023-09-29:
> @JorgeQuevedoC what version of React Native are you using?

Ey @drastus! In my case I am using the version 0.70.9 and with the libraries versions that @JorgeQuevedoC suggested everything is working now! 😄

JorgeQuevedoC, 2023-09-29:
> @JorgeQuevedoC what version of React Native are you using?

"react-native": "0.69.7",

mrousavy, 2023-09-30:
Hey this should be fixed in V3

closed 2023-09-30 by mrousavy

mrousavy, 2023-10-06:
Hey!

JFYI; **VisionCamera V3 now includes a QR/Barcode Scanner!** 😍  Check out the [CodeScanner Documentation](https://www.react-native-vision-camera.com/docs/guides/code-scanning) 🚀 

If you appreciate me dedicating my free time to improving VisionCamera and implementing features like the Code Scanner, please consider [**sponsoring me on GitHub** 💖](https://github.com/sponsors/mrousavy) to show your support.

hardik-javascript, 2023-10-17:
@mrousavy VisionCamera V3 doesn't work with react-native: 0.70.6 as android build fails.

mrousavy, 2023-10-17:
okay 👍

qamarcloud, 2023-12-12:
I want to perform all of the functions, perform OCR, do the barcode scanning and take a photo when I open a camera. I is possible in vision-camera-3

nc-hung, 2023-12-21:
> 
[V3] What you need can be found here:
https://www.react-native-vision-camera.com/docs/guides/frame-processor-plugin-list

[V3] I resolved the frameProcessor error through this note. Hope someone will need it
https://github.com/mrousavy/react-native-vision-camera/issues/1776#issuecomment-1766163792

arazaq917, 2024-09-02:
> ```tsx
> vision-camera-code-scanner
> ```

Hi @qamarcloud 
did you find any solution to perform all at one place?

shengl711, 2025-01-30:
> "react-native": "0.69.7",

@JorgeQuevedoC & @fvalles - I am just trying to confirm the following versions can scan QR code for you?  Also is this local version github:jorgebrunetto/vision-camera-code-scanner will continue to be available?   I like to confirm before installing that version.    I am using 0.72.3 for RN: 

"react-native": "0.72.3",
"react-native-vision-camera": "2.15.6"
"react-native-reanimated": "2.17.0" 
"vision-camera-code-scanner": "github:jorgebrunetto/vision-camera-code-scanner"