margelo/react-native-vision-camera #1840: 🐛 iOS 17.0.2 (iPhone 15) crashing on camera load

thyxus, 2023-09-23:
### What's happening?

While loading camera seems like `AVCaptureColorSpace` throws the following error
```
Fatal error: AVCaptureDevice.Position has unknown state.
```
- Could it be new formats in new iPhone or is this a bug on iOS 17.0.2?
- Is removing this error a good workaround for the time being?
- Could 

### Reproduceable Code

```tsx
const device = cameraDevices.back;
const format = useMemo(() => {
    if (!device?.formats) return undefined;
    const sortedFormats = device.formats.sort(sortFormats);
    const formatsWith60FPS = sortedFormats.filter((f) =>
        f.frameRateRanges.some((r) => frameRateIncluded(r, 60))
    );
    if (formatsWith60FPS.length > 0) return formatsWith60FPS[1];
    if (sortedFormats.length > 0) return sortedFormats[0];
    return undefined;
}, [device?.formats]);
const fps = useMemo(() => {
    if (!format) return undefined;
    return Math.max(
        ...format.frameRateRanges.map((range) => range.maxFrameRate)
    );
}, [format]);
<ReanimatedCamera
    photo={true}
    video={false}
    device={device}
    enableHighQualityPhotos={true}
    enableZoomGesture={false}
    orientation="portrait"
    isActive={focused}
    style={{
        position: 'absolute',
        height: '100%',
        width: '100%',
        padding: 0,
        margin: 0,
    }}
    ref={camera}
    torch={torch}
    format={Platform.OS === 'android' ? undefined : format}
    preset={!format || Platform.OS === 'android' ? 'photo' : undefined}
    fps={Platform.OS === 'android' ? undefined : fps}
    hdr={Platform.OS === 'android' ? undefined : false}
    animatedProps={cameraAnimatedProps}
    onInitialized={() => setCameraReady(true)}
    onError={(err) => {
        console.error(err);
    }}
/>;
```

### Relevant log output

```shell
VisionCamera/AVCaptureColorSpace+descriptor.swift:41: Fatal error: AVCaptureDevice.Position has unknown state.
```

### Camera Device

```json
Unknown as I don't the new iPhone and this is crash report
```

### Device

ex. iPhone 15

### VisionCamera Version

2.15.6

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

stianjensen, 2023-09-23:
Strange. The iOS docs still only mention three possible values for position:
https://developer.apple.com/documentation/avfoundation/avcapturedevice/position

And all three are handled here:
https://github.com/mrousavy/react-native-vision-camera/blob/v2.15.4/ios/Parsers/AVCaptureDevice.Position%2Bdescriptor.swift

McFly78, 2023-09-23:
Hi,

Same issue with iPhone 15 Pro Max.

I have just received a crash from Sentry coming from an iPhone 15 Pro Max in IOS 17.0.2.
EXC_BREAKPOINT
facebook::react::JSIExecutor::defaultTimeoutInvoker
VisionCamera/AVCaptureColorSpace+descriptor.swift:41: Fatal error: AVCaptureDevice.Position has unknown state.

stianjensen, 2023-09-23:
Would be great if someone with access to on iPhone 15 Pro or 15 Pro Max were able to add logging to see what the actual unknown state is, since there is no new reference in the iOS documentation for what the new state value is.

YKSing, 2023-09-23:
[image]
Hope this help.

stianjensen, 2023-09-23:
Hah - I didn't realize until now this is actually coming from the AVCaptureColorSpace+descriptor.swift file and not AVCaptureDevice.Position!

Looks like the problem is similar though. Apple has added a new enum value that is not yet documented. So will need to figure out what it is. Perhaps it's documented inside the latest version of Xcode 15.

Looks like the code is gone in react-native-vision-camera 3 so the crash is probably only happening on react-native-vision-camera 2.x

https://github.com/mrousavy/react-native-vision-camera/blob/9963f4dedc6fbef933fb99e0adca1343f4f5f972/ios/Parsers/AVCaptureColorSpace%2Bdescriptor.swift#L41

thyxus, 2023-09-23:
This patch temporarily resolves the issue by removing `colorSpaces` from AVCaptureDevice.Format
```
diff --git a/node_modules/react-native-vision-camera/ios/CameraView+AVCaptureSession.swift b/node_modules/react-native-vision-camera/ios/CameraView+AVCaptureSession.swift
index 5c9b77d..5ea7ad5 100644
--- a/node_modules/react-native-vision-camera/ios/CameraView+AVCaptureSession.swift
+++ b/node_modules/react-native-vision-camera/ios/CameraView+AVCaptureSession.swift
@@ -195,14 +195,14 @@ extension CameraView {
           device.automaticallyEnablesLowLightBoostWhenAvailable = lowLightBoost!.boolValue
         }
       }
-      if let colorSpace = colorSpace as String? {
-        guard let avColorSpace = try? AVCaptureColorSpace(string: colorSpace),
-              device.activeFormat.supportedColorSpaces.contains(avColorSpace) else {
-          invokeOnError(.format(.invalidColorSpace(colorSpace: colorSpace)))
-          return
-        }
-        device.activeColorSpace = avColorSpace
-      }
+      // if let colorSpace = colorSpace as String? {
+      //   guard let avColorSpace = try? AVCaptureColorSpace(string: colorSpace),
+      //         device.activeFormat.supportedColorSpaces.contains(avColorSpace) else {
+      //     invokeOnError(.format(.invalidColorSpace(colorSpace: colorSpace)))
+      //     return
+      //   }
+      //   device.activeColorSpace = avColorSpace
+      // }
 
       device.unlockForConfiguration()
       ReactLogger.log(level: .info, message: "Device successfully configured!")
diff --git a/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+matchesFilter.swift b/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+matchesFilter.swift
index 35789a6..bff8572 100644
--- a/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+matchesFilter.swift
+++ b/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+matchesFilter.swift
@@ -54,13 +54,13 @@ extension AVCaptureDevice.Format {
         return false
       }
     }
-    if let colorSpaces = filter.value(forKey: "colorSpaces") as? [String] {
-      let avColorSpaces = colorSpaces.map { try? AVCaptureColorSpace(string: $0) }
-      let allColorSpacesIncluded = supportedColorSpaces.allSatisfy { avColorSpaces.contains($0) }
-      if !allColorSpacesIncluded {
-        return false
-      }
-    }
+    // if let colorSpaces = filter.value(forKey: "colorSpaces") as? [String] {
+    //   let avColorSpaces = colorSpaces.map { try? AVCaptureColorSpace(string: $0) }
+    //   let allColorSpacesIncluded = supportedColorSpaces.allSatisfy { avColorSpaces.contains($0) }
+    //   if !allColorSpacesIncluded {
+    //     return false
+    //   }
+    // }
     if let frameRateRanges = filter.value(forKey: "frameRateRanges") as? [NSDictionary] {
       let allFrameRateRangesIncluded = videoSupportedFrameRateRanges.allSatisfy { range -> Bool in
         frameRateRanges.contains { dict -> Bool in
diff --git a/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+toDictionary.swift b/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+toDictionary.swift
index fec83ca..f7bce0e 100644
--- a/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+toDictionary.swift
+++ b/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+toDictionary.swift
@@ -33,7 +33,7 @@ extension AVCaptureDevice.Format {
       "minISO": minISO,
       "fieldOfView": videoFieldOfView,
       "maxZoom": videoMaxZoomFactor,
-      "colorSpaces": supportedColorSpaces.map(\.descriptor),
+      // "colorSpaces": supportedColorSpaces.map(\.descriptor),
       "supportsVideoHDR": isVideoHDRSupported,
       "supportsPhotoHDR": false,
       "frameRateRanges": videoSupportedFrameRateRanges.map {
diff --git a/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift b/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift
index 13a403b..e1781ab 100644
--- a/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift
[... 57 lines cut ...]
-    case .HLG_BT2020:
-      return "hlg-bt2020"
-    case .P3_D65:
-      return "p3-d65"
-    case .sRGB:
-      return "srgb"
-    default:
-      fatalError("AVCaptureDevice.Position has unknown state.")
-    }
-  }
-}
+//   var descriptor: String {
+//     switch self {
+//     case .HLG_BT2020:
+//       return "hlg-bt2020"
+//     case .P3_D65:
+//       return "p3-d65"
+//     case .sRGB:
+//       return "srgb"
+//     default:
+//       fatalError("AVCaptureDevice.Position has unknown state.")
+//     }
+//   }
+// }
```

mrousavy, 2023-09-23:
my bad lol, the error message is wrong here.

Will fix this in V3 to make the enum casting safer!

cristeahub, 2023-09-23:
> my bad lol, the error message is wrong here.
> 
> Will fix this in V3 to make the enum casting safer!

Will there be a release of V2 that fixes this?

> This patch temporarily resolves the issue by removing `colorSpaces` from `AVCaptureDevice.Format`.
> 
> Oh right, apparently this field is removed in v3 as well so upgrading might resolve this as well if that's an option until the new value is figured out.
> ...

This patch seems to be working for me on iPhone 15 Pro 👍🏻

stianjensen, 2023-09-24:
Looks like this is the new enum value they've added:
https://developer.apple.com/documentation/avfoundation/avcapturecolorspace/applelog

I suppose adding support for this requires building the app with xcode 15 / ios 17 SDK.

frenberg, 2023-09-25:
Could a fix for this be as simple as adding 
```
    case .appleLog:
       return "appleLog"
```
to descriptor swith statement? Or what are the side effects of commenting the code as in @thyxus patch?

spsaucier, 2023-09-25:
> Could a fix for this be as simple as adding
> 
> ```
>     case "appleLog":
>         if #available(iOS 17, *) {
>           self = .appleLog
>         } else {
>           throw EnumParserError.unsupportedOS(supportedOnOS: "17")
>         }
>         return
> ```
> 
> to the init function and
> 
> ```
>     case .appleLog:
>        return "appleLog"
> ```
> 
> to descriptor swith statement of AVCaptureColorSpace? Or what are the side effects of commenting the code as in @thyxus patch?
> 
> Found a co worker with the new iphone 15 pro and can confirm that patching 2.15.4 with this code is working! 🎉

Trying this results in build errors on XCode 14:

```
2023-09-25T15:29:18.8227700Z CompileSwift normal arm64 (in target 'VisionCamera' from project 'Pods')
2023-09-25T15:29:18.8227930Z     cd /Users/runner/work/1/s/ios/Pods
2023-09-25T15:29:18.8281630Z     /Applications/Xcode_14.2.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/swift-frontend -c /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVAssetWriter.Status+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVAssetWriterInputPixelBufferAdaptor+initWithVideoSettings.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVAudioSession+trySetAllowHaptics.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVAudioSession+updateCategory.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVAuthorizationStatus+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureConnection+setInterfaceOrientation.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice+isMultiCam.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice+neutralZoom.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice+physicalDevices.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureDevice.DeviceType+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureDevice.FlashMode+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+isBetterThan.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+matchesFilter.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+toDictionary.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureDevice.Format+videoDimensions.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureDevice.Format.AutoFocusSystem+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureDevice.Position+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureDevice.TorchMode+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCapturePhotoOutput+mirror.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCapturePhotoOutput.QualityPrioritization+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVCaptureSession+setVideoStabilizationMode.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureSession.Preset+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureVideoStabilizationMode+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVFileType+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/AVFrameRateRange+includes.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVVideoCodecType+descriptor.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/React\ Utils/Callback.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraError.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraQueues.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+AVAudioSession.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+AVCaptureSession.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+Focus.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+Orientation.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+RecordVideo.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+TakePhoto.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraView+Zoom.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/CameraViewManager.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Extensions/Collection+safe.swift /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/EnumParserError.swift /Users/runner/wo [... cut ...]
2023-09-25T15:29:18.8309270Z                ~^~~~~~~~
2023-09-25T15:29:18.8309730Z /Users/runner/work/1/s/node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift:47:11: error: type 'AVCaptureColorSpace' has no member 'appleLog'
2023-09-25T15:29:18.8309990Z     case .appleLog:
2023-09-25T15:29:18.8310150Z          ~^~~~~~~~
2023-09-25T15:29:18.8310200Z 
```

And [XCode 15 builds aren't working](https://github.com/mrousavy/react-native-vision-camera/issues/1802)...

frenberg, 2023-09-25:
In my case the "Frame processors not enabled" was an issue from static linking (using firebase). I had to add a pre_install section to my podfile.
```
abstract_target 'ReactNative' do
  # static linking was not working with vision camera/document normalizer combo
  pre_install do |installer|
    installer.pod_targets.each do |pod|
      if pod.name.eql?('vision-camera-dynamsoft-document-normalizer') || pod.name.eql?('VisionCamera') || pod.name.eql?('RNReanimated')
        Pod::UI.puts "#{pod.name}: Using overridden static_framework value of 'true'"
        def pod.build_type
          Pod::BuildType.static_library
        end
      end
    end
  end

  use_frameworks! :linkage => :static
```

I suppose it should not really have to do with xcode 15, but i rebased my branch in the same timeframe as the xcode problems occured. (Came back from lunch and osx had updated xcode so nothing was working anymore 🤯 ) Anyway, hope that helps.

KristianSelnas, 2023-09-26:
I had this issue on iOS 17 / iPhone 15 as well, but upgrading to v3.1 seems to have solved it for me. At least the camera now works on my phone during local development.

ANOONBP, 2023-09-26:
I had this issue too. I'm using react-native-vision-camera@2.15.4 and camera crash every time with error AVCaptureDevice.Position has unknown state.

KristianSelnas, 2023-09-26:
> I had this issue too. I'm using react-native-vision-camera@2.15.4 and camera crash every time with error AVCaptureDevice.Position has unknown state.

Have you tried 3.1? It worked for me

osbornZ, 2023-09-27:
This crash is AVCaptureColorSpace enum  add new type AVCaptureColorSpace_AppleLog
`
AVCaptureColorSpace_AppleLog API_AVAILABLE(ios(17.0), macCatalyst(17.0), tvos(17.0)) API_UNAVAILABLE(macos, visionos) = 3,
`
In this file add case parse,  it can work for me.     //  AVCaptureColorSpace+descriptor.swift

jmartin6716, 2023-09-29:
We are having the same issue.  We are still using react-native-camera and not react-native-vision-camera. I am guessing if we upgrade to the latest react-native-vision-camera that will solve this.  We are going to give that a try in the am.  Has anyone else had the issue of the camera going black on an older phone, but on iOS17?

rafaelmotta, 2023-09-29:
> We are having the same issue. We are still using react-native-camera and not react-native-vision-camera. I am guessing if we upgrade to the latest react-native-vision-camera that will solve this. We [...]

I tested with iPhone 12 / iOS 17 - works great, no crashes. With the iPhone 15, though, we're experiencing the issues on version 2.x.

mrousavy, 2023-09-30:
Fixed in latest v2

closed 2023-09-30 by mrousavy

GolliAch, 2023-09-30:
@mrousavy is it fixed on 2.16.1?

pid0r-3blan, 2023-10-01:
> @mrousavy is it fixed on 2.16.1? Or should we wait for a new V2 release?

I think it'll be released in `2.16.2`, but not yet now https://github.com/mrousavy/react-native-vision-camera/commits/v2

wilsolutions, 2023-10-04:
Please, I'm just wondering when we may have a V2 release with this fix, it would help me to plan accordingly...
thanks

mrousavy, 2023-10-05:
Just released this in 2.16.2., hopefully with a better and more stable solution. Also the crash in colorSpace is fixed.

If you appreciate my **free** support for VisionCamera, please [consider sponsoring me on GitHub](https://github.com/sponsors/mrousavy). Thanks. ❤️

2sem, 2023-10-06:
https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.16.2 
2.16.2 released

🐛 Bug Fixes
Fix "Fatal error: AVCaptureDevice.Position has unknown state" ([503114b](https://github.com/mrousavy/react-native-vision-camera/commit/503114b9d1c61d60fa3c69b3bd4cb084758fbf5b)), closes https://github.com/mrousavy/react-native-vision-camera/issues/1840
Fix Xcode 15 crash when using +(void)load ([2ae4439](https://github.com/mrousavy/react-native-vision-camera/commit/2ae4439c844e62131f76d35ffeca76f5a1c7237f))
Updated orientation calculations (https://github.com/mrousavy/react-native-vision-camera/issues/1544) ([f35644d](https://github.com/mrousavy/react-native-vision-camera/commit/f35644df3357d36309c1827ef71aa6e383070b6c))
Use initialize instead of +load ([45742ad](https://github.com/mrousavy/react-native-vision-camera/commit/45742ad969be685117970fec6fb0a60bf61d9549))
Use VISION_CONCAT(initialize_) ([8554aa2](https://github.com/mrousavy/react-native-vision-camera/commit/8554aa27bfc366e821667f3ede49d910012360a8))

2sem, 2023-10-06:
https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.16.2
2.16.2 released

🐛 Bug Fixes
Fix "Fatal error: AVCaptureDevice.Position has unknown state" ([503114b](https://github.com/mrousavy/react-native-vision-camera/commit/503114b9d1c61d60fa3c69b3bd4cb084758fbf5b)), closes https://github.com/mrousavy/react-native-vision-camera/issues/1840
Fix Xcode 15 crash when using +(void)load ([2ae4439](https://github.com/mrousavy/react-native-vision-camera/commit/2ae4439c844e62131f76d35ffeca76f5a1c7237f))
Updated orientation calculations (https://github.com/mrousavy/react-native-vision-camera/pull/1544) ([f35644d](https://github.com/mrousavy/react-native-vision-camera/commit/f35644df3357d36309c1827ef71aa6e383070b6c))
Use initialize instead of +load ([45742ad](https://github.com/mrousavy/react-native-vision-camera/commit/45742ad969be685117970fec6fb0a60bf61d9549))
Use VISION_CONCAT(initialize_) ([8554aa2](https://github.com/mrousavy/react-native-vision-camera/commit/8554aa27bfc366e821667f3ede49d910012360a8))

ferencIOS, 2023-10-09:
[Mukthayar1](https://github.com/Mukthayar1)(https://github.com/mrousavy/react-native-vision-camera/issues/1840#issuecomment-1752964704)

>>> In the released version 2.16.2, I am encountering the issue "__scanCodes not found," while it was working fine in the previous build.

@mrousavy, I am experiencing the same problem!

Danushka96, 2023-10-10:
> Just released this in 2.16.2., hopefully with a better and more stable solution. Also the crash in colorSpace is fixed.
> 
> If you appreciate my **free** support for VisionCamera, please [consider sponsoring me on GitHub](https://github.com/sponsors/mrousavy). Thanks. ❤️

Xcode 15, camera screen crash with `Property '__scanCodes' doesn't exist`. Previous build (2.15.5) works fine but it doesn't work with code 15.

sommcz, 2023-10-10:
Same issue here: `Property '__scanCodes' doesn't exist`

Xcode 15
react-native-vision-camera version 2.16.2
react-native-reanimated version 3.5.4

Mukthayar1, 2023-10-10:
> > Just released this in 2.16.2., hopefully with a better and more stable solution. Also the crash in colorSpace is fixed.
> > If you appreciate my **free** support for VisionCamera, please [consider sponsoring me on GitHub](https://github.com/sponsors/mrousavy). Thanks. ❤️
> 
> Xcode 15, camera screen crashes with `Property '__scanCodes' doesn't exist`. The previous build (2.15.5) works fine but doesn't work with code 15.
> 
> react-native-vision-camera version 2.16.2 react-native-reanimated version 3.5.4

Please switch to react native camera kit

chinieer, 2023-10-11:
Same issue

chinieer, 2023-10-11:
> Same issue here: `Property '__scanCodes' doesn't exist`
> 
> Xcode 15 react-native-vision-camera version 2.16.2 react-native-reanimated version 3.5.4

same issue

nicks78, 2023-10-11:
same issue: Property '__scanCodes' doesn't exist

Xcode 15 react-native-vision-camera version 2.16.2 react-native-reanimated version 3.5.4

JorgeQuevedoC, 2023-10-12:
> Just released this in 2.16.2., hopefully with a better and more stable solution. Also the crash in colorSpace is fixed.
> 
> If you appreciate my **free** support for VisionCamera, please [consider sponsoring me on GitHub](https://github.com/sponsors/mrousavy). Thanks. ❤️

Hi @mrousavy 

I am still using V2 2.15.6 because by the time I implemented it 2.16.1 didn't have support for frameProcessors. 

I am getting the app crashing for iOS 17+. Does 2.16.2 has support for frameProcessors? Or what would you suggest in this case?

mrousavy, 2023-10-16:
Upgrade to V3.

GSFZamai, 2023-10-24:
I applied the following commit as a patch and it worked.
[Commit](https://github.com/mrousavy/react-native-vision-camera/commit/503114b9d1c61d60fa3c69b3bd4cb084758fbf5b)
RNVC 2.15.6.
But soon I'll update to RNVC v3

thongquach, 2023-10-31:
Could you share how you applied that patch, as I didn't see this file in node_modules? @GSFZamai Gs Thanks in advance.

thongquach, 2023-10-31:
I applied the patch on RNVC 2.15.6, but it does not work. @GSFZamai GS

GSFZamai, 2023-10-31:
> I applied the patch on RNVC 2.15.6, but it does not work. @GSFZamai GS

Did you applied using patch-package?
In my case the error war on this line. 
All I did was apply this fix and worked.
1 - Install all packages ( npm i or yarn);
2 - Go to the related file on node_modules>...>ios/Parsers/AVCaptureColorSpace+descriptor.swift;
3 - Make the change;
4 - Run patch-package react-native-vision-camera;

It should create a diff file on your project's folder with the change.

thongquach, 2023-10-31:
I did use `patch-package` and followed your steps. @GSFZamai 
[image]

GSFZamai, 2023-10-31:
> I did use `patch-package` and followed your steps. @GSFZamai [image]

It seems that you are applying at the wrong file.
Correct file is `node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift`
Seems like you are applying at `node_modules/react-native-vision-camera/ios/Parsers/AVCaptureDevice+descriptor.swift`

pke, 2023-11-06:
I've already a patch for this package but creating a new one seems to be impossible. It includes all kind of build artifacts (.cxx, boost) which should not be part of the patch. I tried various include and exclude options to no avail.

pke, 2023-11-10:
Any idea what camera setting might trigger this bug in the first place? I want a reproducible way to see the crash happening before applying the patch.

chramos, 2023-11-16:
> Could a fix for this be as simple as adding
>
> ```
>     case "appleLog":
>         if #available(iOS 17, *) {
>           self = .appleLog
>         } else {
>           throw [...]

I can confirm that it works.
But the build only works with xcode 15.
Unfortunately our CI/CD process is using xcode 14 so I end up applying the [patch](https://github.com/mrousavy/react-native-vision-camera/issues/1840#issuecomment-1732360337)

mikebds, 2023-12-21:
> I can confirm that it works. But the build only works with xcode 15. Unfortunately our CI/CD process is using xcode 14 so I end up applying the [...]

Same case

yasassdk, 2024-01-02:
With version 2.16.2, the camera crash got fixed. Although the camera preview only appears at the top left corner of the screen in about 1/3 of the screen height. 
Does anyone experience the same?

vitas61, 2024-01-10:
look at my solution
[1884467346](https://github.com/mrousavy/react-native-vision-camera/issues/1802#issuecomment-1884467346)

SHINGPH, 2024-01-22:
> It seems that you are applying at the wrong file. Correct file is `node_modules/react-native-vision-camera/ios/Parsers/AVCaptureColorSpace+descriptor.swift` Seems like you are applying at [...]

where is the patch-file?

SHINGPH, 2024-01-22:
currently i'm using
RN:0.70
VISION-CAMERA:2.14.1
REANIMATED:2.10.0
XCODE 14
ISSUE: VISION-CAMERA-FACE-RECOGNITION CRASH

locht, 2024-05-20:
> https://github.com/mrousavy/react-native-vision-camera/releases/tag/v2.16.2 2.16.2 released
>
> 🐛 Bug Fixes Fix "Fatal error: AVCaptureDevice.Position has unknown state" [...]

work for me, thanks

commit 503114b9 (2023-09-30): fix: Fix "Fatal error: AVCaptureDevice.Position has unknown state"

* fixes #1840