margelo/react-native-vision-camera #1843: 🐛 Reanimated 3.5.4 + RNVC V3 Frame Processor causes app crashes(signal SIGABRT)

bglgwyng, 2023-09-24:
### What's happening?

I bumped Reanimated version up to 3.5.4 and my app crashed immediately after launched.
When I unset `frameProcessor` props, the crash was gone.

### Reproduceable Code

```tsx
https://github.com/bglgwyng/react-native-vision-camera/tree/reanimated-3.5.4

The only thing I did was install reanimated 3.5.4.
You could observe that commenting `frameProcessor` props prevents the crash from occurring.
```

### Relevant log output

```shell
libsystem_kernel.dylib`:
    0x1e14ed570 <+0>:  mov    x16, #0x148
    0x1e14ed574 <+4>:  svc    #0x80
->  0x1e14ed578 <+8>:  b.lo   0x1e14ed598               ; <+40>
    0x1e14ed57c <+12>: pacibsp 
    0x1e14ed580 <+16>: stp    x29, x30, [sp, #-0x10]!
    0x1e14ed584 <+20>: mov    x29, sp
    0x1e14ed588 <+24>: bl     0x1e14e89a4               ; cerror_nocancel
    0x1e14ed58c <+28>: mov    sp, x29
    0x1e14ed590 <+32>: ldp    x29, x30, [sp], #0x10
    0x1e14ed594 <+36>: retab  
    0x1e14ed598 <+40>: ret
```

### Camera Device

```json
{
  "hardwareLevel": "full",
  "supportsLowLightBoost": false,
  "isMultiCam": false,
  "hasTorch": true,
  "hasFlash": true,
  "name": "Back Camera",
  "sensorOrientation": "portrait",
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:0",
  "maxZoom": 121.875,
  "supportsFocus": true,
  "supportsRawCapture": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "position": "back",
  "minZoom": 1
}
```

### Device

iPhone 13 (iOS 16.6.1)

### VisionCamera Version

3.0.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-24:
Hey! Hmm, maybe there is some incompatibility with the Babel plugins? Which one do you use? Both REA and Worklets?

The log output is not really relevant in this case, did you run the app through Xcode? Is there a stacktrace shown on the left?

bglgwyng, 2023-09-25:
@mrousavy I didn't touch the existing [babel.config.js](https://github.com/mrousavy/react-native-vision-camera/blob/9dd91a40014492acd90b98171839843ee490b6ae/package/example/babel.config.js
) at all.

Here are the logs from `bt` command. Is it more helpful?
```
(lldb) bt
* thread #5, queue = 'mrousavy/VisionCamera.video', stop reason = signal SIGABRT
  * frame #0: 0x00000001e14ed578 libsystem_kernel.dylib`__pthread_kill + 8
    frame #1: 0x0000000202368118 libsystem_pthread.dylib`pthread_kill + 268
    frame #2: 0x00000001a9a58178 libsystem_c.dylib`abort + 180
    frame #3: 0x00000002022a6bf8 libc++abi.dylib`abort_message + 132
    frame #4: 0x0000000202296428 libc++abi.dylib`demangling_terminate_handler() + 320
    frame #5: 0x000000019b5d1eb4 libobjc.A.dylib`_objc_terminate() + 160
    frame #6: 0x00000002022a5fbc libc++abi.dylib`std::__terminate(void (*)()) + 16
    frame #7: 0x00000002022a5f60 libc++abi.dylib`std::terminate() + 56
    frame #8: 0x000000019b5e7d04 libobjc.A.dylib`objc_terminate + 16
    frame #9: 0x00000001a99f7ec0 libdispatch.dylib`_dispatch_client_callout + 40
    frame #10: 0x00000001a99fb330 libdispatch.dylib`_dispatch_continuation_pop + 504
    frame #11: 0x00000001a9a0e908 libdispatch.dylib`_dispatch_source_invoke + 1588
    frame #12: 0x00000001a99ff40c libdispatch.dylib`_dispatch_lane_serial_drain + 372
    frame #13: 0x00000001a9a000a4 libdispatch.dylib`_dispatch_lane_invoke + 384
    frame #14: 0x00000001a9a0acdc libdispatch.dylib`_dispatch_workloop_worker_thread + 648
    frame #15: 0x0000000202361ddc libsystem_pthread.dylib`_pthread_wqthread + 288
    ```

mrousavy, 2023-09-25:
Hm, idk there are no native changes since REA 3.5.0, and I remember that 3.5.0 worked at some point. Maybe there is something wrong happening in the worklets, I can try to debug this once I'm done with the new devices API/docs!

chrfalch, 2023-09-25:
Could it be that the new version has some updates in the babel plugin architecture (which will be automatically updated by upgrading Reanimated) - or that having two babel plugins for worklets makes for some undefined behaviour? 

Could you remove the worklet plugin after installing Reanimated and report back with your findings? (in the babel.config.js file).

bglgwyng, 2023-09-26:
@chrfalch I've already tried switching the order of plugins, but the crash still occurred.
> Could you remove the worklet plugin after installing Reanimated and report back with your findings?

Does this mean leaving `react-native-reanimated/plugin` only in babel.config.js?

chrfalch, 2023-09-26:
Yes, that was the suggestion. The theory was that the two might be interfering with each other.

mrousavy, 2023-09-27:
Okay then it might be a REA compatibility thing? We need to check what has changed since their last release.

Alternatively, you could downgrade REA but that's also not an ideal solution.

It sucks that we have to have two separate Worklet implementations :/

jackster-humi, 2023-09-27:
+1 Just ran into this as well. 

Had issues with REA v `3.5.4` when attempting an upgrade.

I tested with REA v `3.4.2` and had no issues.

carlgrob5171989, 2023-09-29:
REA v 3.5.0 mean  "react-native-reanimated": "^3.5.0" ? @jackster-humi

carlgrob5171989, 2023-09-29:
"react-native-reanimated": "^3.4.2",
"react-native-vision-camera": "^2.15.4",
"vision-camera-code-scanner": "^0.2.0"
in react native 0.69.0

jackster-humi, 2023-09-29:
Yes, that's correct, I was referring to `reanimated` version `3.5.0`.

I've written a face detector `frameProcessor` using Apple Vision with `vision-camera` `3.1.0` and saw the crash described in this thread.

I'll try to spend some time this weekend getting a good stracktrace + minimum reproducible example.

I'm now also getting a similar crash when upgrading reanimated from `2.12.0` to `3.4.2`, using the following (can't be sure it's the same issue but it's also a SIGABRT error):

```
    "react-native-vision-camera": "^3.1.0",
    "react-native-worklets-core": "^0.2.0",
    "react-native-reanimated": "3.4.2",
    "react-native": "0.71.13",
```

bglgwyng, 2023-10-17:
I encountered an interesting yet positive occurrence. I updated my RNVC version from 3.2.2 to 3.4.0 while keeping Reanimated at 3.3.0. This led to a crash in my app. However, when I examined the crash log, it seemed strangely familiar, and I recalled encountering the same error message before.

In an attempt to resolve the issue, I decided to upgrade Reanimated to 3.5.4, hoping for a stroke of luck. Surprisingly, it resolved the problem. To sum it up:

RNVC 3.2.2 + Reanimated 3.3.0 → Functioning properly
RNVC 3.2.2 + Reanimated 3.5.4 → Caused a crash
RNVC 3.4.0 + Reanimated 3.3.0 → Caused a crash
RNVC 3.4.0 + Reanimated 3.5.4 → Functioning properly

mrousavy, 2023-10-17:
Gotcha, I think I know a fix @bglgwyng

mrousavy, 2023-10-17:
So I think the SWM team renamed this again in their babel plugin:

[image]

This is how react-native-worklets-core works:

[image]

pvducTaureauAI, 2023-10-17:
@bglgwyng and @mrousavy, do you have any example that frameProcessor works? i need your configuration for react-native-camera with REA and RNWORKLET

bglgwyng, 2023-10-17:
@pvducTaureauAI Have you tried the example app in the package? Which platform do you have problems with?

carlgrob5171989, 2023-10-17:
https://github.com/mrousavy/react-native-vision-camera/issues/1802

does it work?

mrousavy, 2023-10-18:
@chrfalch just fixed this issue in this PR (I think) - we'll test more then release an update! https://github.com/margelo/react-native-worklets-core/pull/128

closed 2023-11-22 by bglgwyng