margelo/react-native-vision-camera #1850: 🐛 V3 Convert Frame to byte array nv21Data

basharmalkawi, 2023-09-26:
### What's happening?

I have this  dependencies 
    "react-native-reanimated": "^3.5.3",
    "react-native-vision-camera": "^3.0.0",
    "react-native-worklets-core": "^0.2.0"

, I try to convert frame to  byte [] nv21Data 
        @SuppressLint("UnsafeOptInUsageError") Image imageObj = frame.getImage();
        Plane[] planes = imageObj.getPlanes();
        ByteBuffer yBuffer = planes[0].getBuffer();
        ByteBuffer uBuffer = planes[1].getBuffer();
        ByteBuffer vBuffer = planes[2].getBuffer();
        int ySize = yBuffer.remaining();
        int uSize = uBuffer.remaining();
        int vSize = vBuffer.remaining();
        byte[] yData = new byte[ySize];
        byte[] uData = new byte[uSize];
        byte[] vData = new byte[vSize];
        yBuffer.get(yData);
        uBuffer.get(uData);
        vBuffer.get(vData);

but always I got the error: Cannot convert Frame with Format 34 to byte array! 

### Reproduceable Code

```tsx
const [camera, setCamera] = useState(false)
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const faces = documentRecognition(frame)
    console.log('---before---',faces)

  }, [])

  const devices = useCameraDevices()
  const device = devices.back 

  if (device == null) {
    return <View />;
  }
  return (
    <View style={{flex:1}} >
      <TouchableOpacity title={'Open Camera'} onPress={() => setCamera(true)} style={{backgroundColor: 'red', marginTop: 100, padding: 100, zIndex: 2}}>
        <Text>Open</Text>
      </TouchableOpacity>
      {camera&&  <Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        isActive={camera}
        device={device}
        torch={"on"}
        zoom={0}
      />}
    
    </View>
  )
```

### Relevant log output

```shell
error: Cannot convert Frame with Format 34 to byte array!
```

### Camera Device

```json
test
```

### Device

Samsung s23

### VisionCamera Version

"react-native-vision-camera": "^3.0.0"

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-26:
Your code does not throw that error, and you didnt share the full code.

closed 2023-09-26 by mrousavy

basharmalkawi, 2023-09-26:
> Your code does not throw that error, and you didnt share the full code.

this my code :
   @Override
    public Object callback(Frame frame, Map<String, Object> arguments) {
        Bitmap bitmap = convertImageProxyToBitmap(frame);
        return "test";
    }

    private Bitmap convertImageProxyToBitmap(Frame frame) {
        @SuppressLint("UnsafeOptInUsageError") Image imageObj = frame.getImage();

        Plane[] planes = imageObj.getPlanes();

        ByteBuffer yBuffer = planes[0].getBuffer();
        ByteBuffer uBuffer = planes[1].getBuffer();
        ByteBuffer vBuffer = planes[2].getBuffer();
        int ySize = yBuffer.remaining();
        int uSize = uBuffer.remaining();
        int vSize = vBuffer.remaining();
        byte[] yData = new byte[ySize];
        byte[] uData = new byte[uSize];
        byte[] vData = new byte[vSize];
        yBuffer.get(yData);
        uBuffer.get(uData);
        vBuffer.get(vData);
        int srcWidth = frame.getWidth();
        int srcHeight = frame.getHeight();
        int dstWidth = srcWidth;
        int dstHeight = srcHeight;
        boolean mirror = false;
        byte[] nv21Data = Utils.concatenateYUV(yData, uData, vData);
      //  Bitmap bitmap = TTVOCRSdk.yuvToBitmap(nv21Data, srcWidth, srcHeight, dstWidth, dstHeight, 90, mirror);
        return bitmap;
    }

rodgomesc, 2023-09-26:
hmm, also experiencing this, `imageObj.getPlanes();` returns an empty array

mrousavy, 2023-09-27:
I _think_ I can use the GPU HardwareBuffer APIs and get data even for private formats - this is much more reliable but as far as I remember it was slow for some reason. Will try in a new PR soon

genesiscz, 2023-11-17:
@mrousavy I see that you now expose hardwareBuffer in Frame, but how would one get the planes/Bitmap out of it?

rodgomesc, 2023-11-18:
`frame.image` is just a `android.media.Image` in format `private` or `YUV_420_888` you need to create  a bitmap from it you need to parse the image for `ARGB` first and then use `Bitmap.createBitmap` from `android.graphics.Bitmap`

genesiscz, 2023-11-18:
"yuv" is crashing on testing physical xiaomi redmi 5g even on newest version, "private", as far as I know, doesn't have planes. Is there a tutorial somewhere for this? I am digging around for hours and kinda lost here.

rodgomesc, 2023-11-18:
if you just need to access the planes you can do `frame.image.planes` but be aware that this is `yuv`

genesiscz, 2023-11-18:
I am rewriting the plugin our team wrote for v2 of vision-camera and it worked flawlessly. Could you share more of the code maybe? Would be greatly appreciated

rodgomesc, 2023-11-18:
> w, doesn't have planes. Is there a tutorial somewhere for this? I am digging around for hours and kinda lost here.

sorry, there's not straightforward way of doing that but there's a lot of tutorials on internet and i provided all keywords you should need to figure out the solution

rodgomesc, 2023-11-18:
@genesiscz let's not spam here, please  join t discord server and try to provide a initial code to show your problem  https://discord.com/invite/6CSHz2qAvA

genesiscz, 2023-11-18:
Is "yuv" pixelFormat working for you? To make it all worse, toByteArray was somehow removed from the Frame :(

mrousavy, 2023-11-20:
@genesiscz could you try replacing this:

https://github.com/mrousavy/react-native-vision-camera/blob/deb3c41ee0a55e64cbb83ead912ccdd53d10988f/package/android/src/main/java/com/mrousavy/camera/core/VideoPipeline.kt#L97-L107

with

```kt
imageReader = ImageReader.newInstance(width, height, format, MAX_IMAGES)
imageWriter = null
```

Note that this will break video recording functionality, but for only Frame Processing it should work.

I am still not sure if it is even possible on Android to implement both Frame Processing and Video Recording in one, as the `ImageReader` + `ImageWriter` setup I have here just sometimes fails on some devices - as you noticed on your Xiamoi Redmi 5 @genesiscz. And there's no other way to get the GPU buffers apparently. 

Please play around here and let me know your findings!