margelo/react-native-vision-camera #1851: ✨ RAW (Apple ProRAW/Bayer) support ($2,000)

mrousavy, 2023-09-26:
### What feature or enhancement are you suggesting?

It would be great to be able to capture RAW images (Apple ProRAW or Bayer) in VisionCamera. See [this apple guide](https://developer.apple.com/documentation/avfoundation/photo_capture/capturing_photos_in_raw_and_apple_proraw_formats).

For $2,000, I will implement RAW support in VisionCamera.

### What Platforms whould this feature/enhancement affect?

iOS, Android

### Alternatives/Workarounds

RAW Capture is not possible today.

### Additional information

- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar feature requests in this repository](https://github.com/mrousavy/react-native-vision-camera/issues?q=is%3Aopen+is%3Aissue+label%3A%22✨+enhancement%22) and found none.

angelo-hub, 2024-04-19:
if i implement it can i get that?

mrousavy, 2024-04-20:
No, because no one donated to the pool yet. It's at $0.

mrousavy, 2024-10-29:
> Is there any update on this?

Again, this issue is a sponsor pool. Once it's full, I'll implement this.

stefanoschrs, 2026-03-13:
Any updates on this?