margelo/react-native-vision-camera #1852: 🐛 V3 Convert Frame to byte array nv21Data

basharmalkawi, 2023-09-26:
### What's happening?

I have this dependencies
"react-native-reanimated": "^3.5.3",
"react-native-vision-camera": "^3.0.0",
"react-native-worklets-core": "^0.2.0"

I try to convert frame to byte [] nv21Data  , but always I got the error: Cannot convert Frame with Format 34 to byte array!

### Reproduceable Code

```tsx
react native code : 
import React, { useState } from 'react';
import { SafeAreaView, Text, View, StyleSheet, Button, TouchableOpacity } from 'react-native';
import {documentRecognition} from './src/Helpers';
import { Camera, runAsync, useCameraDevices, useFrameProcessor } from 'react-native-vision-camera';

export const App = () => {

  const [camera, setCamera] = useState(false)
  const frameProcessor = useFrameProcessor((frame) => {
    'worklet';
    const faces = documentRecognition(frame)
    console.log('---before---',faces)

  }, [])

  const devices = useCameraDevices()
  const device = devices.back 

  if (device == null) {
    return <View />;
  }
  return (
    <View style={{flex:1}} >
      <TouchableOpacity title={'Open Camera'} onPress={() => setCamera(true)} style={{backgroundColor: 'red', marginTop: 100, padding: 100, zIndex: 2}}>
        <Text>Open</Text>
      </TouchableOpacity>
      {camera&&  <Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        isActive={camera}
        device={device}
        torch={"on"}
        zoom={0}
      />}
    
    </View>
  )
}
=======
native android code :
public class documentRecognitionFrameProcessorPlugin extends FrameProcessorPlugin {
 

    @Override
    public Object callback(Frame frame, Map<String, Object> arguments) {
        Bitmap bitmap = convertImageProxyToBitmap(frame);
        return "test";
    }

    private Bitmap convertImageProxyToBitmap(Frame frame) {
        @SuppressLint("UnsafeOptInUsageError") Image imageObj = frame.getImage();

        Plane[] planes = imageObj.getPlanes();

        ByteBuffer yBuffer = planes[0].getBuffer();
        ByteBuffer uBuffer = planes[1].getBuffer();
        ByteBuffer vBuffer = planes[2].getBuffer();
        int ySize = yBuffer.remaining();
        int uSize = uBuffer.remaining();
        int vSize = vBuffer.remaining();
        byte[] yData = new byte[ySize];
        byte[] uData = new byte[uSize];
        byte[] vData = new byte[vSize];
        yBuffer.get(yData);
        uBuffer.get(uData);
        vBuffer.get(vData);
        int srcWidth = frame.getWidth();
        int srcHeight = frame.getHeight();
        int dstWidth = srcWidth;
        int dstHeight = srcHeight;
        boolean mirror = false;
        byte[] nv21Data = Utils.concatenateYUV(yData, uData, vData);
      //  Bitmap bitmap = TTVOCRSdk.yuvToBitmap(nv21Data, srcWidth, srcHeight, dstWidth, dstHeight, 90, mirror);
        return bitmap;
    }

}
```

### Relevant log output

```shell
No response
```

### Camera Device

```json
No response
```

### Device

Samsung s23 , OPPO A5s

### VisionCamera Version

"react-native-vision-camera": "^3.0.0"

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

basharmalkawi, 2023-09-26:
I upgrade  "react-native-vision-camera": "^3.1.0",
 ,  I am using the same code mentioned above but changed this part

    private Bitmap convertImageProxyToBitmap(Frame frame) {
        @SuppressLint("UnsafeOptInUsageError") Image imageObj = frame.getImage();

        if (imageObj == null) {
            Log.e("Error", "Image is null");
            return null; // Handle the case where the image is null
        }

        Plane[] planes = imageObj.getPlanes();

        if (planes.length < 3) {
            Log.e("Error", "Not enough planes in the image");
            return null; // Handle the case where the expected number of planes is not available
        }

        ByteBuffer yBuffer = planes[0].getBuffer();
        ByteBuffer uBuffer = planes[1].getBuffer();
        ByteBuffer vBuffer = planes[2].getBuffer();
        int ySize = yBuffer.remaining();
        int uSize = uBuffer.remaining();
        int vSize = vBuffer.remaining();
        byte[] yData = new byte[ySize];
        byte[] uData = new byte[uSize];
        byte[] vData = new byte[vSize];
        yBuffer.get(yData);
        uBuffer.get(uData);
        vBuffer.get(vData);
        int srcWidth = frame.getWidth();
        int srcHeight = frame.getHeight();
        int dstWidth = srcWidth;
        int dstHeight = srcHeight;
        boolean mirror = false;
        byte[] nv21Data = Utils.concatenateYUV(yData, uData, vData);
        Bitmap bitmap = TTVOCRSdk.yuvToBitmap(nv21Data, srcWidth, srcHeight, dstWidth, dstHeight, 90, mirror);
        return bitmap;
    }
every time print   Log.e("Error", "Not enough planes in the image"); 
when I remove it I got
 JSI rethrowing JS exception: Exception in HostFunction: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0
                                                                                                    
                                                                                                    Error: Exception in HostFunction: java.lang.ArrayIndexOutOfBoundsException: length=0; index=0 ,
 in  "react-native-vision-camera": "^2.15.4" I use it with ImageProxy it work fine , why with frame it not work , do you have any solution ?

jjbwii, 2023-09-26:
Hello @basharmalkawi,

At the current state of the V3 improvement we have no ImageProxy over the Frame. 

New V3 version are using the Standard Camera 2 low Level API instead of the high level wrapping API (with low flexibility , low performance) used with "react-native-vision-camera": "^2.XX.X".

I have no time to test new 3.1.0 but reading some forum post you should configure the Camera with ‘rgb’ option not Format 34.

In theory, with the rgb option you should receive the planes. Could you try this option and share the results?

Regards

basharmalkawi, 2023-09-27:
Hello @jjbwii  , 
I tried your suggestion but  I got the Camera preview is a black screen , do you have any solution .
<Camera
        frameProcessor={frameProcessor}
        style={StyleSheet.absoluteFill}
        isActive={true}
        device={device}
        torch={"off"}
        zoom={0}
        pixelFormat={'rgb'}
      />
Regards

jjbwii, 2023-09-27:
Hello,

You must replace <Camera> with <ReanimatedCamera> it works for me (with 3.0.0 version and frontal camera)

see the example below:

https://github.com/mrousavy/react-native-vision-camera/blob/main/package/example/src/CameraPage.tsx

mrousavy, 2023-09-27:
I can use HardwareBuffer instead of planes which should work for every type - will try this in a PR soon

basharmalkawi, 2023-09-27:
@jjbwii  
hello sir , 
I replaced it by 'ReanimatedCamera' , but still the same issue

jjbwii, 2023-09-27:
Hello @basharmalkawi,

I use this configuration with 3.0.0 and it worked with my four years old mobile phone (cheap chinesse processor)

const devices = useCameraDevices();
const device = devices.front;

<ReanimatedCamera
          style={StyleSheet.absoluteFill}
          device={device}
          isActive={true}
          frameProcessor={frameProcessor}
          pixelFormat={"rgb"}
          zoom={0.5}
          photo={true}
          video={true}
        />

Which mobile phone are you using? Samsung perhaps…

basharmalkawi, 2023-09-28:
@jjbwii  
hello , 
this Device
 OPPO A5s

mrousavy, 2023-09-30:
This should be fixed in 3.2.0 as I now use the GPU HardwareBuffer directly!

closed 2023-09-30 by mrousavy