margelo/react-native-vision-camera #1854: 🔧 Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.

VGabriel45, 2023-09-26:
### How were you trying to build the app?

Using 'yarn android' (react-native run-android)

### Full build logs

```tsx
Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:processDebugResources'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/gabi/Desktop/Work/POP/rn-app/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/gabi/Desktop/Work/POP/rn-app/node_modules/jsc-android/dist/com/facebook/react/react-android/maven-metadata.xml
       - https://maven.pkg.github.com/Banuba/banuba-ve-sdk/com/facebook/react/react-android/maven-metadata.xml
       - https://github.com/Banuba/banuba-ar/com/facebook/react/react-android/maven-metadata.xml
       - https://dl.google.com/dl/android/maven2/com/facebook/react/react-android/maven-metadata.xml
       - https://www.jitpack.io/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/gabi/Desktop/Work/POP/rn-app/node_modules/@notifee/react-native/android/libs/com/facebook/react/react-android/maven-metadata.xml
     Required by:
         project :app > project :react-native-vision-camera
```

### Project dependencies

```json
"dependencies": {
    "@apollo/client": "^3.7.14",
    "@bam.tech/react-native-image-resizer": "^3.0.5",
    "@enzoferey/ethers-error-parser": "^0.2.3",
    "@ethersproject/shims": "^5.6.0",
    "@notifee/react-native": "^7.7.1",
    "@phinx-app/phinx-auth": "^1.1.8",
    "@react-native-async-storage/async-storage": "^1.19.0",
    "@react-native-camera-roll/camera-roll": "5.3.1",
    "@react-native-clipboard/clipboard": "^1.11.1",
    "@react-native-community/blur": "^4.3.0",
    "@react-native-community/geolocation": "^3.0.6",
    "@react-native-community/netinfo": "^9.4.1",
    "@react-native-community/push-notification-ios": "^1.11.0",
    "@react-native-community/slider": "^4.4.3",
    "@react-native-firebase/analytics": "17.4.3",
    "@react-native-firebase/app": "17.4.3",
    "@react-native-firebase/messaging": "17.4.3",
    "@react-navigation/bottom-tabs": "^6.3.3",
    "@react-navigation/native": "^6.0.11",
    "@react-navigation/native-stack": "^6.7.0",
    "@sendbird": "http://git.popoo.foundation/app/sendbird.git",
    "@sendbird/react-native-scrollview-enhancer": "^0.2.0",
    "@shopify/flash-list": "^1.4.3",
    "@tanstack/react-query": "^4.28.0",
    "@toruslabs/react-native-web-browser": "^1.1.0",
    "@walletconnect/modal-react-native": "^1.0.0-rc.9",
    "@web3auth/react-native-sdk": "^3.5.0",
    "ahooks": "^3.7.4",
    "appcenter": "^5.0.0",
    "appcenter-analytics": "^5.0.0",
    "appcenter-crashes": "^5.0.0",
    "axios": "^0.27.2",
    "babel-plugin-root-import": "^6.6.0",
    "babel-plugin-transform-remove-console": "^6.9.4",
    "eslint-plugin-react-hooks": "^4.6.0",
    "ethereum-blockies-base64": "^1.0.2",
    "ethers": "5.7.2",
    "events": "^3.3.0",
    "firebase": "^9.13.0",
    "graphql": "^16.6.0",
    "i18n-js": "^4.1.1",
    "i18next": "^22.5.0",
    "i18next-http-backend": "^2.2.1",
    "is-ipfs": "^8.0.1",
    "lodash": "^4.17.21",
    "lottie-react-native": "^5.1.4",
    "metro-config": "^0.76.2",
    "moment": "^2.29.4",
    "nativewind": "^2.0.11",
    "node-libs-browser": "^2.2.1",
    "patch-package": "^7.0.0",
    "react": "18.1.0",
    "react-i18next": "^12.3.1",
    "react-native": "0.70.6",
    "react-native-adjust": "^4.33.0",
    "react-native-animatable": "^1.3.3",
    "react-native-camera-kit": "^13.0.0",
    "react-native-circular-progress": "^1.3.7",
    "react-native-code-push": "^7.0.2",
    "react-native-color-matrix-image-filters": "^6.0.3",
    "react-native-compressor": "^1.6.1",
    "react-native-country-picker-modal": "^2.0.0",
    "react-native-create-thumbnail": "^1.6.4",
    "react-native-crypto": "^2.1.0",
    "react-native-date-picker": "^4.2.13",
    "react-native-device-info": "^10.3.0",
    "react-native-document-picker": "^8.2.0",
    "react-native-elements": "^3.4.3",
    "react-native-fast-image": "^8.6.3",
    "react-native-file-access": "^2.6.0",
    "react-native-fs": "^2.20.0",
    "react-native-gesture-handler": "^2.6.1",
    "react-native-get-random-values": "^1.9.0",
    "react-native-image-crop-picker": "^0.38.0",
    "react-native-image-picker": "4.10.0",
    "react-native-image-zoom-viewer": "^3.0.1",
    "react-native-linear-gradient": "^2.6.2",
    "react-native-localize": "^2.2.3",
    "react-native-modal": "^13.0.1",
    "react-native-orientation-locker": "^1.5.0",
    "react-native-os": "^1.2.6",
    "react-native-pager-view": "6.1.3",
    "react-native-permissions": ">=3.8.0",
    "react-native-pressable-opacity": "^1.0.10",
    "react-native-qrcode-svg": "^6.2.0",
    "react-native-randombytes": "^3.0.0",
    "react-native-reanimated": "^2.12.0",
    "react-native-render-html": "^6.3.4",
    "react-native-root-siblings": "^4.1.1",
    "react-native-root-toast": "^3.4.0",
    "react-native-safe-area-context": "^4.4.1",
    "react-native-screens": "^3.20.0",
    "react-native-shadow": "^1.2.2",
    "react-native-share": "^8.2.1",
    "react-native-splash-screen": "^3.3.0",
    "react-native-static-safe-area-insets": "^2.2.0",
    "react-native-svg": "^13.9.0",
    "react-native-swipe-list-view": "^3.2.9",
    "react-native-swiper": "^1.6.0",
    "react-native-tab-view": "^3.5.2",
    "react-native-tcp": "^4.0.0",
    "react-native-toast-message": "^2.1.6",
    "react-native-ui-lib": "^6.28.1",
    "react-native-uuid": "^2.0.1",
    "react-native-vector-icons": "^10.0.0",
    "react-native-version-check": "^3.4.7",
    "react-native-video": "^5.2.1",
    "react-native-view-shot": "^3.6.0",
    "react-native-vision-camera": "3.1.2",
    "react-native-webview": "^13.2.2",
    "readable-stream": "^4.3.0",
    "rn-placeholder": "^3.0.3",
    "stream-browserify": "^3.0.0",
    "url": "^0.11.0",
    "zustand": "^4.1.1"
  },
```

### VisionCamera Version

3.1.2

### Target platforms

Android

### Operating system

MacOS

### Can you build the VisionCamera Example app?

Yes, I can successfully build the Example app here

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

zahrat, 2023-09-27:
same issue when upgrade to "react-native-vision-camera": "^3.1.0"

mrousavy, 2023-09-27:
There is no VisionCamera 3.1.2.
[image]

closed 2023-09-27 by mrousavy

zahrat, 2023-09-27:
@mrousavy this error happened for version 3.1.0 and react-native version 0.70.5

jwoodmansey, 2023-10-04:
Seeing this on the latest 

```
"react-native-vision-camera": "^3.3.0",
"react-native": "0.70.13",
```

fvalles, 2023-10-06:
Same error when trying to build android app 

```
* What went wrong:
Could not determine the dependencies of task ':app:processDebugResources'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not find any matches for com.facebook.react:react-android:+ as no versions of com.facebook.react:react-android are available.
     Searched in the following locations:
       - file:/Users/fvalles/Repositories/chefs-app/node_modules/react-native/android/com/facebook/react/react-android/maven-metadata.xml
       - file:/Users/fvalles/Repositories/chefs-app/node_modules/jsc-android/dist/com/facebook/react/react-android/maven-metadata.xml
       - https://dl.google.com/dl/android/maven2/com/facebook/react/react-android/maven-metadata.xml
       - https://www.jitpack.io/com/facebook/react/react-android/maven-metadata.xml
     Required by:
         project :app > project :react-native-vision-camera
```

with the following versions:

```
"react-native-vision-camera": "^3.3.1",
"react-native": "0.70.9",
```

hardik-javascript, 2023-10-17:
Any solutions i am getting same error with `react-native: 0.70.6` and `react-native-vision-camera: 3.3.1`?

mrousavy, 2023-10-20:
you need react-native 0.71 or higher

jibarozzibehr, 2023-11-01:
What's a good version of react-native-vision-camera v3 for react-native 0.70.0?

jibarozzibehr, 2023-11-01:
> you need react-native 0.71 or higher

What's a good version of react-native-vision-camera v3 for react-native 0.70.0?

hardik-javascript, 2023-11-02:
@jibarozzibehr I can confirm `react-native-vision-camera: 2.16.2` working fine with react native 0.70.1

DarkcoderSe, 2023-11-24:
React native 0.70.* will work with vision-camera@2, here you go
https://www.npmjs.com/package/react-native-vision-camera/v/2.16.2

EL-MEHDI-ESSAADI, 2023-12-07:
@mrousavy you should mention that react native vision camera v3 require 0.71 RN o higher in the README o docs so people wouldn't lose time trying to upgrade to v3 if they have less than 0.71 RN

chellongCoder, 2023-12-11:
any solution for react native 0.64? :(

EL-MEHDI-ESSAADI, 2024-01-04:
@chellongCoder I think the only solution is upgrading

MSerj, 2024-04-11:
same issue on **react-native v0.69.12**

msaiducar64, 2024-04-27:
any solution
 "react-native": "0.70.5",
"react-native-vision-camera": "^3.9.1",

charleyoliveira, 2025-02-03:
any solution for 
"react-native": "0.69.6",
"react-native-vision-camera": "^3.9.1",