margelo/react-native-vision-camera #1858: 🐛 Why Camera preview is a black screen?

pepperav, 2023-09-26:
### What's happening?

Hi 👋,
first of all, I want to thank you @mrousavy in advance for your incredible work! 👏

Now my question: why I'm getting a black screen when app starts?

**package.json**
```
{
  "name": "visioncamera",
  "version": "1.0.0",
  "main": "node_modules/expo/AppEntry.js",
  "scripts": {
    "start": "expo start",
    "android": "expo start --android",
    "ios": "expo start --ios",
    "web": "expo start --web"
  },
  "dependencies": {
    "@types/react": "~18.2.14",
    "expo": "~49.0.12",
    "expo-dev-client": "~2.4.11",
    "expo-status-bar": "~1.6.0",
    "react": "18.2.0",
    "react-native": "0.72.5",
    "react-native-vision-camera": "^3.1.0",
    "react-native-worklets-core": "^0.2.0",
    "typescript": "^5.1.3"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0"
  },
  "private": true
}
```

### Reproduceable Code

```tsx
import { Camera, getCameraFormat, useCameraDevice, useFrameProcessor } from 'react-native-vision-camera';
import { useEffect } from 'react';
import { StyleSheet, Text, View } from 'react-native';
import { StatusBar } from 'expo-status-bar';

export default function App() {
	const device = useCameraDevice('back');

	useEffect(() => {
		const cameraPermissions = async () => {
			const cameraPermission = await Camera.getCameraPermissionStatus();
			console.log("cameraPermission", cameraPermission);
			await Camera.requestCameraPermission();
		};

		cameraPermissions();
	}, []);

	if (device == null) return <View style={{ flex: 1, justifyContent: 'center', alignItems: 'center' }}><Text>Loading...</Text></View>

	return (
		<>
			<StatusBar style='light' />
			<Camera
				style={StyleSheet.absoluteFill}
				device={device}
				isActive={true}
			/>
		</>
	);
}
```

### Relevant log output

```shell
09-26 13:35:18.077 26277 26277 D ApkSoSource: not allowing consideration of lib/arm64-v8a/libVisionCamera.so: deferring to libdir
09-26 13:35:40.324 26277 26643 I VisionCameraProxy: Initializing VisionCameraProxy...
09-26 13:35:40.324 26277 26643 I VisionCameraProxy: Creating Worklet Context...
09-26 13:35:40.330 26277 26643 I VisionCameraProxy: Worklet Context created!
```

### Camera Device

```json
{
  "sensorOrientation": "landscape-right",
  "hardwareLevel": "full",
  "maxZoom": 8,
  "minZoom": 1,
  "supportsLowLightBoost": false,
  "neutralZoom": 1,
  "physicalDevices": [
    "wide-angle-camera"
  ],
  "supportsFocus": true,
  "supportsRawCapture": true,
  "isMultiCam": false,
  "name": "BACK (0)",
  "hasFlash": true,
  "hasTorch": true,
  "position": "back",
  "id": "0"
}
```

### Device

RedMi - MIUI Global 12.5.1

### VisionCamera Version

3.1.0

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [X] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-27:
Hey! Thanks man.

This sounds like a classic android issue. Can you try commenting out line by line here and see what makes it fail? https://github.com/mrousavy/react-native-vision-camera/blob/fb6ebd9ee1b831ce1c2e0b42d123257f5f9c5459/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L461-L492

pepperav, 2023-09-27:
Ehi! Thanks for the kind reply.

I'm pretty new here so can you guide me a little bit more? I am in an Expo project, so, if I've understood well, I have to do:

1. Find `CameraSession.kt` file into `node_modules/react-native-vision-camera` package
2. Starting to comment first couple of lines (eg: 462 and 463 for FPS)
3. Launch a new build using EAS CLI in order to generate a new expo dev client .APK
4. Run it and test
5. Iterate for all lines

Is it right?
Thank you 👋

mrousavy, 2023-09-27:
Hey - yep!
I don't know how it'd work in Expo, but I guess that's how you do it yea.

Runnjng yarn android won't rebuild, does it? cc @EvanBacon

pepperav, 2023-09-27:
Unfortunately my proposal does not work! 😭 The command `eas build` command triggers a pipeline in the cloud witch download fresh `node_modules` remotely, ignoring my local edits with commented lines (of course! 😓).

I've tried to build locally as well but I got strange errors such as:
[image]

I'm stuck! Any suggestions ?

mrousavy, 2023-09-28:
Hey!
If the issue is still there after commenting all lines in there then I haven't seen it before.

I need the full adb logcat output, and a way to reproduce (but I don't have that phone)

idrissakhi, 2023-09-28:
@mrousavy i took the time to debug this issue, it's only happening on android 10 and less so in the code we need to adapt in two places => preview (CameraSession) & photoCapture (CameraDevice+createPhotoCaptureRequest) and the issue is related to the Zoom utility you have created, it's changing the coordinates of the Rect even if zoom == 1.0f so i have tested with a small if to resolve it, and now it's working, but the zoom is broken here's what i have done

@pepperav FYI
`
      val cameraCharacteristics = cameraManager.getCameraCharacteristics(cameraId!!)
      val size = cameraCharacteristics.get(CameraCharacteristics.SENSOR_INFO_ACTIVE_ARRAY_SIZE)!!
      if (zoom != 1.0f) {
        captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
      } else {
        captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size)
      }
`

pepperav, 2023-09-28:
ehi! tnx @idrissakhi for your insight! 🏆

I have updated `CameraSession.kt` file with your tip:

```
    // Zoom
    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.R) {
      captureRequest.set(CaptureRequest.CONTROL_ZOOM_RATIO, zoom)
    } else {
      val cameraCharacteristics = cameraManager.getCameraCharacteristics(cameraId!!)
      val size = cameraCharacteristics.get(CameraCharacteristics.SENSOR_INFO_ACTIVE_ARRAY_SIZE)!!
      
      //captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
      
      if (zoom != 1.0f) {
        captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
      } else {
        captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size)
      }
    }
```

But the black screen on preview is still there 😓.

Am I miss something else?

@mrousavy I'm going to save the full adb logcat

idrissakhi, 2023-09-28:
@pepperav you have a black screen after taking a picture ? if it's the case you need to do the same thing on CameraDevice+createPhotoCaptureRequest line 93 

```
captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
```

pepperav, 2023-09-28:
> @pepperav you have a black screen after taking a picture ? if it's the case you need to do the same thing on CameraDevice+createPhotoCaptureRequest line 93
> 
> ```
> captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
> ```

nop, always! Since the app is started (all permissions granted), in the preview of camera...check my `App.tsx` file

pepperav, 2023-09-28:
Here the full log. Hope it should be enough to understand the issue

[black_screen_log.txt](https://github.com/mrousavy/react-native-vision-camera/files/12749379/black_screen_log.txt)

igperez-ar, 2023-09-28:
It worked for me by commenting line **487**. How can it be resolved? Do I need to use [patch-package](https://www.npmjs.com/package/patch-package)?

```kotlin
    // Zoom
    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.R) {
      captureRequest.set(CaptureRequest.CONTROL_ZOOM_RATIO, zoom)
    } else {
      val cameraCharacteristics = cameraManager.getCameraCharacteristics(cameraId!!)
      val size = cameraCharacteristics.get(CameraCharacteristics.SENSOR_INFO_ACTIVE_ARRAY_SIZE)!!
      // --> captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
    }
```

I'm coming from version **2.15.4** to **3.1.0** hoping to resolve the error `java.lang.IllegalStateException - Surface was requested when the Recorder had encountered error c.c.a.l1.l: c.c.a.l1.o.a1: Encoder cannot created :null` which was a crash happening on some devices, but I still don't know if it's stable enough to make the change.

## Device
Motorola G8 Plus - Android 10

## VisionCamera Version
3.1.0

## Can you reproduce this issue in the VisionCamera Example app?
No, I cannot reproduce the issue in the Example app

## Additional information
- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [x] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [x] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [x] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

pepperav, 2023-09-29:
> It worked for me by commenting line **487**. How can it be resolved? Do I need to use [patch-package](https://www.npmjs.com/package/patch-package)?
>
> ```kotlin
>     // Zoom
>     if [...]

I've tried commenting line 487 but I get always black screen 😓

idrisssakhi, 2023-09-29:
@pepperav you need to add formats for android => 
        <AnimatedCamera
          testID={`${testID}`}
          ref={cameraRef}
          style={styles.cameraView}
          device={device}
          isActive={isActive}
          photo
          animatedProps={animatedProps}
          orientation="portrait"
          {...(isAndroid && { format: format })}
        />

pepperav, 2023-09-29:
Ehi @idrisssakhi , thanks but nothing happens. 

Check my code below:
```
import { Camera, useCameraDevice, useCameraFormat, useFrameProcessor } from 'react-native-vision-camera';
import { useEffect, useRef } from 'react';
import { Platform, StyleSheet, Text, TouchableOpacity, View } from 'react-native';
import { StatusBar } from 'expo-status-bar';

export default function App() {
	const device = useCameraDevice('back');

	const camera = useRef<Camera>(null);

	const format = useCameraFormat(device, [
		{ videoResolution: 'max' },
  		{ photoResolution: 'max' },
  		{ fps: 30 }
	]);

	useEffect(() => {
		const cameraPermissions = async () => {
			const cameraPermission = await Camera.getCameraPermissionStatus();
			console.log("cameraPermission", cameraPermission);
			await Camera.requestCameraPermission();

			const microphonePermission = await Camera.getMicrophonePermissionStatus();
			console.log("microphonePermission", microphonePermission);
			await Camera.requestMicrophonePermission();
		};

		cameraPermissions();
	}, []);

	if (device == null) 
		return 
			<View style={styles.flexCenter}>
				<Text style={{ fontSize: 16 }}>⌛️ Loading...</Text>
			</View>

	return (
		<>
			<StatusBar style='light' />

			<Camera
				ref={camera}
				style={StyleSheet.absoluteFill}
				device={device}
				isActive={true}
				photo
				{ ...(Platform.OS === 'android' && { format }) }
			/>
		</>
	);
};

const styles = StyleSheet.create({
	flexCenter: { 
		flex: 1, 
		justifyContent: 'center', 
		alignItems: 'center' 
	}
});
```

mrousavy, 2023-10-06:
Hey!

Quick update: I think there could be four problems here:

1. You are using a `limited`/`legacy` device, and use Preview (always on), `photo` and `video`/`frameProcessor` outputs (3 outputs) which are by default at maximum resolution. LIMITED/LEGACY devices might not support full-resolution streaming, see [the Android capture session documentation](https://developer.android.com/reference/android/hardware/camera2/CameraDevice#regular-capture) on supported configuration maps.
    We need `PRIV PREVIEW` + `YUV/PRIV RECORD` + `JPEG MAXIMUM`. anything lower, needs to be explicitly handled by me, but I can easily add that no problemo. I just need the device to test it on.
    [image]
    **What you can do:** Try disabling `photo`, `video` and `frameProcessor` and check if that works. If only PREVIEW is attached, everything should work.
2. We are using some setting in the Capture Request that is not supported (e.g. FPS, HDR, Zoom, ...) but we don't previously check if those settings are supported on the device. I can easily add some checks here, so that the Camera doesn't just swallow any errors and fails silently, but instead throws and says something like "60 FPS is not supported on this device". https://github.com/mrousavy/react-native-vision-camera/blob/fb6ebd9ee1b831ce1c2e0b42d123257f5f9c5459/package/android/src/main/java/com/mrousavy/camera/core/CameraSession.kt#L461-L492
     **What you can do:** Try disabling each line in `getPreviewCaptureRequest` and see what happens. I've heard `zoom` is often the cause.
3. Some manufacturers (cough* cough Samsung) explicitly say they support something (e.g. 60 FPS), but then don't support it and just swallow errors and show blackscreens when trying to use it. This is the most ridiculous sh*t I've ever seen, but yea, that's how it is.
    **What you can do:** Also try other steps here and remove code by code to see what it is. Maybe also browse similar issues in the [CameraX issue tracker](https://issuetracker.google.com/hotlists/1551123) or StackOverflow to see if someone had similar issues, explicitly by searching for your device.
4. There could be some synchronization issue. I can try to make everything fully atomic so nothing ever goes out of sync with mutexes. This is pretty tricky, but I'll do it

pepperav, 2023-10-09:
Tnx @mrousavy  for your full detailed response 🙏. I will do some tries but, as you said, this device is too old! I will move to another one (a Samsung Galaxy J7 - Android 9) which seems more promising.

pepperav, 2023-10-13:
I'm going to paste the camera2-probe report for both my Android devices. Do you see some helpful info?

```
Model:Redmi Note 7
Manufacturer:Xiaomi
Build:10
SDK:29
------------------------------------------
Camera ID: 0
Camera Facing: BACK
Hardware Support Level:LEVEL_3
Manual exposure:1
Auto exposure:1
Auto exposure, always flash:1
Auto exposure, auto flash:1
Auto exposure, auto flash, red eye reduction:1
AE Lock:1
Compensation Range: [-24, 24]
Compensation Step: 1/6
Target FPS  Range: [15, 15]
Target FPS  Range: [20, 20]
Target FPS  Range: [24, 24]
Target FPS  Range: [7, 30]
Target FPS  Range: [30, 30]
Manual focus:1
Auto focus:1
Auto focus macro:1
Auto focus continuous picture:1
Auto focus continuous video:1
Auto focus EDOF:0
White balance off:1
Automatic white balance:1
WB: cloudy day:1
WB: daylight:1
WB: fluorescent:1
WB: incandescent:1
WB: shade:1
WB: twilight:1
WB: warm fluorescent:1
AWB Lock:1
RawCapture:1
BurstCapture:1
availableCapability: BACKWARD_COMPATIBLE
availableCapability: MANUAL_SENSOR
availableCapability: MANUAL_POST_PROCESSING
availableCapability: READ_SENSOR_SETTINGS
availableCapability: BURST_CAPTURE
availableCapability: PRIVATE_REPROCESSING
availableCapability: YUV_REPROCESSING
availableCapability: CONSTRAINED_HIGH_SPEED_VIDEO
availableCapability: RAW
Stabilization Mode: Video
previewSize: 4000 * 3000 (4:3)
previewSize: 4000 * 2250 (16:9)
previewSize: 4000 * 1844 (1000:461)
previewSize: 3840 * 2160 (16:9)
previewSize: 2592 * 1944 (4:3)
previewSize: 2340 * 1080 (13:6)
previewSize: 2048 * 1536 (4:3)
previewSize: 1920 * 1080 (16:9)
previewSize: 1600 * 1200 (4:3)
previewSize: 1440 * 1080 (4:3)
previewSize: 1440 * 720 (2:1)
previewSize: 1280 * 960 (4:3)
previewSize: 1280 * 768 (5:3)
previewSize: 1280 * 720 (16:9)
previewSize: 1280 * 480 (8:3)
previewSize: 1280 * 400 (16:5)
previewSize: 800 * 480 (5:3)
previewSize: 720 * 480 (3:2)
[... 132 lines cut ...]
photoSize: 1280 * 720 (16:9)
photoSize: 960 * 720 (4:3)
photoSize: 720 * 480 (3:2)
photoSize: 640 * 480 (4:3)
photoSize: 352 * 288 (11:9)
photoSize: 320 * 240 (4:3)
photoSize: 256 * 144 (16:9)
photoSize: 176 * 144 (11:9)
videoSize: 4128 * 3096 (4:3)
videoSize: 4128 * 2322 (16:9)
videoSize: 3264 * 2448 (4:3)
videoSize: 3264 * 1836 (16:9)
videoSize: 3088 * 3088 (1:1)
videoSize: 2048 * 1536 (4:3)
videoSize: 2048 * 1152 (16:9)
videoSize: 1920 * 1080 (16:9)
videoSize: 1280 * 720 (16:9)
videoSize: 960 * 720 (4:3)
videoSize: 720 * 480 (3:2)
videoSize: 640 * 480 (4:3)
videoSize: 352 * 288 (11:9)
videoSize: 320 * 240 (4:3)
videoSize: 256 * 144 (16:9)
videoSize: 176 * 144 (11:9)
```

mrousavy, 2023-10-18:
Hey! Can anyone try with VisionCamera 3.4.1. and see if the issue still persists? Thanks!

pepperav, 2023-10-18:
Sure! I'm gonna try it asap and I will update you. Thanks

mrousavy, 2023-10-24:
Awesome! If you appreciate my work please consider sponsoring me on GitHub to say thanks.

closed 2023-10-24 by mrousavy