margelo/react-native-vision-camera #1862: [iOS v3] NSArray returned as object to JS 🐛

mnecchi, 2023-09-26:
### What's happening?

Just testing the example app with `v3.1.0` on iOS 17. I've added a `console.log` after calling the example plugin: 
```js
const frameProcessor = useFrameProcessor((frame) => {
    'worklet'

    console.log(`${frame.timestamp}: ${frame.width}x${frame.height} ${frame.pixelFormat} Frame (${frame.orientation})`)
    const result = examplePlugin(frame)
    console.log('Result: ', result)
  }, [])
```
This are the logs:
```
 LOG  Re-rendering camera page with inactive camera. Device: "Back Dual Wide Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
 LOG  147576488.926875: 1080x1920 yuv Frame (landscape-right)
 LOG  Result:  {"example_array": {"0": "Hello", "1": 1, "2": 17.38}, "example_bool": 1, "example_double": 5.3, "example_str": "Test"}
 LOG  147576505.60375: 1080x1920 yuv Frame (landscape-right)
 LOG  Result:  {"example_array": {"0": "Hello", "1": 1, "2": 17.38}, "example_bool": 1, "example_double": 5.3, "example_str": "Test"}
 LOG  Re-rendering camera page with inactive camera. Device: "Back Dual Wide Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
```

### Reproduceable Code

```tsx
const frameProcessor = useFrameProcessor((frame) => {
    'worklet'

    console.log(`${frame.timestamp}: ${frame.width}x${frame.height} ${frame.pixelFormat} Frame (${frame.orientation})`)
    const result = examplePlugin(frame)
    console.log('Result: ', result)
  }, [])
```

### Relevant log output

```shell
LOG  Re-rendering camera page with inactive camera. Device: "Back Dual Wide Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
 LOG  147576488.926875: 1080x1920 yuv Frame (landscape-right)
 LOG  Result:  {"example_array": {"0": "Hello", "1": 1, "2": 17.38}, "example_bool": 1, "example_double": 5.3, "example_str": "Test"}
 LOG  147576505.60375: 1080x1920 yuv Frame (landscape-right)
 LOG  Result:  {"example_array": {"0": "Hello", "1": 1, "2": 17.38}, "example_bool": 1, "example_double": 5.3, "example_str": "Test"}
 LOG  Re-rendering camera page with inactive camera. Device: "Back Dual Wide Camera" (4224x2376 photo / 1920x1080 video @ 60fps)
```

### Camera Device

```json
{
  "position": "back",
  "maxZoom": 123.75,
  "hasTorch": true,
  "supportsLowLightBoost": false,
  "supportsFocus": true,
  "physicalDevices": [
    "ultra-wide-angle-camera",
    "wide-angle-camera"
  ],
  "supportsRawCapture": false,
  "neutralZoom": 2,
  "name": "Back Dual Wide Camera",
  "hasFlash": true,
  "id": "com.apple.avfoundation.avcapturedevice.built-in_video:6",
  "sensorOrientation": "portrait",
  "hardwareLevel": "full",
  "isMultiCam": true,
  "minZoom": 1
}
```

### Device

iPhone 14 iOS 17.0.1

### VisionCamera Version

3.1.0

### Can you reproduce this issue in the VisionCamera Example app?

Yes, I can reproduce the same issue in the Example app here

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-27:
Hm that's weird, because I convert it like this: https://github.com/mrousavy/react-native-vision-camera/blob/fb6ebd9ee1b831ce1c2e0b42d123257f5f9c5459/package/ios/Frame%20Processor/JSINSObjectConversion.mm#L71-L74

mrousavy, 2023-09-27:
Maybe that's just how it gets logged by console.log? Did you do an `Array.isArray` check on that?

mnecchi, 2023-09-28:
> Maybe that's just how it gets logged by console.log? Did you do an `Array.isArray` check on that?

`Array.isArray` returns `false` and `length` is `undefined`. That's weird I agree.

mrousavy, 2023-09-30:
`Array.isArray` returns `true` and `length` is `3`. 🤨 

[image]

[image]

Just spent 20 mins debugging this only to find out that it is actually an array and console.log just logs it weirdly. Why did you say that `Array.isArray` is false? How did you test that?

closed 2023-09-30 by mrousavy

mrousavy, 2023-09-30:
Actually I know what's happening - react-native-worklets-core wraps an Array as a jsi::HostObject (see code [here](https://github.com/margelo/react-native-worklets-core/blob/main/cpp/wrappers/WKTJsiArrayWrapper.h)), but for some reason fails to unwrap it again when passing it to another context (e.g. `console.log` is another context because it calls into React JS) - and that's why Array.isArray is null.

This needs to be fixed in react-native-worklets-core somehow, but I currently don't have the time to investigate this