margelo/react-native-vision-camera #1864: Uncaught Error: Frame Processors are not enabled 🐛

CarlSchader, 2023-09-26:
### What's happening?

I'm on react-native-vision-camera version 2.15.4 and all of the sudden my team started getting this error without making code changes: 
[image]

I did find a similar issue from an earlier version of rn vision camera and all the fixes in that issue didn't work for this version.

### Reproduceable Code

```tsx
package.json:

{
	"name": "cardbook",
	"private": true,
	"scripts": {
		"android": "react-native run-android",
		"ios": "react-native run-ios",
		"start": "react-native start",
		"test": "jest",
		"lint": "eslint .",
		"jetify": "npx jetify"
	},
	"dependencies": {
		"@react-navigation/material-top-tabs": "^6.6.3",
		"@react-navigation/native": "^6.1.7",
		"@react-navigation/native-stack": "^6.9.13",
		"@rneui/base": "^4.0.0-rc.7",
		"@rneui/themed": "^4.0.0-rc.7",
		"@stripe/stripe-react-native": "^0.28.0",
		"currency-symbol-map": "^5.1.0",
		"match-sorter": "^6.3.1",
		"react": "18.2.0",
		"react-native": "0.71.4",
		"react-native-auth0": "^2.17.1",
		"react-native-dotenv": "^3.4.8",
		"react-native-fs": "^2.20.0",
		"react-native-in-app-review": "^4.3.3",
		"react-native-outside-press": "^1.2.1",
		"react-native-pager-view": "^6.2.1",
		"react-native-phone-number-input": "^2.1.0",
		"react-native-reanimated": "2.14.4",
		"react-native-safe-area-context": "^4.5.1",
		"react-native-screens": "^3.22.1",
		"react-native-set-version": "^1.1.2",
		"react-native-splash-screen": "^3.3.0",
		"react-native-swiper": "^1.6.0",
		"react-native-tab-view": "^3.5.2",
		"react-native-vector-icons": "^9.2.0",
		"react-native-vision-camera": "^2.15.4",
		"uuid": "^9.0.0",
		"yarn": "^1.22.19"
	},
	"devDependencies": {
		"@babel/core": "^7.21.3",
		"@babel/runtime": "^7.21.0",
		"@bam.tech/react-native-make": "^3.0.3",
		"@react-native-community/eslint-config": "^2.0.0",
		"babel-jest": "^26.6.3",
		"eslint": "^7.32.0",
		"jest": "^26.6.3"
	},
	"jest": {
		"preset": "react-native"
	},
	"version": "1.3.10"
}

Javascript usage:

[... 264 lines cut ...]
  "supportsDepthCapture": false,
  "position": "back",
  "formats": null
}
```

### Device

iPhone 14 and Google Pixel 6a

### VisionCamera Version

2.15.4

### Can you reproduce this issue in the VisionCamera Example app?

No, I cannot reproduce the issue in the Example app

### Additional information

- [ ] I am using Expo
- [X] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

mrousavy, 2023-09-30:
This should be fixed in latest

closed 2023-09-30 by mrousavy

2sem, 2023-10-06:
2.16.2 same