margelo/react-native-vision-camera #1867: Front Camera not loading in s23 ultra 🐛

victorkvarghese, 2023-09-27:
### What's happening?

@mrousavy I tried the latest **v3 example code.** By default front camera is loading in s23 ultra.

Then i removed everything and made it

` const devices = useCameraDevices();
`
now back camera loads, but switching to front crashes the app.

### Reproduceable Code

```tsx
CameraPage.tsx

import * as React from 'react';
import {useRef, useState, useCallback, useMemo} from 'react';
import {StyleSheet, Text, View} from 'react-native';
import {
  PinchGestureHandler,
  PinchGestureHandlerGestureEvent,
  TapGestureHandler,
} from 'react-native-gesture-handler';
import {
  CameraRuntimeError,
  PhotoFile,
  useCameraDevice,
  //useCameraDevices,
  useCameraFormat,
  // useFrameProcessor,
  VideoFile,
} from 'react-native-vision-camera';
import {Camera} from 'react-native-vision-camera';
import {
  CONTENT_SPACING,
  MAX_ZOOM_FACTOR,
  SAFE_AREA_PADDING,
  SCREEN_HEIGHT,
  SCREEN_WIDTH,
} from './Constants';
import Reanimated, {
  Extrapolate,
  interpolate,
  useAnimatedGestureHandler,
  useAnimatedProps,
  useSharedValue,
} from 'react-native-reanimated';
import {useEffect} from 'react';
import {useIsForeground} from './hooks/useIsForeground';
import {StatusBarBlurBackground} from './views/StatusBarBlurBackground';
import {CaptureButton} from './views/CaptureButton';
import {PressableOpacity} from 'react-native-pressable-opacity';
import MaterialIcon from 'react-native-vector-icons/MaterialCommunityIcons';
import IonIcon from 'react-native-vector-icons/Ionicons';
import type {Routes} from './Routes';
import type {NativeStackScreenProps} from '@react-navigation/native-stack';
import {useIsFocused} from '@react-navigation/core';
// import {examplePlugin} from './frame-processors/ExamplePlugin';

const ReanimatedCamera = Reanimated.createAnimatedComponent(Camera);
Reanimated.addWhitelistedNativeProps({
  zoom: true,
});

const SCALE_FULL_ZOOM = 3;
const BUTTON_SIZE = 40;

type Props = NativeStackScreenProps<Routes, 'CameraPage'>;
export function CameraPage({navigation}: Props): React.ReactElement {
  const camera = useRef<Camera>(null);
[... 455 lines cut ...]
                                                                                                        videoHeight: 2160,
                                                                                                        photoWidth: 3648,
                                                                                                        fieldOfView: 75.70461464712741,
                                                                                                        maxFps: [TOO BIG formatValueCalls 201 exceeded limit of 200] },
```

### Device

S23 Ultra Android

### VisionCamera Version

3.1.0

### Can you reproduce this issue in the VisionCamera Example app?

I didn't try (⚠️ your issue might get ignored & closed if you don't try this)

### Additional information

- [ ] I am using Expo
- [ ] I have enabled Frame Processors (react-native-worklets-core)
- [X] I have read the [Troubleshooting Guide](https://react-native-vision-camera.com/docs/guides/troubleshooting)
- [X] I agree to follow this project's [Code of Conduct](https://github.com/mrousavy/react-native-vision-camera/blob/main/CODE_OF_CONDUCT.md)
- [X] I searched for [similar issues in this repository](https://github.com/mrousavy/react-native-vision-camera/issues) and found none.

victorkvarghese, 2023-09-27:
@mrousavy 

I think this is related to s23 ultra. I was able to run this properly in an Oppo v27 pro. I ran the example project disabling frame processors.

Flash not working in S23 ultra
HDR click 0 crashes the App.
60 FPS was not listed in the camera module.

Not sure if this will help. Adding for clarity

victorkvarghese, 2023-09-27:
Might be related: https://github.com/mrousavy/react-native-vision-camera/issues/1801#issuecomment-1719943232

I will check removing lines

victorkvarghese, 2023-09-27:
@mrousavy 

Thing are working after removing those lines of code. I tried adding them one by one.

Device: **S23 Ultra**

_Switching from back camera to front camera crashed_ when i added **zoom** related piece of code.

```
  // Zoom
    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.R) {
      captureRequest.set(CaptureRequest.CONTROL_ZOOM_RATIO, zoom)
    } else {
      val cameraCharacteristics = cameraManager.getCameraCharacteristics(cameraId!!)
      val size = cameraCharacteristics.get(CameraCharacteristics.SENSOR_INFO_ACTIVE_ARRAY_SIZE)!!
      captureRequest.set(CaptureRequest.SCALER_CROP_REGION, size.zoomed(zoom))
    }
```

mrousavy, 2023-09-27:
1. Please stop tagging me.
2. I even explained how you can get the native (adb) logs, and even put a placeholder in there of how those should look like - yet you still ignored that and didn't paste them in there. How do you expect me to help you?

Thanks for trying the other approach though, I think we can pinpoint this to the zoom here - maybe Samsung can't handle non-clamped zoom values? What if you pass something like `0.5` or `1` here in that prop?

victorkvarghese, 2023-09-27:
Sorry for the multiple number of messages/tags.

I have attached what i could salvage from android studio. (Not able to get the entire logcat)

not sure, I will check and get back. Im using same code as the example folder. App crashes when switching the camera, not when zooming happens.

will pass the prop and get back

victorkvarghese, 2023-09-28:
Hi,

i tried to log the zoom value. 

- Intially back camera is opened and zoom value is 1
- Then if i switch to front camera it crashes
- Open app again
- Zoom out the back camera after loaded, value becomes 0.6
- Now if i switch to front camera , it works.

neutral zoom value is 1 for both back and front camera

I trying to see if there is a way to reset zoom to preferred value when camera position changes. Not sure if this is a reanimated issue that causing the camera to have an unsupported value

victorkvarghese, 2023-09-29:
Update

I passed zoom={1} as a prop. And i think issues got fixed. App is not breaking when user zooms in/out. I think without zoom prop app crashes when cameras are switched

mrousavy, 2023-09-30:
Okay so we need to clamp zoom

victorkvarghese, 2023-09-30:
Do we need to add any permissions for flash. I tried 3 different devices , flash is not working on any of them. 

HDR is also crashing for me.

mrousavy, 2023-09-30:
Flash is not yet implemented in V3. there is a feature request for that.

HDR is crashing; please create a separate bug report and share the logs.

xts-bit, 2023-10-02:
i also get this issue in Samsung devices, just a black screen on Camera view

victorkvarghese, 2023-10-02:
> i also get this issue in Samsung devices, just a black screen on Camera view

@xts-bit try const devices = useCameraDevices() and check if any cameras are detected.

xts-bit, 2023-10-02:
@victorkvarghese It didn't work black screen on V3.1

muhammedjailam, 2023-10-02:
Same issue... any fixes?

mrousavy, 2023-10-18:
Hey! Can anyone try with VisionCamera 3.4.1. and see if the issue still persists? Thanks!

mrousavy, 2023-10-18:
@xts-bit does it work if you remove those lines in `CameraSession.kt`?

victorkvarghese, 2023-10-19:
@mrousavy 

I just checked in S23 Ultra. 

1. Camera switching is working fine without issues.
2. HDR Fails ( CamerXXError [unknown/unknown: [unknown/unknown] CaptureRequest contains unconfigured Input/Output Surface!])
3. Night mode is not crashing, but i am not sure if that mode is indeed fucntional.

mrousavy, 2023-10-19:
1. Nice!
2. Okay, will try to fix that.
3. Night mode is just an Android Camera2 scene mode - it will probably not do a lot but that's how it works on Android.

mrousavy, 2023-10-24:
Hey! Did #2049 (released in 3.6.0) fix the HDR issue? Thanks!

rutvik24, 2023-10-25:
Hey! @mrousavy the same error is occurring while we process image via [react-native-image-crop-picker](https://www.npmjs.com/package/react-native-image-crop-picker) after capture and return back from native screen of package for cropping to app.

victorkvarghese, 2023-11-01:
Hi @mrousavy 
i tested the latest code with S23 ultra. 

Camera hangs up with the exception         
                                                                                            `com.mrousavy.camera.core.CameraSessionCannotBeConfiguredError: [session/cannot-create-session] Failed to create a Camera Session for Camera 0!`

Also front camera is kind of stretched out , weird aspect ratio. I tried commenting all aspect ratio related code, but same  results

Victor-Varghese, 2023-11-02:
I think this may be related to Samsung devices, not sure, I tried on vivo v27pro, and the front camera and back camera are working fine (without providing an aspect ratio)

victorkvarghese, 2023-11-03:
Update: things are fine if i downgrade to 3.4.1 , Front camera is not stretched out

mrousavy, 2023-11-07:
Okay so there is some wrong stretching going on in the preview? cc @blancham is that something you noticed as well?

victorkvarghese, 2023-11-08:
I think this is also related: https://github.com/mrousavy/react-native-vision-camera/issues/2123

Have the same issue. front camera photos are horizontal oriented, while back camera is working okay.

closed 2024-01-15 by mrousavy

bglgwyng, 2024-01-17:
I'm still facing the same issue. I opened the example app and repeated flipping the direction.
My device is Galaxy S10, Android 12.

logs: https://gist.github.com/bglgwyng/4b287419ff61a02d7c09efb9a53e8cad

mrousavy, 2024-01-17:
> Unable to retrieve camera characteristics for unknown device 1: No such file or directory (-2)

Lol that is a weird error.

ben-qiu, 2024-02-02:
I had an issue with `com.mrousavy.camera.core.CameraSessionCannotBeConfiguredError: [session/cannot-create-session] Failed to create a Camera Session for Camera 0!`
This only happens to the users who are using Samsung S23

I tried to replicate the issue and got the log
```
02-02 15:38:13.056 10536 11563 W CameraDevice-JV-0: Stream configuration failed due to: createSurfaceFromGbp:482: Camera 0: No 10-bit supported stream configurations with format 0x21 defined and profile 2, failed to create output stream
02-02 15:38:13.058 10536 11563 E CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
02-02 15:38:13.058 10536 11563 E CreateCaptureSession: Camera #0: Failed to create CameraCaptureSession #1!
02-02 15:38:13.058 10536 11563 E CameraSession: Failed to configure CameraSession! Error: [session/cannot-create-session] Failed to create a Camera Session for Camera #0!, isRunning: false, Config-Diff: Difference(deviceChanged=true, outputsChanged=true, sidePropsChanged=true, isActiveChanged=true)
02-02 15:38:13.058 10536 11563 E CameraSession: pl.k: [session/cannot-create-session] Failed to create a Camera Session for Camera #0!
```
So the problem is `CameraDevice-JV-0: Stream configuration failed due to: createSurfaceFromGbp:482: Camera 0: No 10-bit supported stream configurations with format 0x21 defined and profile 2, failed to create output stream`

I can fix the problem by setting the `photoHdr={false}`

mrousavy, 2024-02-02:
@ben-qiu does your `format` even have `supportsPhotoHdr`?

ben-qiu, 2024-02-02:
> @ben-qiu does your `format` even have `supportsPhotoHdr`?

Hi @mrousavy , I didn't log the value, but I believe it is `true`, because the original code was `photoHdr={format.supportsPhotoHdr}`

mrousavy, 2024-02-02:
Okay yea makes sense.

bglgwyng, 2024-02-05:
I've occasionally observed this behavior with VisionCamera 3.8.2 on Galaxy S21.
However, after upgrading to version 3.9.0, now I consistently experience this behavior.

mrousavy, 2024-02-05:
@bglgwyng wow - weird. Can you show me how you're using the `<Camera>` component, and also the adb logcat logs?

mrousavy, 2024-02-05:
If you are using a Frame Processor, maybe it's this change? https://github.com/mrousavy/react-native-vision-camera/compare/v3.8.2...v3.9.0-beta.0#diff-2335891be153a7966add00cea63f840e8c3c5597af11fb9f7156dac0d1fb57b1R84-R87

Could you try to disable a Frame Processor and see if that works? Also, maybe try explicitly passing a different format? Either `native` or `yuv`